reuterbal/photobooth #70: Greenscreen / Chromakey function

jopaing, 2018-11-23:
Hi,

Is a greenscreen / chromakey feature planned?

If it helps for implementation, i can provide the commands to create the final image from photo and bachground.
Unfortunately i am not able to add this directly into your scrips.

reuterbal, 2018-11-28:
I haven't had it on my list yet, but this is certainly something that would be nice to have. I'll include it in the list of feature requests.

I'd appreciate if you could point me to some resources describing how to proceed!

jopaing, 2018-11-28:
I need some tome to evaluate a good and fast solution. I will come back to you.

The coolest function without a special background, i fould here [](https://gist.github.com/drscotthawley/2d6bbffce9dda5f3057b4879c3bd4422). Unfortunately i am sure it will not work with a photobox, because of the changing light  and maybe moved camera on an event. But it is cool and link has also a demo video.

reuterbal, 2018-11-28:
This is certainly a good starting point - thanks a lot!
I think it could still work for a photobooth because the camera stands still during a shot. But, as you say, it will probably require some re-calibration once in a while because of lighting etc.
Would it be too bad to have a unicolor background as a requirement for this feature? Because I think this would then simplify things a lot.

jopaing, 2018-11-28:
sorry - that was unclear from my side.
The example was ony to show what is possible. For the project, i think only a single colour background makes sense, because of the not possible calibration during a event.
I will evaluate code snipets and give feedback.
It will need some time...