reuterbal/photobooth #71: Birthday succes

thyristor59, 2018-11-27:
**Describe your photobooth**
 I recently used this software for a test before friend marriage. Everything passed well!
My material(equipment) was only a digital reflex, a screen, a mouse connected(bound) in my raspberry pi.

**Hardware (please complete the following information):**
 - Device Raspberry Pi 3B+
 - Camera Nikon D5000
 - GPIO: No
- Printer Canon CP1300
 
 **Additional equipment**
  - No

**Software (please complete the following information):**
 - OS raspbian
 - Python version 3.6

**Problems:**
 Rasperry stop 3x 
Complete stop(ruling) without reason of the software but has cause of the heat. The raspberry was locked and just behind the screen. I think that it is the reason

**Pictures:**

[image]

50 was print

reuterbal, 2018-11-28:
Thanks for taking the time to write your feedback! Glad to hear that it worked mostly fine. 

Regarding the heat issue: are you using a heat sink on your Raspberry Pi? I found that this significantly improves heat issues and, in the same way, ensures better performance because the clock frequency isn't reduced as often.

wikijm, 2018-11-28:
Hi @thyristor59,

I'll be able to see you 'Success story' on the [wiki](https://github.com/reuterbal/photobooth/wiki/Success-stories) now 😃

Thanks for your feedback 👍

wikijm, 2018-11-28:
As this is included in the Wiki now, I'll close this.

closed 2018-11-28 by wikijm