reuterbal/photobooth #72: Problem with EOS 450D

botyas, 2018-12-03:
good again, I got a Canon Eos 450D at a good price to put in my photobooth, but when you start it gives an error because you do not recognize it, going out and entering the application several times you get to take a photo but most of the time you stay the account back in 1 does not happen I have to force the closing and restart.

With the command

gphoto2 --force-overwrite --quiet --capture-image-and-download --filename /dev/shm/photobooth.jpg

from CLI if he always takes the pictures and recognizes the camera perfectly

[image]

Log files:

[photobooth.log.2018-11-29.txt](https://github.com/reuterbal/photobooth/files/2641167/photobooth.log.2018-11-29.txt)
[photobooth.log.2018-12-02.txt](https://github.com/reuterbal/photobooth/files/2641168/photobooth.log.2018-12-02.txt)

jopaing, 2018-12-03:
I remember, that i also had the 105 error, but do not exact remember the solution.
Please check the following:
1. picture must only be saved as jpg, not jpg+raw
2. try different modes of camera. P should work
3. ensure, that auto power off is disabled

botyas, 2018-12-03:
testing a bit more all the options I realized that it is for the live preview during the countdown. Removing that option works without problem. But I would like to have this option available to be able to preview before shooting.

jopaing, 2018-12-03:
Gphoto2 supports liveview with your camera.
Try different modes on camera as pointed out on hint nr. 2

botyas, 2018-12-03:
Canon EOS 450D (PTP mode) | Image Capture, Trigger Capture, Liveview, Configuration
-- | --
Canon EOS 60D | Image Capture, Trigger Capture, Liveview, Configuration

The 2 cameras have the same options for gphoto2, with the 60D if everything works fine but with the 450D the Live Preview fails.

botyas, 2018-12-03:
In AV TV mode if it works but it shows all these errors in the log

> 2018-12-03 22:05:40,946 - root - INFO - Camera configuration:
> Camera Actions [actions]: 
>     Synchronize camera date and time with PC [syncdatetime]: 0
>     Bulb Mode [bulb]: 2
>     UI Lock [uilock]: 2
>     Drive Canon DSLR Autofocus [autofocusdrive]: 0
>     Drive Canon DSLR Manual focus [manualfocusdrive]: None (Near 1, Near 2, Near 3, None, Far 1, Far 2, Far 3, )
>     Canon EOS Zoom [eoszoom]: 0
>     Canon EOS Zoom Position [eoszoomposition]: 0,0
>     Canon EOS Viewfinder [viewfinder]: 2
>     PTP Opcode [opcode]: 0x1001,0xparam1,0xparam2
> Camera Settings [settings]: 
>     Camera Date and Time [datetime]: 1543871073
>     Quick Review Time [reviewtime]: None (None, 2 seconds, 4 seconds, 8 seconds, Hold, )
>     Camera Output [output]: Off (TFT, PC, Off, )
>     EVF Mode [evfmode]: 1 (0, 1, )
>     Owner Name [ownername]: 
>     Custom Functions Ex [customfuncex]: d4,4,1,20,2,101,1,0,10f,1,0,2,38,4,201,1,0,202,1,0,203,1,0,204,1,0,3,2c,3,50e,1,0,511,1,0,60f,1,0,4,38,4,701,1,0,704,1,0,811,1,0,80f,1,0,
>     Auto Power Off [autopoweroff]: 0
>     Depth of Field [depthoffield]: 0
>     Capture Target [capturetarget]: Internal RAM (Internal RAM, Memory card, )
>     Capture [capture]: 0
> Camera Status Information [status]: 
>     Serial Number [serialnumber]: 5bdd23d9acbd47dc9989f30ea2de510b
>     Camera Manufacturer [manufacturer]: Canon Inc.
>     Camera Model [cameramodel]: Canon EOS 450D
>     Device Version [deviceversion]: 3-1.1.0
>     Vendor Extension [vendorextension]: None
>     Camera Model [model]: 2147484022
>     PTP Version [ptpversion]: 256
>     DPOF Version [dpofversion]: 257
>     Battery Level [batterylevel]: 100%
>     Serial Number [eosserialnumber]: 330119900
>     Shutter Counter [shuttercounter]: 36286
>     Available Shots [availableshots]: 59788
> Image Settings [imgsettings]: 
>     Image Format [imageformat]: Large Fine JPEG (Large Fine JPEG, Large Normal JPEG, Medium Fine JPEG, Medium Normal JPEG, Small Fine JPEG, Small Normal JPEG, RAW + Large Fine JPEG, RAW, )
>     Image Format SD [imageformatsd]: Large Fine JPEG (Large Fine JPEG, Large Normal JPEG, Medium Fine JPEG, Medium Normal JPEG, Small Fine JPEG, Small Normal JPEG, RAW + Large Fine JPEG, RAW, )
>     Image Format CF [imageformatcf]: RAW (RAW, )
>     ISO Speed [iso]: 800 (Auto, 100, 200, 400, 800, 1600, )
>     WhiteBalance [whitebalance]: Auto (Auto, Daylight, Shadow, Cloudy, Tungsten, Fluorescent, Flash, Manual, )
>     Color Temperature [colortemperature]: 0
>     WhiteBalance Adjust A [whitebalanceadjusta]: 0 (-9, -8, -7, -6, -5, -4, -3, -2, -1, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, )
>     WhiteBalance Adjust B [whitebalanceadjustb]: 0 (-9, -8, -7, -6, -5, -4, -3, -2, -1, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, )
>     WhiteBalance X A [whitebalancexa]: 0
>     WhiteBalance X B [whitebalancexb]: 0
>     Color Space [colorspace]: sRGB (sRGB, AdobeRGB, )
> Capture Settings [capturesettings]: 
>     Exposure Compensation [exposurecompensation]: 0 (-2, -1.6, -1.3, -1, -0.6, -0.3, 0, 0.3, 0.6, 1, 1.3, 1.6, 2, )
>     Focus Mode [focusmode]: One Shot (One Shot, AI Focus, AI Servo, )
>     Storage Device [storageid]: 00020001
>     Canon Auto Exposure Mode [autoexposuremode]: P (P, TV, AV, Manual, Bulb, A_DEP, DEP, Custom, Lock, Green, Night Portrait, Sports, Portrait, Landscape, Closeup, Flash Off, )
>     Drive Mode [drivemode]: Single (Single, Continuous, Timer 10 sec, Timer 2 sec, Unknown value 0007, )
>     Picture Style [picturestyle]: Faithful (Standard, Portrait, Landscape, Neutral, Faithful, Monochrome, User defined 1, User defined 2, User defined 3, )
>     Aperture [aperture]: implicit auto (implicit auto, )
>     Shutter Speed [shutterspeed]: 1/60 (1/60, )
>     Metering Mode [meteringmode]: Center-weighted average (Evaluative, Partial, Spot, Center-weighted average, )
>     Bracket Mode [bracketmode]: 0
> Other PTP Device Properties [other]: 
>     PTP Property 0xd402 [d402]: Canon EOS 450D
>     PTP Property 0xd407 [d407]: 1
>     PTP Property 0xd406 [d406]: Unknown Initiator
> 
> 2018-12-03 22:05:40,948 - root - INFO - Using camera with preview functionality
> 2018-12-03 22:05:42,620 - root - INFO - Using background "/home/pi/photobooth/supplementals/backgrounds/fondo_MilaAntonio.jpg"
> 2018-12-03 22:06:02,121 - root - INFO - Saving picture as /home/pi/Pictures/Boda/Boda_Mila&antonio00138.jpg
> 2018-12-03 22:06:02,628 - root - INFO - Printing picture
> 2018-12-03 22:06:36,949 - root - INFO - Saving picture as /home/pi/Pictures/Boda/Boda_Mila&antonio00139.jpg
[... 68 lines cut ...]
> 2018-12-03 22:06:53,870 - gphoto2 - WARNING - (b'camera_prepare_canon_eos_capture [config.c:414]') b"'ptp_canon_eos_requestdevicepropvalue (params, PTP_DPC_CANON_EOS_SerialNumber)' failed: PTP General Error (0x2002)"
> 2018-12-03 22:06:53,871 - gphoto2 - WARNING - (b'camera_prepare_canon_eos_capture [config.c:414]') b"'ptp_canon_eos_requestdevicepropvalue (params, PTP_DPC_CANON_EOS_SerialNumber)' failed: PTP General Error (0x2002)"
> 2018-12-03 22:06:53,872 - gphoto2 - WARNING - (b'camera_prepare_canon_eos_capture [config.c:414]') b"'ptp_canon_eos_requestdevicepropvalue (params, PTP_DPC_CANON_EOS_SerialNumber)' failed: PTP General Error (0x2002)"
> 2018-12-03 22:06:53,873 - gphoto2 - WARNING - (b'camera_prepare_canon_eos_capture [config.c:414]') b"'ptp_canon_eos_requestdevicepropvalue (params, PTP_DPC_CANON_EOS_SerialNumber)' failed: PTP General Error (0x2 [... cut ...]
> 2018-12-03 22:06:54,186 - gphoto2 - WARNING - (b'camera_unprepare_canon_eos_capture [config.c:557]') b"'ptp_canon_eos_resetuilock (params)' failed: PTP Device Busy (0x2019)"
> 2018-12-03 22:06:54,187 - gphoto2 - WARNING - (b'camera_unprepare_canon_eos_capture [config.c:557]') b"'ptp_canon_eos_resetuilock (params)' failed: PTP Device Busy (0x2019)"
> 2018-12-03 22:06:54,187 - gphoto2 - WARNING - (b'camera_unprepare_canon_eos_capture [config.c:557]') b"'ptp_canon_eos_resetuilock (params)' failed: PTP Device Busy (0x2019)"
> 2018-12-03 22:06:54,244 - root - INFO - Exiting photobooth with status code 0
>

jopaing, 2018-12-03:
So it seems to be a mode setting issue.
Has your camera the latest firmware?
What about the other modes. eg P-Mode

reuterbal, 2018-12-05:
Hi @botyas 

all those warnings in your latest log are normal - they are warnings that the camera replies with "not yet ready" and gphoto logs all of them. This happens e.g. when the mirror is not yet back in place but the photobooth asks already for a preview image. This is annoying but not an error in itself, as the gphoto2 library simply blocks the function call until it is able to retrieve the preview image.

The 450D was one of the first consumer-level cameras to support these interfaces, thus I'd not be surprised if some things don't work as expected. Some have reported before that they had problems in certain modes, but at least the semi-automatic modes Tv/Av seemed to work usually. As this is something that seems to be camera specific I deem this a gphoto2-problem.

Regarding the -105-errors: I experience them sometimes as well with an EOS500D - it seems that the communication is then stuck somehow. The only thing that seems to help then is to turn off the camera, disconnect the USB cable, reconnect and switch the camera back on. This usually happens when there was some error before.

wikijm, 2018-12-31:
Hi @botyas

Does @reuterbal's answer solve your problem?
Thank you in advance for your feedback 😃 

Sincerely,
@wikijm

reuterbal, 2019-01-22:
Closed due to inactivity

closed 2019-01-22 by reuterbal