reuterbal/photobooth #73: Portable Photobooth Success

jul1792, 2018-12-28:
Hi @reuterbal, hi everyone,

I already finished my photobooth in August, but now I finally have the time to share my finished booth with you.

I didn't build it for a special event, it was just fun building it (even my girlfriend thought it was awesome). I wanted to have a booth, that I can easily take to parties of friends or family. I did take it to a family gathering over the Holidays and my cousins couldn't stop taking pictures. It was great!

Hardware:
 - Raspberry Pi 3B+
 - Raspberry Camera Module v2.1
 - GPIO: Yes, [with a big red button on top](https://www.amazon.de/gp/product/B01LZMANZ7/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1).
 
 Additional equipment
  - For the lighting I used [this ](https://www.amazon.de/gp/product/B01N02NWOJ/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1) camera light.
- Canon Selphy CP800
- [This 7" Screen](https://www.pollin.de/p/7-17-78-cm-display-set-ls-7-hdmi-dvi-vga-cvbs-120942)
- An Ikea Frame
- Wood from the hardware store
- Powerstrip
- HDMI cable

Software:
 - Some Raspbian OS, I don't the exact version.
- some software for Raspbian to get the printer working. I can't remember what it is called.

Problems:
 - No Touchscreen: I accidentally bought a no-touchscreen, so I couldn't do anything without a mouse or a keyboard. So the real problem was then, that you had to choose beetween "Print the Photo" or "do not print the Photo". Thanks to @reuterbal because he changed his program so you can choose in the settings, if the program should ask to print it or just to print it without asking.
- I also needed assistance to get the photobooth program to start on booth, so I don't need a keyboard etc. when I take this thing to a party. Again, thanks to @reuterbal I could manage this.

So thanks again @reuterbal! This wouldn't have been possible, if it wasn't for your work! Thanks a lot!

[image]
[image]
[image]
[image]

thyristor59, 2018-12-29:
hello, great installation !!!
can you inform me about the cabling of your lighting and how did you program it on the GPIO plz???
I am greatly interested!!

moritzschieder, 2019-01-03:
@jul1792 can you pls send a picture or describe how you have wired the big red button to the GPIO.
Many thanks!

reuterbal, 2019-01-10:
This looks really cool, thank you for sharing your story! The frame around the screen looks really great!

@moritzschieder For help on wiring, have a look at [my schematics](https://github.com/reuterbal/photobooth/blob/master/supplementals/wiring/wiring_bb.png)

reuterbal, 2019-01-11:
Closing this as it is already in the wiki.

closed 2019-01-11 by reuterbal