reuterbal/photobooth #74: Initial pip installation fails

lunarcloud, 2018-12-29:
running the initial "pip install -e ." fails.

**To Reproduce**
Following the INSTALL.md up to [Install photobooth with dependencies](https://github.com/reuterbal/photobooth/blob/master/INSTALL.md#install-photobooth-with-dependencies)

**Screenshots/Screencast**
https://asciinema.org/a/Npb46CFOsyNnEOAzHLlpRjBkT 

**Hardware (please complete the following information)**
Raspberry Pi 3B+

**Software (please complete the following information)**

 - OS = Raspbian 9.6 stretch
 - Python version 2.7.13 / 3.5.3
 
 **Installed packages**
 
 ```
 asciinema==1.3.0
automationhat==0.1.0
blinker==1.3
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
chardet==2.3.0
click==6.6
colorama==0.3.7
cryptography==1.7.1
drumhat==0.1.0
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==0.12.1
fourletterphat==0.1.0
gpiozero==1.4.1
idna==2.2
itsdangerous==0.24
Jinja2==2.8
keyring==10.1
keyrings.alt==1.3
MarkupSafe==0.23
microdotphat==0.2.1
mote==0.0.4
motephat==0.0.2
numpy==1.12.1
oauthlib==2.0.1
pantilthat==0.0.7
pgzero==1.2
phatbeat==0.1.1
pianohat==0.1.0
picamera==1.13
picraft==1.0
piglow==1.2.4
pigpio==1.38
Pillow==4.0.0
pkg-resources==0.0.0
pyasn1==0.1.9
pycrypto==2.6.1
pygame==1.9.3
pygobject==3.22.0
pyinotify==0.9.6
PyJWT==1.4.2
pyOpenSSL==16.2.0
pyserial==3.2.1
python-apt==1.1.0b5
pyxdg==0.25
rainbowhat==0.1.0
requests==2.12.4
requests-oauthlib==0.7.0
RPi.GPIO==0.6.5
RTIMULib==7.2.1
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
sense-emu==1.1
sense-hat==2.2.0
simplejson==3.10.0
six==1.10.0
skywriter==0.0.7
sn3218==1.2.7
spidev==3.3
ssh-import-id==5.6
touchphat==0.0.1
twython==3.4.0
unicornhathd==0.0.4
urllib3==1.19.1
virtualenv==15.1.0
Werkzeug==0.11.15
 ```

lunarcloud, 2018-12-29:
Tried suggestions from here: https://www.reddit.com/r/elementaryos/comments/79wulm/need_some_help_installing_a_python_program/dp5pqi9/

to do
```
sudo apt install libudisks2-dev libgudev-1.0-dev libgexiv2-dev libgphoto2-dev
```
and that solved my issue. (probably only needed the last one)

closed 2018-12-29 by lunarcloud

reuterbal, 2019-01-10:
Yes. See [Install required packages](https://github.com/reuterbal/photobooth/blob/master/INSTALL.md#install-required-packages) in the installation manual.