reuterbal/photobooth #75: Email Option and Gallery?

raspblu, 2018-12-31:
Is it possible for you to add in the features to email out the pictures that was just taken, and to display all pictures taken in a gallery?

reuterbal, 2019-01-10:
The software is written such that it is relatively simple to add additional postprocessing tasks. 
Simply add another class derived from `WorkerTask` in [`photobooth/worker/__init__.py`](https://github.com/reuterbal/photobooth/blob/master/photobooth/worker/__init__.py), similarly to `PictureSaver`that implements the desired actions and enable them in `Worker` in the same file.

MarioJanssen27, 2019-01-31:
Hello,

wie weit ist das standalone network mit der Gallerie?
Wo kann man helfen?

oelegeirnaert, 2019-01-31:
Hi, currently working on a webserver to be able to make a gallery:
https://github.com/oelegeirnaert/photobooth/tree/webserver/photobooth

reuterbal, 2019-02-01:
Hi @oelegeirnaert 
that looks very promising. Let me know if you need to know anything in order to integrate it!

MarioJanssen27, 2019-02-01:
Is it possible to run a standalone network with a gallery with the project?

MarioJanssen27, 2019-02-01:
Can a standalone network with a gallery be operated with the project?

---------------------------------------------------------------
in deutsch

Kann man mit dem Projekt auch ein Standalone Netzwerk mit einer Gallerie betreiben?

hotdogreen, 2019-02-09:
Well ... I’m not very pro with these things. 
Do i need to install the software again or can I just copy paste your additional files to make the gallery happen?! 
(Sorry)

oelegeirnaert, 2019-02-10:
I just did some changes in my code.

MarioJanssen27, 2019-02-11:
I cannot start the Phothbooth from (https://github.com/oelegeirnaert/photobooth). What am I doing wrong?

I get the following error message:

Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.5/runpy.py", line 142, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.5/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/pi/photobooth/photobooth/__init__.py", line 21, in <module>
    from .main import main
  File "/home/pi/photobooth/photobooth/main.py", line 40, in <module>
    from .webserver import Webserver
  File "/home/pi/photobooth/photobooth/webserver/__init__.py", line 4, in <module>
    from simple_http_server import request_map
ImportError: No module named 'simple_http_server'

wikijm, 2019-02-11:
Hi @MarioJanssen27 

It might be better to create an issue on @oelegeirnaert, because your issue is specific to it.

oelegeirnaert, 2019-02-12:
Indeed, you should create issues on my fork, if everything is finnished, we can merge again :-)
But to answer your questions, you need to install the http_server via PIP.

**pip install simple_http_server**
Another thing... I've changed the startup params so that the gui wasn't showing all the time while developing. So you beter start it like this:

**python -m photobooth --debug -w --gui**
I should change this that the GUI is popping up by default.

Enjoy!

Kr,
Oele Geirnaert

serpentes80, 2019-02-25:
@oelegeirnaert
not so easy to create an issue on your fork ;-)
there's no "issue" tab :-P

btw: tested the latest version yesterday and the website doesn't show anything. Neither pictures nor qr code or anything

hotdogreen, 2019-02-25:
Not able to get it running, either 👀

oelegeirnaert, 2019-02-26:
Just enabled it!