reuterbal/photobooth #77: Succès

thyristor59, 2019-01-03:
Hello!!!
Here is another example of success!
[image]

But:

however, with the advice of my friends to optimize the wedding day:

-how do I put the "print" and "start again" functions at the top of the screen
[image]

- flash / lighting problem (under research)
- huge slowdown of the printer.

all his questions are according to my furniture creation.
[image]

if you have ideas or solutions I would be delighted!
thank you in advance

reuterbal, 2019-01-10:
Thank you for sharing your story!

To modify the layout you can try to achieve this by changing the stylesheet [`photobooth/gui/Qt5Gui/stylesheets/pastel-1024x600.qss`](https://github.com/reuterbal/photobooth/blob/master/photobooth/gui/Qt5Gui/stylesheets/pastel-1024x600.qss). I'm not sure how much you can change the positioning of items in there using the Qt-CSS-subset.
Otherwise you will have to modify `PostprocessMessage` in [`photobooth/gui/Qt5Gui/Frames.py`](https://github.com/reuterbal/photobooth/blob/master/photobooth/gui/Qt5Gui/Frames.py)

reuterbal, 2019-01-10:
When I find time I will include your success story in the Wiki. Maybe you could add some details about the employed hardware (camera, Raspberry, screen, Printer, etc.)?

wikijm, 2019-01-11:
Hi @reuterbal 

🎉My best wishes for 2019  🎉

It's seems that @thyristor59 already did the job with #71, which is then already in the Wiki 😜

reuterbal, 2019-01-11:
Well observed - I included the link to this one.

closed 2019-01-11 by reuterbal

MarioJanssen27, 2019-02-06:
Is that a mirror as a display?

thyristor59, 2019-02-25:
Yes, it is an unstained window for the mirror effect when the software does not take pictures.
it looks more like a mirror connect than a classic photobooth