reuterbal/photobooth #78: Webcam do not work / OpenCV issue?

qwertyFrank, 2019-01-04:
Hi all,
I’m using Raspbian Desktop and rpi 3B+.
I would like to use a webcam instead a picam. I plugged in a webcam (apeman 4K) via USB to rpi and selected OpenCV by using configuration screen in Photobooth. But it doesn’t work. Even Photobooth did not start and “hang” in start mode. Do I have to install OpenCV (ver 3 or 4?) in addition?
Thanks for any hint.
Best regards,
Frank

wikijm, 2019-01-04:
Hi @qwertyFrank 

In order to help you as much as possible, can you provide what is required when creating an issue ticket?

Sincerely,
WikiJM

reuterbal, 2019-01-10:
Are you sure that your installation went through without errors? There is no package `opencv-python` installed, which is required to use a webcam.
Try `pip install opencv-python` to add the failing dependency.

reuterbal, 2019-01-25:
Did that help?

qwertyFrank, 2019-01-26:
Hi,
I'm using RPI camera now. Did not follow up Web cam anymore, therefore did not try your proposal.
Thanks

reuterbal, 2019-01-28:
Ok, closing this.

closed 2019-01-28 by reuterbal

Wikinaut, 2019-02-18:
@qwertyFrank see https://github.com/reuterbal/photobooth/issues/68