reuterbal/photobooth #79: Canon Selphy CP800 suddenly not printing anymore

jul1792, 2019-01-05:
Hi everybody,

I already used the photobooth a couple of times and already posted a success story, which you can find [here](https://github.com/reuterbal/photobooth/issues/73). Everything worked great!

But last week I went to a New Years Eve party and brought my photobooth with me. I plugged it in and took a test shot. Everything worked great, the photo was printed automatically. Then the photobooth stood there, turned on and ready. But after two hours, when we wanted to start taking some snapshots, it stopped working: It took photos perfectly, but it did not print anymore (I set the settings to not ask if I wanted them to print, but just to print every photo automatically). And this is the status right now. Photos are taken, but my Canon Selphy CP800 just doesn't react anymore. The photos are even saved.

When I type in `lpstat -p -d` it says: "No paper tray loaded, aborting!"

I also tried to delete the printer in cups, but then I have to authenticate but my username "pi" and the default password do not work. Also I tried to print a test photo, but that didn't work either. In Cups the status is the same as `lpstat -p -d`: "No paper tray loaded, aborting!"

I then unplugged the printer, inserted a SD card and printed a photo. This did work great! So the printer should not be the problem.

What can I do to solve the problem? Has anyone any suggestions?

Thanks a lot in advance!

My setup:

- Raspberry Pi 3
- Raspberry Camera Module v2.1
- Canon Selphy CP800

[image]
[image]
[image]

reuterbal, 2019-01-10:
To modify the printer settings in CUPS, you have to add the user `pi` to the group `lpadmin`, as explained in the [installation instructions](https://github.com/reuterbal/photobooth/blob/master/INSTALL.md#add-user-pi-to-group-lpadmin)

I'd try removing and re-adding the printer.

reuterbal, 2019-01-25:
Any progress?

jul1792, 2019-01-29:
Sorry, I didn't answer for so long.
I tried to add the user `pi` to the group `lpadmin` as described in the installation instructions. But in CUPS I could not delete the printer with user `pi`. I even changed the default password of `pi` with the command `sudo passwd`. But still no luck.

I will have some spare time on the weekend and do a backup of the current System. Then I will begin from the start and will install everything from scratch again. I hope this does the trick. I will let you know.

If you have any other solutions in the meantime, I will be happy to try them out :)

Thanks a lot!

reuterbal, 2019-01-30:
With `sudo passwd` you have changed the password of the root user. Simply run `passwd` to change the password of the user `pi`.

reuterbal, 2019-01-30:
What does `getent group lpadmin` say?

jul1792, 2019-02-03:
`getent group lpadmin` Says `lpadmin:x:114:pi`

With command `passwd` I have to enter the current password and I just can‘t remember it. 

Is there a way to remove CUPS and install it again?

I also tried to install everything from scratch but that failed in another part. I opened an issue with this here: 
https://github.com/reuterbal/photobooth/issues/100

reuterbal, 2019-02-04:
To reset the password of the `pi` user, run `sudo passwd pi` and enter the root password (i.e., the one that you specified by running `sudo passwd` earlier). If you don't remember that either, re-install everything and read up a little bit about Linux in general.

To reinstall CUPS, run the following:
1. Remove everything including configuration
`sudo apt purge cups`
(Pay attention to the list of packages that is removed. If something is not working later on, you might have to add some of the automatically removed packages by hand later on).
2. Install CUPS
`sudo apt install cups`
3. Install gutenprint or whatever else you need.

closed 2019-06-21 by reuterbal