reuterbal/photobooth #80: Using External Flash in Canon "M" mode makes liveview too dark

vomo1, 2019-01-07:
@first thank you providing this great project!

**My issue:** 
Preview dark (black display) in manual camera mode.

**My Setup:**
Raspi 3B+ with 9" touch display
Canon EOS 550D 
External Flash Elinchrome RXONE
Printer DNP DS620

**Description:**
I'm using a Canon EOS 550D in my booth. 
In auto setup mode everything works fine.
In lager setups I want to use a external flash. Then, the picture taken is too bright because camera do not know usage of external flash. Therefore I have to switch to manual mode to controll the camera.
In this mode, liveview isnt possible because its - driven by the EOS - too dark.

Improvement: By using Canon addon Firmware "Magic Lantern" 
_https://magiclantern.fm_
its possible to reduce liveview FPS to values in range 3-7 fps, wich is sufficient to pose in front of the box. By this mod, liveview is bright enough to display.

Problem: Photobooth Software does not switch the cam in this mode automatically. after each photo taken, I have to switch the cam manually in live view.

**Question:** (i'm a noob)
In which region of your software do you control the cam for liveview? Maybe I can figure out the gphoto commands to bring the cam in correct mode.

Best regards and thank you very much!
Vomo

vomo1, 2019-01-07:
... same issue in ml forum, maybe no solution until now.

https://www.magiclantern.fm/forum/index.php?topic=18937.0

sratxp, 2019-01-08:
Had the same problem as you in my setup. My solution was to use a different shutter speed for the preview.
You can find the code in my fork (check the last commit in the preparing_for_wed branch.

Hope that helps

sratxp, 2019-01-12:
@serpentes80 
I haven't followed the changes in the master branch from @reuterbal . He might have done some changes that doesn't work with my code. It would be best if you could download my repo and check if it runs. I currently don't have time to work on the photobooth.

If its not working you could try and hardcode the shutterspeed for preview and for pictures. Therefore you only need the code in line 164 of CameraGphoto.py  `gp.check_result(gp.gp_widget_set_value(shutterspeed, "1"))`
This line sets the shutterspeed for priview to 1 sec. Which should be bright enough for dark environments. I would simply add this line to the CameraGphoto.py and see if your preview is brighter.  If it works you will observe that your actual shots are also made with sutterspeed = 1sec. To prevent this you then hardcode a different shutterspeed for your pictures something like 1/125 sec (this value should be sufficient for pictures with flash and you don't need to adjust it)

vomo1, 2019-01-13:
Thank you all for your help!

I wasn't able to add the feature in original actual reuterbal SW, but sratxp's repo runs.
Maybe Balthasar Reuter is able to add this feature (configurable shutterspeed would be great) in his Code?! ;-)

reuterbal, 2019-01-14:
I'm not sure if this is a good idea. It seems, the names of all configuration options are highly dependent on camera manufacturer and model, thus I would want to avoid adding too many specific options in there (see the issues with Nikon cameras in another bug report).
If someone can provide a portable solution I'm happy to integrate it.

serpentes80, 2019-02-05:
@sratxp 
I've downloaded your repo... and yes, it runs...
but as you already said... you didn't follow @reutebal s changes...

and in fact I would need a mixture of different forks :(

longer live view exposure from @sratxp + webserver from another fork...

sadly I'm not a coder to work this out on my own :(

reuterbal, 2019-02-06:
@sratxp If you implement the features in a generic way, such that it is configurable and easy to switch off to make it work with different camera models, I'm happy to merge a pull request!

sratxp, 2019-02-06:
@reuterbal I will try to make it generic and create a pull requiest. Need to dig a bit deeper into gphoto and qt

MarioJanssen27, 2019-03-18:
What do I have to do to fix the same problem?

nexusfatal, 2019-08-02:
Hi @vomo1 The easiest way to fix a (too) dark liveview image is to turn off exposure simulation in the liveview settings.
However, this solution is kind of Canon specific, but it will at least fix your problem.