reuterbal/photobooth #81: Button screen image too large

ther3zz, 2019-01-08:
Currently using the office raspberry pi 7 inch touch screen (pi 3b+). I have the photobooth application set to full screen and I've configured the proper resolution within its settings (800x480) but when on the "Hit the button" screen, the word "button" and the arrow are cut from the screen.

Does anyone know how I can fix this? (I've added a [screenshot](https://app.box.com/s/m21oftddev1vyau6x4qoriwsrjj20t13) of what I see)

Thanks for the help!

matthieusalvat, 2019-01-10:
Hi,

I have the same problem on my raspberry screen. I proposed a pull request (https://github.com/reuterbal/photobooth/pull/83) to add 2 appearances styles that work with low resolution screen.

You can test this code by copy/paste this lines in your photobooth directory : 
git remote set-url origin https://github.com/matthieusalvat/photobooth.git
git pull

Then you can launch photobooth and choose "dark (800*600)"  or "pastel (800*600)" in "appearance" settings.

reuterbal, 2019-01-10:
My stylesheets are not very sophisticated - I simply made them such that they work for my setup. Thanks to @matthieusalvat for the PR, it is included in the master branch now.

If someone feels like making stylesheets more generic, I'm happy to merge any PRs.

closed 2019-01-10 by reuterbal

ther3zz, 2019-01-10:
> Hi,
>
> I have the same problem on my raspberry screen. I proposed a pull request (#83) to add 2 appearances styles that work with low resolution screen.
>
> You can test this code by copy/paste this [...]

Thank you very much! I really appreciate it!

and thanks to reuterbal for merging the fix with the master!