reuterbal/photobooth #82: Change Caption

hotdogreen, 2019-01-09:
Hi there, 

great project. 

I'm pretty new to all this stuff but got your software running due to the good documentation.
My Question..How can I change the caption Hit the Button, because I am using a food switch instead. :)

MarioJanssen27, 2019-01-13:
Hello,

can you change the themes? I want to change the colors and the one arrow. I didn't find any pictures to change themes. In the menu I found three possibilities but I wanted to make my own themes.

Greetings
_____________________________________________________________________________
Weil mein Englisch nicht so gut ist. Das gleiche nochmal in deutsch.

_____________________________________________________________________________
Hallo,

kann man die Themes ändern? Ich möchte die Farben und den einen Pfeil  ändern. Ich habe keine Bilder zum ändern Themes gefunden. Im Menu habe ich drei Möglichkeiten gefunden aber ich wollte meine eigenen Themes machen.

Gruß

reuterbal, 2019-01-14:
Hi @MarioJanssen27 ,
your English is fine!
All files corresponding to the themes are located in [`photobooth/gui/Qt5Gui`](https://github.com/reuterbal/photobooth/tree/master/photobooth/gui/Qt5Gui):
- folder `images` contains the used images (camera, arrow)
- folder `stylesheets` contains the stylesheet files that modify the appearance

You can easily add another stylesheet file there to create your own theme. There is only one thing that you have to do afterwards to make it availabe: 
- Open the file `__init__.py` in the same folder 
- Add your new stylesheet to the list of available stylesheets, e.g. like this:
```python
styles = (('default', 'stylesheets/default.qss'),
          ('dark (1024 x 600 px)', 'stylesheets/dark-1024x600.qss'),
          ('dark (800 x 600 px)', 'stylesheets/dark-800x600.qss'),
          ('pastel (1024 x 600 px)', 'stylesheets/pastel-1024x600.qss'),
          ('pastel (800 x 600 px)', 'stylesheets/pastel-800x600.qss'),
          ('name of new stylesheet (choose anything', 'stylesheets/filename-of-new-stylesheet.qss'))
```

MarioJanssen27, 2019-01-14:
Hello, Reuterbal,

thanks for your quick help. I have added another theme to Photobooth! That worked well. I also found the arrow.

But how can only the background change colors? I thought it was just a picture I had to change. I understood that I had to change the file "photobooth/gui/Qt5Gui/__init__.py" in the line Background. But I can't see a preview of what I changed.
Is there a program for this?

and 

I wanted to have a new straight arrow point up center. When the text comes:
"#: photobooth/gui/Qt5Gui/Frames.py:68
msgid "photobooth"
msgstr "Photobooth"

#: photobooth/gui/Qt5Gui/Frames.py:87
&lt;font color="#ffff00"&gt;-==- sync:ßÇÈâÈâ
msgstr "Squeeze the""

But how do I get the arrow in the Miite

and 

the text under the arrow?

------------------------------------------------------------------------------------------------------
Weil mein Englisch nicht so gut ist. Das gleiche nochmal in deutsch.
-------------------------------------------------------------------------------------------------------
Hallo Reuterbal,

danke für deine schnelle Hilfe. Ich habe die Photobooth um ein weiteres Theme erweitert! Das hat gut funkioniert. Den Pfeil habe ich auch gefunden.

Aber wie kann nur die Hintergrund Farben ändern. Ich habe gedacht das wäre nur ein Bild was ich ändern müsste. Ich habe wohl verstanden das ich in die Datei "photobooth/gui/Qt5Gui/__init__.py" im Zeile Background  ändern muss. Aber ich kann keine Vorschau sehen was ich geändert habe.
Gibt es dafür ein Programm?

und 

Ich wollte einen neuen geraden Pfeil mitte nach oben zeigen lassen. Wenn der text kommt:
"#: photobooth/gui/Qt5Gui/Frames.py:68
msgid "photobooth"
msgstr "Photobooth"

#: photobooth/gui/Qt5Gui/Frames.py:87
msgid "Hit the"
msgstr "Drück' den""

Aber wie bekomme ich den Pfeil in die Miite

und 

den Text unter den Pfeil?

reuterbal, 2019-01-15:
For colors, sizes etc. you will have to modify the stylesheet file.
You might also be able to rearrange the layout in there, if not, you will have to modify `IdleMessage` in [`photobooth/gui/Qt5Gui/Frames.py`](https://github.com/reuterbal/photobooth/blob/master/photobooth/gui/Qt5Gui/Frames.py)

MarioJanssen27, 2019-01-15:
Thanks for the quick help. With the colors have already changed the theme the arrow I take tomorrow.

Danke für die schnelle Hilfe. Mit den Farben haben das Theme schon geändert den Pfeil nehme ich mir morgen vor.

hotdogreen, 2019-02-04:
Thanks..my question is answered

closed 2019-02-04 by hotdogreen