reuterbal/photobooth #84: Live Preview problems

call-me-matt, 2019-01-12:
**Describe the bug**

Live Preview does not work for my camera (Canon PowerShot G5X). Please help me to identify the problem.

**To Reproduce**

I get the following error messages:
- python-gphoto2: Before first preview is shown the error message: ```Camera: [-110] I/O in progress``` appears and the user can only return to main menu or retry
- gphoto2-cffi: the lenses extract and retract several time, followed by the following error message: ```Unspecified Error```. User can only return to main menu or retry.
- gphoto2-commandline: The camera seems to take one picture at startup, then the lenses retract and I see the "push the button" screen. The countdown is shown without preview (white image only), for each picture the lenses extract and retract afterwards. No error message is shown, even when started with the ``` --debug``` option.

**Expected behavior**

The photobooth software has been tested with an EOS D70 of a friend, live preview works fine with this camera.
On the other hand, my camera shows a working live preview with the software "entangle" which is also based on gphoto2. This is why I am hoping to adapt the photobooth software to work as well.

**Hardware (please complete the following information)**

 - Device Ubuntu Laptop
 - Camera Canon PowerShot G5X
 - GPIO: No

reuterbal, 2019-01-12:
Hi @call-me-matt 
what version of gphoto2 are you using? According to the [compatibility list](http://gphoto.sourceforge.net/doc/remote/), this model requires at least version 2.5.21.1 or newer, which was released only recently in December.
Try updating gphoto2 to the latest version (you can use [this script](https://github.com/gonzalo/gphoto2-updater)) and see if that helps.

call-me-matt, 2019-01-12:
Hi and thank you for your quick reply. I resolved this issue by disabling ```set_idle``` in camera state machine. I think its a problem with a camera owner property which takes some time to retrieve and thus blocks the camera for other actions.

closed 2019-01-12 by call-me-matt

call-me-matt, 2019-01-14:
Just to precise: the idle-active settings did not work and the camera froze after some time. I resolved this now by settiing only once to active in the prepareCapture state and then teardown after each round. This results in lense retraction after each session, but it resolves the problem.

reuterbal, 2019-01-15:
You can also try to figure out the idle/active command from the list of configuration values printed in the log.

reuterbal, 2019-01-22:
The 'Unknown Value...' thing looks fishy. Have you updated your gphoto2 installation?

call-me-matt, 2019-01-22:
I am using the latest version of gphoto2:
```
gphoto2         2.5.20.1       gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.22.1       all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
```
But I had to [implement](https://github.com/gphoto/libgphoto2/issues/344) new opcodes for my camera to work, there might be other bugs still to fix. What output would you expect?

reuterbal, 2019-01-22:
Actually no idea what the correct value would be, but probably something like 'screen' or 'PC'. But I think you have to follow up on this upstream at gphoto2.

SparkyCola, 2019-02-05:
@call-me-matt I've got image-preview and image-capture (though not capture-and-download) working for the G7X (the G5X's younger sister which seems quite similar in the way PTP-IP is implemented). [My changes](https://github.com/SparkyCola/libgphoto2) are very hacky since I can't get the config (filed a bug report for it already), I just thought maybe you want to give it a try anyway.

call-me-matt, 2019-02-05:
@SparkyCola - thank you for your feedback. By chance I have also G7X and it works fine - image-preview, image-capture as well as capture-and-download (with my adaptions that are now included in libgphoto2 since version 2.5.22.1). The only issue is power save mode which I prevent by retracting the lenses after each "round".

SparkyCola, 2019-02-06:
@call-me-matt I have seen some idle issues, from what I can see, libgphoto seems to close the session which never happened in my Canon Connect live view dumps. Btw how did you get your cameras into the live view mode, I wrote a little python program which handles SSDP and HTTP to act like the Canon app.