reuterbal/photobooth #85: Success story: Photobooth 1.0

reuterbal, 2019-01-14:
**Describe your photobooth**
This photobooth was the reason for the rewrite of the initial software version. The original motivation was a friend's wedding but it has been used before that and afterwards at a couple of events

**Hardware (please complete the following information):**
 - Raspberry Pi 3B
 - Canon EOS 500D with Canon EF 28mm f2.8 lens
 - GPIO: Trigger-button, Trigger-LED, Exit-LED (inside the housing), color LED-strip
 - Canon Selphy CP1300 (connected via USB)
 
 **Additional equipment**
  - Cheap studio lamp from Amazon
  - External USB battery
  - External power supply for camera
  - 1024x600px touch screen from Pollin
  - Emergency-switch with lamp from eBay (230V lamp replaced by 12V LED)
  - LED strip for lighting during idle, red blinking during countdown, and white light (as "flash")

**Software (please complete the following information):**
 - Latest Raspbian
 - 3.5.x

**Modifications:**
 - This is the original setup from the repository. Modifications not included in the repository are only custom backgrounds for the assembled pictures.

**Problems:**
 - Printer hangs (presumably only when there is a longer print queue). It just stays displaying "busy". Most of the times it helps to unplug the power supply of the printer and to turn it back on. It then continues to print missing photos. Sometimes, jobs in the queue are put on hold and must be re-enabled in CUPS. No idea how to fix this, as CUPS has the jobs...
- I experienced once or twice out-of-memory issues or a random hang of the photobooth without anything in the logs. Keep an eye out ;-)

**Pictures:**
Some pictures from the setup of an event at the weekend. This is usually what it looks like:
- Photobooth + Studio lamp
- Some background
- Printer and masks on a nearby table
- Photo-album and some handicraft work equipment to glue photos in and create a guest-book on-site.
[image]
[image]
[image]
Some older photos from the interior and tests in my living room:
[image]
[image]
[image]

qwertyFrank, 2019-01-14:
Looks very good. Can you share the schematics, componentes used to drive color LED-strip? Which LED-strip did you use?
Thanks in advance.

botyas, 2019-01-14:
I have the assembly that appears on this plane  `photobooth/supplementals/wiring/wiring_bb.png`
       with the circuits that I retrubal indicated to me [in this answer](https://github.com/reuterbal/photobooth/issues/65#issuecomment-435306123)  and it works correctly.

reuterbal, 2019-01-15:
The wiring schematics linked to by @botyas are mine, the LED strip is included there as red/green/blue LEDs. Some info about the LED strip is included in #26

thyristor59, 2019-01-15:
great, great job !!
how did you get the liveview shown on the screen?
would you have the reference of the amazon lamp?
thank you in advance !

reuterbal, 2019-01-16:
Live view is an intrinsic component of the software package? How do you not get it?

The lamp could be this one: https://www.amazon.de/dp/B01EFDROV0/

reuterbal, 2019-01-25:
Closing this as the story is in the wiki.

closed 2019-01-25 by reuterbal