reuterbal/photobooth #86: [Question] Add logo/text to printout only

call-me-matt, 2019-01-14:
Hello.

I would like show the compilation image as currently implemented after having taken four pictures but print a different assembly. This is because I would like to print two photostrips per printout, as shown in the example below. As my screen is in landscape mode, the overview at the end shall remain unchanged.
I have written the python code to create the print picture, but I am unsure where to add this. Is that a postprocessor task? What would you recommend?

[image]

reuterbal, 2019-01-15:
At the moment, the picture is assembled in [`photobooth/camera/__init__.py`](https://github.com/reuterbal/photobooth/blob/master/photobooth/camera/__init__.py) to avoid having to send the data of all pictures but only of the assembled picture to a different process.

You will either have to include the data of all pictures in the CameraEvent used to trigger the review state (this way you can use this to create a postprocess task that assembles your print picture) or keep the pictures after assembling the displayed picture in above linked class and send a direct message to create the printout once triggered (seems more complicate).

For sake of simplicity I would choose the first option and test if it makes too much of a perfomance impact. That way, the printout is assembled in the background and won't interfere with the photobooth frontend but it can impact the overall performance and you might hit some memory limits.

call-me-matt, 2019-01-22:
Implemented as proposed by you. The performance is OK on my Raspi3B - you can notice some delay for some seconds if a new capture session is started right away, but that is alright.
From my side, this topic can be closed.

reuterbal, 2019-01-22:
Glad it works. If you find time, I appreciate a success story, maybe including a link to your modifications. It might help others looking to achieve the same!

closed 2019-01-22 by reuterbal

kun93, 2019-03-21:
Is there anyway you can share your code? I would like to print two photostrips per printout As well

call-me-matt, 2019-03-21:
sure. This is my modified print-function from PrinterPyQt5.py:

```python
from PIL import Image, ImageOps, ImageFilter, ImageQt

    def print(self, pictures):
    
        logging.info ("Generating printout compilation")
        
        x = 547
        y = 365
        x1 = 16
        y1 = 26
        x2 = 636
        abstand_y = 365+31
        border = 10

        pic = Image.open("template.jpg").copy()

        for n in range(4):

            shot = Image.open(pictures[n])
            shot = ImageOps.expand(shot, border=(border,border))
            shot.thumbnail((x,y))
            # left
            resized = shot.crop((0,0,x,y))
            box = (x1, y1+n*abstand_y, x+x1, y+y1+n*abstand_y)
            pic.paste(resized, box)
            # right
            shot = shot.convert('L').filter(ImageFilter.SHARPEN)
            resized = shot.crop((0,0,x,y))
            box = (x2, y1+n*abstand_y, x+x2, y+y1+n*abstand_y)
            pic.paste(resized, box)

        #logging.info ("writing compiled image to disk")
        #pic.save(time.strftime("%Y-%m-%d/%Hh%Ms%S.png"))
        
        picture = ImageQt.ImageQt(pic)

        if self._print_pdf:
            self._printer.setOutputFileName('print_%d.pdf' % self._counter)
            self._counter += 1

        logging.info('Printing picture')

        picture = picture.scaled(self._printer.paperRect().size(),
                                 QtCore.Qt.KeepAspectRatio,
                                 QtCore.Qt.SmoothTransformation)

        printable_size = self._printer.pageRect(QPrinter.DevicePixel)
        origin = ((printable_size.width() - picture.width()) // 2,
                  (printable_size.height() - picture.height()) // 2)

        painter = QtGui.QPainter(self._printer)
        painter.drawImage(QtCore.QPoint(*origin), picture)
        painter.end()
```

kun93, 2019-03-21:
Thank you ! Did you have to modify anything else ? I’m wondering if I have to adjust the picture px in the config .

kun93, 2019-03-21:
@call-me-matt Sorry to bug again. I'm running into a few issues with that code. 

1. Is the final image suppose to be named `template.jpg` or are you using that for the background image? I keep getting the error below, I put in a placeholder but I don't think that's right.
`FileNotFoundError: [Errno 2] No such file or directory: 'template.jpg'`

2. If I add a placeholder image, I get further but a new error is produced. It get's to this part `Generating printout compilation` and then errors with `TypeError: 'ImageQt' object does not support indexing`

Fork
https://github.com/kun93/photobooth

call-me-matt, 2019-03-21:
Well, yes. I had to change quite a bit in order to pass the pictures to the post-processor. Let's see if I get everything together again (there might be mistakes)...

In the gui/Qt5Gui/PyQt5Gui.py:
```python
class PyQt5Gui(GuiSkeleton):
        self._picture = None
        self._pictures = []
(...)
    def showReview(self, state):

        picture = Image.open(state.picture)
        pictures = state.pictures
        self._picture = ImageQt.ImageQt(picture)
        self._pictures = pictures
        review_time = self._cfg.getInt('Photobooth', 'display_time') * 1000
        self._setWidget(Frames.PictureMessage(self._picture))
        QtCore.QTimer.singleShot(
            review_time,
            lambda: self._comm.send(Workers.MASTER, GuiEvent('postprocess')))
        self._postprocess.do(self._picture, self._pictures)

    def showPostprocess(self, state):

        tasks = self._postprocess.get(self._picture, self._pictures)
        postproc_t = self._cfg.getInt('Photobooth', 'postprocess_time')

        Frames.PostprocessMessage(
            self._gui.centralWidget(), tasks, self._worker,
            lambda: self._comm.send(Workers.MASTER, GuiEvent('idle')),
            postproc_t * 1000)
```

In the StateMachine.py:
```python
class CameraEvent(Event):

    def __init__(self, name, picture=None, pictures=[]):

        super().__init__(name)
        self._picture = picture
        self._pictures = pictures

    @property
    def picture(self):

        return self._picture
        
        
    @property
    def pictures(self):

        return self._pictures
```

In the camera/__init__.py:
```python
    def prepareCapture(self):
    
        logging.debug('prepare capture')

        self.setActive()
        self._pictures = []

    def assemblePicture(self):

        #self.setIdle()
        self.teardown('wtf')
        
        # full screen compilation
        
        picture = self._template.copy()
        for i in range(self._pic_dims.totalNumPictures):
            shot = Image.open(self._pictures[i])
            resized = shot.resize(self._pic_dims.thumbnailSize)
            picture.paste(resized, self._pic_dims.thumbnailOffset[i])

        byte_data = BytesIO()
        picture.save(byte_data, format='jpeg')
        self._comm.send(Workers.MASTER,
                        StateMachine.CameraEvent('review', byte_data, self._pictures))
                        
        self._pictures = []
```

In the gui/GuiPostprocessor.py
```python
class GuiPostprocessor:

    def __init__(self, config):

        super().__init__()

        self._get_task_list = []
        self._do_task_list = []

        if config.getBool('Printer', 'enable'):
            module = config.get('Printer', 'module')
            paper_size = (config.getInt('Printer', 'width'),
                          config.getInt('Printer', 'height'))
            pdf = config.getBool('Printer', 'pdf')
            if config.getBool('Printer', 'confirmation'):
                self._get_task_list.append(
                    PrintPostprocess(module, paper_size, pdf))
            else:
                self._do_task_list.append(
                    PrintPostprocess(module, paper_size, pdf))

    def get(self, picture, pictures):

        return [task.get(picture,pictures) for task in self._get_task_list]

    def do(self, picture, pictures):

        for task in self._do_task_list:
            task.get(picture,pictures).action()

class PostprocessTask:

    def __init__(self):

        super().__init__()

    def get(self, picture, pictures):

        raise NotImplementedError()

class PrintPostprocess(PostprocessTask):

    def __init__(self, printer_module, paper_size, is_pdf, **kwargs):

        super().__init__(**kwargs)

        Printer = lookup_and_import(printer.modules, printer_module, 'printer')
        self._printer = Printer(paper_size, is_pdf)

    def get(self, picture, pictures):
    
        return PostprocessItem('Print', lambda: self._printer.print(pictures))
```

In the printer/PrinterPyQt5.py
as described above

For the template - yes, you need to store the background template as you found out.

alopezbu, 2019-06-02:
@call-me-matt 
I have copied your version of the code, and I got the following error and the photobooth does not start:

2019-06-02 19:59:11,288 - root - INFO - Initializing photobooth
2019-06-02 19:59:11,290 - root - INFO - Photobooth version: 0.4.dev81+g471fb07
2019-06-02 19:59:11,297 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2019-06-02 19:59:11,310 - root - INFO - Reading config file "photobooth.cfg"
2019-06-02 19:59:11,457 - root - INFO - Number of last existing file: 0
2019-06-02 19:59:11,469 - root - INFO - Saving pictures as "2019-06-02/photoboothXXXXX.jpg"
2019-06-02 19:59:11,482 - root - INFO - Number of last existing file: 0
2019-06-02 19:59:11,485 - root - INFO - Saving pictures as "2019-06-02/photobooth_shot_XXXXX.jpg"
2019-06-02 19:59:11,856 - root - INFO - Using PiCamera
2019-06-02 19:59:11,900 - root - INFO - GPIO enabled (lamp_pin=4, trigger_pin=26, exit_pin=23, rgb_pins=(27, 21, 17))
2019-06-02 19:59:12,081 - root - INFO - Using camera with preview functionality
Process GuiProcess-3:
Traceback (most recent call last):
  File "/usr/lib/python3.5/multiprocessing/process.py", line 249, in _bootstrap
    self.run()
  File "/home/pi/photobooth/photobooth/main.py", line 87, in run
    'gui')
  File "/home/pi/photobooth/photobooth/util.py", line 35, in lookup_and_import
    'photobooth.' + package + '.' + result[0])
  File "/home/pi/photobooth/.venv/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 673, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/pi/photobooth/photobooth/gui/Qt5Gui/__init__.py", line 27, in <module>
    from .PyQt5Gui import PyQt5Gui  # noqa
  File "/home/pi/photobooth/photobooth/gui/Qt5Gui/PyQt5Gui.py", line 41, in <module>
    class PyQt5Gui(GuiSkeleton):
  File "/home/pi/photobooth/photobooth/gui/Qt5Gui/PyQt5Gui.py", line 42, in PyQt5Gui
    self._picture = None
NameError: name 'self' is not defined

I would appreciate if you can help me.