reuterbal/photobooth #88: Question: Can I center the text?

raeMaem, 2019-01-15:
Hello and thanks for this awesome, well made, really complete project.

Instead of the text saying "Press the button" with the arrow, I've deleted the arrow files and replaced the text by "Press the screen" because I want to use a touchscreen instead. Until here I was fine, but can you tell me how I can center the text please? The text is still here as if the arrows were on the right and it's not centered.

Once again thanks for this awesome project

MarioJanssen27, 2019-01-15:
Hello raeMaem, 

I think your problem is already being discussed and explained here. Link: [https://github.com/reuterbal/photobooth/issues/82#issuecomment-454300816]

raeMaem, 2019-01-15:
Hello MarioJanssen27,

You're totally right. I had started to read that discussion but didn't follow through the end, sorry.

Thank you for your help.

reuterbal, 2019-01-15:
Thanks for the link, closing this as a duplicate.

closed 2019-01-15 by reuterbal