reuterbal/photobooth #89: Three problems: 1. Gpio, 2. The photo quality is reduced and 3. Save on a USB stick.

MarioJanssen27, 2019-01-15:
Hello, I have three problems:

1. **Gpio**: I wanted to build my Photobooth without a touchscreen.  I don't need the printer. The only problem I have is the first selection screen (see photo 1). I can't get any further with the Gpio.  I would have to work with the mouse or the touch screen. I only have two Gpios.

2. **The photo quality is reduced.** I don't mean the number of pixels but the photo quality. I wanted to increase the photo quality. I use Gphoto2 because I have a Canon.

3. **Save on a USB stick.** I cannot use a USB stick because I have to make the following settings (see photo 2).
File manager/ edit/ settings/ disk management/ 
The following settings are not checked:
- Mount data carrier automatically at program start.
- Mount removable media automatically when inserting.
I must remove the hooks otherwise my camera is not found by Gphoto2.

______________________________________________________
in Deutsch
______________________________________________________
Hallo ich habe drei Probleme:

1. **Gpio**: Ich wollte meine Photobooth ohne Touchscreen aufbauen.  Den Drucker brauche ich nicht. Das einzige Probleme das ich habe ist der Erste Auswahl Bildschirm (siehe Foto 1). Da komme ich nicht mit den Gpio weiter. Ich müsste mit Mouse oder mit dem Touchscreen arbeiten. Ich habe habe bloß zwei Gpios.

2. **Die Foto Qualität ist heruntergesetzt.** ich meine nicht die Anzahl der Pixel sondern die Foto Qualität. Ich wollte die Foto Qualität erhöhen. Ich benutze Gphoto2 weil ich eine Canon habe.

3. **Speichern auf einen USB-Stick.** Ich kann keinen USB-Stick verwenden weil ich folgende Einstellungen vornehmen muss (siehe Foto 2).
Dateimanager/ Bearbeiten/ Einstellungen/ Datenträgerverwaltung/ 
Es sind folgenden Einstellungen keine Haken bei:
- Datenträger bei Programmstart automatisch einhängen.
- Wechseldatenträger automatisch beim Einlegen einbinden.
Ich muss die Haken entfernen sonst wird meine Kamera von Gphoto2 nicht gefunden.

reuterbal, 2019-01-15:
Hi,

I think you forgot to attach the photos.

1. See [this Wiki entry](https://github.com/reuterbal/photobooth/wiki/Tips-and-tricks#start-photobooth-directly-instead-of-showing-welcomesettings-screen) regarding direct start of the photobooth.

2. I don't understand. What is reduced?

3. Install `autofs` and add the USB drive there or add it in `/etc/fdisk`

reuterbal, 2019-01-16:
Regarding 3.: Did you follow [these instructions](https://github.com/reuterbal/photobooth/blob/master/INSTALL.md#remove-some-files-to-get-gphoto2-working)? This should avoid mounting the SD card of the camera.

reuterbal, 2019-01-16:
1. Great.

2. You can pass additional options in [`assemblePicture`](https://github.com/reuterbal/photobooth/blob/master/photobooth/camera/__init__.py) to the save() function.
See [the documentation of the save function](https://pillow.readthedocs.io/en/3.1.x/reference/Image.html#PIL.Image.Image.save) and [the available options](https://pillow.readthedocs.io/en/3.1.x/handbook/image-file-formats.html#jpeg). Try, e.g. `quality=95` and see, if this helps. Please let me know if it makes a difference.

3. Read [this](https://www.modmypi.com/blog/how-to-mount-an-external-hard-drive-on-the-raspberry-pi-raspian) (first hit in a search).

MarioJanssen27, 2019-01-16:
> 1. Great.
> 2. You can pass additional options in [`assemblePicture`](https://github.com/reuterbal/photobooth/blob/master/photobooth/camera/__init__.py) to the save() function.
>    See [the [...]

**2.** Now I didn't hear you. What do you mean by that? I found this in the file but. I don't understand what it says!
    def assemblePicture(self):

        self.setIdle()

        picture = self._template.copy()
        for i in range(self._pic_dims.totalNumPictures):
            shot = Image.open(self._pictures[i])
            resized = shot.resize(self._pic_dims.thumbnailSize)
            picture.paste(resized, self._pic_dims.thumbnailOffset[i])

        byte_data = BytesIO()
        picture.save(byte_data, format='jpeg')
        self._comm.send(Workers.MASTER,
                        StateMachine.CameraEvent('review', byte_data))
        self._pictures = []

**3.** I managed quite easily. The link didn't work or the tip.

But it was quite simple.

With this help I was able to reset all options and the USB drive was mounted again.

> Regarding 3.: Did you follow [these instructions](https://github.com/reuterbal/photobooth/blob/master/INSTALL.md#remove-some-files-to-get-gphoto2-working)? This should avoid mounting the SD card of the camera.

But now I can't write on the stick. But I think this is due to the NTFS format. I'm trying that again this week with another stick.

-----------------------------------------------------------------------------------------
In German/ In Deustch

**2.**  Jetzt habe ich dich nicht mehr verstanden. Was meinst du damit? Ich habe das in der Datei gefunden. Aber ich verstehe nicht, was da steht!?

def assemblePicture(self):

        self.setIdle()

        picture = self._template.copy()
        for i in range(self._pic_dims.totalNumPictures):
            shot = Image.open(self._pictures[i])
            resized = shot.resize(self._pic_dims.thumbnailSize)
            picture.paste(resized, self._pic_dims.thumbnailOffset[i])

        byte_data = BytesIO()
        picture.save(byte_data, format='jpeg')
        self._comm.send(Workers.MASTER,
                        StateMachine.CameraEvent('review', byte_data))
        self._pictures = []

**3.** Ich habe ganz einfach hinbekommen. Der Link hat nicht funktioniert bzw. der Tipp.

Aber er was ganz einfach.

Durch diese Hilfe konnte ich auch wieder alle Optionen wieder zurück stellen und das USB-Laufwerk wurde wieder eingebunden.

> Regarding 3.: Did you follow [these instructions](https://github.com/reuterbal/photobooth/blob/master/INSTALL.md#remove-some-files-to-get-gphoto2-working)? This should avoid mounting the SD card of the camera.

Aber jetzt kann ich nicht auf dem Stick schreiben. Aber das liegt glaube ich am Format NTFS. Ich probiere das diese Woche nochmal mit einen anderen Stick aus.

reuterbal, 2019-01-17:
2. Can you give an example of a "low quality" picture and include your `photobooth.log`?

3. NTFS can sometimes cause trouble although I was under the impression that these days were finally over. You can try formatting the drive as FAT, then everything should work as expected.

MarioJanssen27, 2019-01-17:
2. I uploaded the files. A photo (Quality=95; 1,4 MB), a photo (With GPhoto2--capture-and-Download from the terminal) and a log file.

[image]
[image]
[photobooth.log](https://github.com/reuterbal/photobooth/files/2769608/photobooth.log)

3.  I'll work on it tomorrow.

--------------------------------------------------------------------------------------------------------------
In deutsch

2. Ich habe die Dateien hochgeladen. Ein Foto (Qualität=95; 1,4 MB), ein Foto (mit GPhoto2--Capture-and-Download vom Terminal) und eine Log-Datei.

3. Bearbeite ich morgen.

MarioJanssen27, 2019-01-18:
2. I've adjusted the resolution now. Nevertheless the jPG of Photobooth is smaller than that of ghpoto2. Is this only due to the JPG compression?

3. I could solve the problem with the USB stick. It was because of the NTFS format with FAT32 there are no problems.

----------------------------------------------------------------------
in deutsch

2. Ich habe die Auflösung jetzt angepasst. Trotzdem ist die jPG von Photobooth kleiner als von ghpoto2. Liegt das jetzt nur an der JPG Kompression?

3.  Das Problem mit dem USB-Stick konnte ich klären. Es lag an dem NTFS Format mit FAT32 gibt es keine Probleme.

reuterbal, 2019-01-22:
2. Possibly. 
3. Great.

reuterbal, 2019-01-25:
Closing this as all issues seem to be resolved.

closed 2019-01-25 by reuterbal