reuterbal/photobooth #90: Editing language file has no effect

mfausin, 2019-01-15:
Hello,

first of all I want to say its a very great work, thanks a lot! I was searching really long to find such a great solution!

Installing was very simple, i can take photos and print them. 
Now I want to edit the text, i.e. "Drück den Knopf!". I tried to edit the language file .../photobooth/locale/de/LC_MESSAGES/photobooth.po but it has no effect wenn I restart the Photo Booth, it always show me the original German text.

What else I have to edit? Or did I something wrong?

reuterbal, 2019-01-16:
You will have to compile .mo-files from the .po-files.
For that I am using Babel and have included the respective commands into setuptools. Read the [Babel documentation](http://babel.pocoo.org/en/latest/setup.html) on how to proceed.

mfausin, 2019-01-19:
Thanks a lot!

it took some time to understand how it works, but now it works fine! 

Just a hint:
I found a "string" which is not in the .pot or .po files. I edit the Frames.py and extract a new .pot-file
(Babel is a very great tool ;) )

`#: photobooth/gui/Qt5Gui/Frames.py:151
msgid "Taking a photo..."
msgstr ""`

But I just cannot find a messageID for the "Print"-Button at the PostprocessMessage. There ist no msgid in the .po-file and I have no idea where I can find it in the GUI-code.
Can you please tell me where I found it?

Its not a big thing. I think everybody understand the English word "Print", but then it would be complete.

reuterbal, 2019-01-22:
Yes, you are absolutely right. The file affected is [`photobooth/gui/GuiPostprocessor.py`](https://github.com/reuterbal/photobooth/blob/master/photobooth/gui/GuiPostprocessor.py).