reuterbal/photobooth #91: Raspberry PI 3 with RaspiCam > Not smooth during preview

oelegeirnaert, 2019-01-16:
Hello,
**Thank you for creating such a great project!**

I'd like to know if other people are also having some problem with the raspberry pi 3 webcam.

I have the following one:
[https://www.sossolutions.nl/raspberry-camera-module-v2](https://www.sossolutions.nl/raspberry-camera-module-v2)

During preview, the stream is not smooth, how to resolve this?
Has it something to do with the FPS?

reuterbal, 2019-01-16:
I have only tested with a different camera module (with lower resolution). There everything worked fine.
Maybe try to reduce the resolution and see, if that helps.

oelegeirnaert, 2019-01-22:
Mmmm, i found the issue...
It's related to the IO stream in CameraPicamera.py > def getPreview(self):
self._cap.capture(stream, format='jpeg'... > jpeg

I've changed it to:
self.setActive()
return self._cap.start_preview()

But then I have an error when taking picture:
Error: 'PiPreviewRenderer' object has no attribute transpose.

reuterbal, 2019-01-22:
You probably have to call `stop_preview()` or something like that before being able capture pictures.

reuterbal, 2019-05-11:
Did using a different Raspbian edition fix the issue?

oelegeirnaert, 2019-05-21:
No, but you can close this ticket. As I'm using a Canon D4000 right now.
https://www.coolblue.be/nl/product/807173/canon-eos-4000d-18-55mm-dc.html

closed 2019-05-21 by oelegeirnaert