reuterbal/photobooth #92: Additional menu for Photobooth tasks/ game

MarioJanssen27, 2019-01-20:
Additional menu for Photobooth tasks/ game

Additional menu for Photobooth tasks/game to at game like this game. (https://www.amazon.de/Photobooth-Tube-Fotoaufgaben-Hochzeitsspiel-Spielekarten/dp/B07DP58NQS/ref=sr_1_4/261-8987724-2598531?s=toys&ie=UTF8&qid=1547986315&sr=1-4&keywords=Fotobox)

Where and how can I best add an additional menu? In my opinion it would be best to add it in the same screen where it says "Press the button". In which file would you have to add it?

The menu should look like this: a picture shows where the item is located. In addition there are two keys. The first button would be a release button for the photos (which is already present). Like a button (and Gpio) for a random playback of the tasks/games. I mean if the displayed task/game doesn't like another task/game in random mode. 

Where and how can I do this?

Technically that shouldn't be a problem, should it?

--------------------------------------------------------------------------------------
in deutsch/ in German

Zusatzmenü für Photobooth Aufgaben/ Spiel

Zusatzmenü für Photobooth Aufgaben/ Spiel zum bei Spiel wie dieses Spiel. 
[((https://www.amazon.de/Photobooth-Tube-Fotoaufgaben-Hochzeitsspiel-Spielekarten/dp/B07DP58NQS/ref=sr_1_4/261-8987724-2598531?s=toys&ie=UTF8&qid=1547986315&sr=1-4&keywords=Fotobox)](url)

Wo und wie kann man am besten ein zusatzmenu hinzufügen? Meiner Meinung wäre es am besten im gleichen Bildschirm wo steht "Drück den Knopf". In welche Datei müsste man das Einbauen?

Das Menü sollte so aus sehen das ein Anzeige in Form eines Bilds anzeigt wo die Aufagbe steht. Dazu kommt zwei Tasten. Die erste Taste wäre ein Auslöse Knopf für die Fotos (der schon vorhanden ist. So wie eine Schaltfläche (und Gpio) für  eine zufalls Wiedergabe der Aufgaben/ Spiele. Ich meine damit wenn die angezeigte Aufgabe/ Spiel nicht gefällt per zufallsmodus eine andere Aufgabe/ spiel bekommt. 

Wo und wie kann man das am besten einbauen?

Technisch dürfte das kein Problem sein oder?

reuterbal, 2019-01-22:
This requires a number of changes because it means to add a new state in the global state chart and then create handlers for this state in the GUI and GPIO drivers. Files affected are probably:
- `StateMachine.py`
- `gpio/__init__.py`
- `gui/Qt5Gui/PyQt5Gui.py`
- `gui/Qt5Gui/Frames.py`

MarioJanssen27, 2019-02-06:
Intermediate result:

I programmed a random generator for Jpg/Jpeg files. I am currently programming the menu.

------------------------------------------------------------------------------
in detusch

Zwischenergebnis:

Ich habe einen Zufallsgenerator für Jpg/Jpeg Dateien programmiert. Ich bin gerade am programmieren Menu.

reuterbal, 2019-03-11:
In case it helps. I finally found the time to draw a state chart: https://github.com/reuterbal/photobooth/blob/master/supplementals/state-chart/state-chart.pdf