reuterbal/photobooth #94: Photo printing seems to go awry between Photobooth and CUPS

smarquis, 2019-01-24:
Hi Reuterbal!

First off, thank you so much for making this whole repository available to the public! Ithas been a big help for me to try and figure out a photobooth for my upcoming wedding.

Now - on to the issue!

I was able to find my specific printer's ppd file in CUPS (HP Officejet 6600) And have been able to print test pages and regular pdf pages flawlessly from PDF viewers and image vieweres with no issue. However, the issue i'm having is that the print alignment seems to go hay wire within photobooth, as in the first attached picture. My initial photobooth settings are width: 152mm, Length: 102mm, and it results in a print that is off set to the right of the page. However, when I flip the dimensions in the photobooth software (i.e. Width: 102mm, Length: 152mm) It results in a properly centered picture, but put in the wrong orientation (as in the 2nd picture).

I've deducted that it can't be CUPS or the printer itself that's throwing the alignment off since I can print from a PDF viewer no problem using default settings and the prints come out fine.

What are my options here? I've narrowed it down to the command being sent from photobooth to CUPS since I am able to print fine by default out of any other application under default settings. Is there something i'm missing here? WHat is the specific command that gets sent to CUPS from the photobooth app?

Sorry if this comes off as vague, but it would be nice to start a dialogue to figure this out as i've tried searching for other solutions but can't seem to fix this.

Thanks for any help!!

smarquis, 2019-01-24:
It would help if I attached the photos, sorry about that:

Image 1: Mis-aligned Print
[image]

Image 2:Flipped but centered photo:
[image]

reuterbal, 2019-01-24:
Hi @smarquis 
Printing is happening [here](https://github.com/reuterbal/photobooth/blob/master/photobooth/printer/PrinterPyQt5.py)

In ll. 56 and 60 it should probably read `paperRect` both times, not `pageRect` and `paperRect`. Try setting them to the same value. See the [Qt documentation](http://doc.qt.io/qt-5/qprinter.html#pageRect) for the difference.
You can try additionally to specify `setFullPage` (see [here](http://doc.qt.io/qt-5/qprinter.html#setFullPage)) before that.
Please let me know if that fixed the issue.

smarquis, 2019-01-25:
Hi Reuterbal,

Thanks for getting back to me so quickly. Unfortunately, setting the setfullpage variable doesn't seem to work, and neither does replacing pagerect with paperrect. 

However, i've made a few observations:

I messed around with the origin variable,and when I change the origin value to manually inputted numbers,the image seems to move, but the print area stays put. Most obvious is the transition from setting origin=(0,-2000) to origin=(1000,000). The image clearly moves, but it's being cut off by some unknown invisible line that stays the same on all prints:
[image]

Another obeservation i've found is that when I go through the photobooth GUI and tell it to print to PDF, it prints perfectly within the margins, and when I print that PDF to a physical printer, it prints fine without having to mess around with the print settings at all.

Is there any more options I could explore? I'm sure it's not the printer driver because I can print easily to the physical printer. it's only when photobooth prints to the printer that I get the problem.

What do you think? 

Thanks a bunch!

reuterbal, 2019-01-25:
Hi @smarquis 
maybe dig a bit further into the Qt5 Printing support and see if there is something that might be related to that.

I intentionally kept the printing module as simple as possible because printing is always a pain in the a** (welcome to 21st century... we can send messages around the world in less than a second but putting that message on a sheet of paper right next to me is rocket science).
That's why there is only the bare minimum of necessary logic implemented: determining size and offset. 

You should also check the default values in CUPS for that printer - most other applications specify implicitly a bunch of settings but the photobooth relies on system defaults for that.

How did you print the PDF? Best would be to simply run `lp filename.pdf`, which should be as close as possible to what the photobooth does.

reuterbal, 2019-01-25:
#69 is probably related

smarquis, 2019-01-26:
Hi Reuterbal,

It looks like I was able to fix it, but it's more of a brute force method than actually coming up with a robust way to print. Instead of working through the code of photobooth, it got me thinking - couldn't I just mess with the CUPS ppd file and manually set an origin?

So I did exactly that. I directed my terminal to /etc/cups/ppd and text edited the driver ppd file default paper size. 

Specifically, the line originally looked like this:
*ImageableArea Photo4x6/Photo 4x6in: " 9.3599999 9.3599999 278.64 422.64 "

And I changed it to:
*ImageableArea Photo4x6/Photo 4x6in: " -172 -60 278.64 422.64 "

Effectively moving the imageable area 172 units to the left and 60 units to the top of the page.

I had to print through about 50 pages of expensive image paper to get to that value through trial and error, and that value is specific only to my setup, so this is by no means a robust solution. If I wanted to change the paper size, or pixel density through the photobooth GUI, the print would once again be out of whack and I would have to go through another 50 pages of paper to get the settings correct, so I do not recommend doing this whatsoever.

However, this works for me since I don't plan on changing the camera or paper size in the foreseeable future.

So I guess it's kind of solved? I don't know, what do you think Reuterbal? :p

reuterbal, 2019-01-28:
Wow, that is an expensive solution. Honestly, it sounds like the PPD-file has some issues... You could also try to look online for another one. Often there are more than one to be found for a printer.

As you have a working solution and the problem seems to be specific to your printer, I'm closing this for now.

closed 2019-01-28 by reuterbal