reuterbal/photobooth #97: Success story: Photo Booth #2

mfausin, 2019-01-28:
**Describe your photobooth**
This is my second Photo Booth, the first one was even more a prototype for my dad's birthday. However, I have received many inquiries there if I would lend my Photo Booth. But I wanted to build a one that really works.

**Hardware (please complete the following information):**
 - Raspberry Pi 3B+
 - Canon EOS 450D
 - GPIO: Yes
 
 **Additional equipment**

Picture | Name | Price
-- | -- | --
 [image] | Canon EOS 450D SLR Digital Camera Lens Kit 18-55mm --> Ebay | 153,50 €
[image] | Neewer Z Flex Tilt Kugelkopf  --> Amazon | 18,99 €
[image] | Pixel TF-326 Blitzschuh Adapter für Studioblitze mit PC-Sync Buchse für Canon --> Amazon | 11,90 €
[image] | Walimex Pro Newcomer 100 Studioblitz --> Amazon | 90,50 €
 [image] | Walimex Pro Beauty Dish VC Serie (40 cm) --> Amazon | 56,35 €
[image] | kj-vertrieb Netzteil mit Akkuadapter für Canon EOS 450D --> Amazon | 18,95 €
[image] | König & Meyer 19580-000-55 Boxenflansch schwarz --> Amazon | 19,80 €
[image] | Raspberry Pi 3 Model B+ (plus) BCM2837B0 SoC, IoT, PoE Enabled Single Board Computer --> Eckstein | 39,95 €
[image] | 3D printed case for Raspberry Pi 3+ --> thingiverse.com | --
 -- | Raspberry Pi Kühler Cooler Kühlkörper | 2,95 €
[image] | Noctua NF-A4x10 5V Lüfter, 5V (40mm, Braun) | 12,90 €
[image] | Waveshare 10.1 inch Raspberry Pi Display 1280x800 capacitive Touchscreen HDMI LCD with case IPS | 98,99 €
[image] | Fußtaster PFS-135A, 10 A/250 V~ --> Pollin | 7,95 €
[image] | CANON SELPHY CP 1300, Fotodrucker --> Saturn | 123,99 €
-- | Cables, wood for the housing,......... |

**Software (please complete the following information):**
 - OS: Raspbian Stretch
 - Python version: 3.6.x

**Modifications:**
 - Stylesheet adapted to the larger display with 1280x800 pixels
 - Location of the finished pictures and the individual shots were separated.
 - Web server with a simple image gallery installed to allow guests to access the images. For this I used the gallery SFPG (Single File PHP Gallery). https://sye.dk/sfpg/
- To access the images, I wanted to make the RasPi a WLAN access point. Unfortunately that has led to problems .....

**Problems:**
This solution for a photobooth works really well and is really well built! Thanks again to reuterbal !!!
During the installation and the first attempts, I hardly had any problems.
I only had problems when I tried to make the RasPi an access point, but unfortunately something went wrong, so he got stuck while booting. I have not found the reason yet, but that's why I have to take care of it later.

Otherwise, I had only 2-3 times a problem that the camera was apparently not responsive. The result was that the countdown just stopped, it only helped to restart, because the program did not react at all.

**Pictures:**
[image]

[image]

[image]

[image]

MarioJanssen27, 2019-01-28:
Hello,

Is this a standalone network? Where can I download the photo from my smartphone? If so, do you have a manual?

Web server with a simple image gallery installed to allow guests to access the images. For this I used the gallery SFPG (Single File PHP Gallery). https://sye.dk/sfpg/
To access the images, I wanted to make the RasPi a WLAN access point. Unfortunately that has led to problems .....

reuterbal, 2019-01-28:
This looks really great, thank you for sharing!
Here's a tutorial on how to create an access point: https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md

MarioJanssen27, 2019-01-28:
> 
> 
> This looks really great, thank you for sharing!
> Here's a tutorial on how to create an access point: https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md

I did that a week ago and it worked for me too. But I can't do it with Gallery!?

Is there a manual or project that describes everything completely?

mfausin, 2019-01-29:
Yes, it was planned to set up a standalone network. But I had problems as I said, I'll try again with the instructions of reuterbal!

As Gallery I use the SFPG (Single File PHP Gallery) as written above. It is a gallery without database. As the name implies, the whole gallery actually consists of a single PHP file.

The "installing" and setting up the SFPG is really simple. You only need a web server (Apache2, PHP, PHP-GD, ....), all requirements of the SFPG can be found on the website of the developer. For the webserver itself you can find enough instructions on the internet.
All further steps to set up the SPFG can be found in the readme.txt!

What the guests have to do then, is to connect to the WLAN of the RasPi and enter in a browser the IP address or host name.

oelegeirnaert, 2019-01-30:
I'll try to write an additional process that can be enabled from the settings menu.
(https://github.com/reuterbal/photobooth/blob/master/photobooth/main.py)
It will implement SimpleHTTPServer.
Once this server is running:

You'll be able to enable:

1. A slideshow that could be presented during the event making use of .
2. A QR-code that will pop-up after the guests tooks their picture to download their picture.

reuterbal, 2019-01-30:
This would be awesome, I would be very much interested in such an addition! There is another issue for that (#75) where the discussion for that should be happening.

reuterbal, 2019-01-31:
Yes, but let's move the discussion to #75

MarioJanssen27, 2019-01-31:
Okay to #75

gpsless, 2019-02-03:
@mfausin I did not take a look at your code yet, but I assume that the liveview comes from the DSLR via gphoto2, and that via gphoto2 you also manage to shoot and store the picture once the trigger is hit.
From a CPU load and thermal stress perspective, what stats have you seen on the RPi?
In other words, is the liveview process pushing the RPi CPU to its limit, that if cooling is not properly taken care of, the whole system is at risk of crashing because of overheating?

mfausin, 2019-02-04:
Hallo @gpsless ,
its not my code, its from reuterbal!
But yes, gphoto2 is used to get the live view from the DSLR and to take and store the photo.

I only watched the CPU load while I was testing with the camera dummy, so without gphoto. There it comes up to 70-80%. Only for a few seconds. The temperature didn't get over 50°C. 
I think it wouldn't be really much stress for the RPi. 

By the way, last Saturday my Photo Booth were on a birthday party. 229 photos were taken, without any problems!

reuterbal, 2019-02-04:
@gpsless In my experience, the real issue is the poor USB performance of the RPI. It limits the frame rate in live preview and thus prevents the RPi from overheating ;-)
Besides, the processor of the RPi usually reduces clock rate early enough to prevent it from crashing due to overheating. Generally, I recommend using a passive heatspreader and use an open case for the RPi so that temperature doesn't rise so fast. This will simply improve performance.

gpsless, 2019-02-04:
Hallo @mfausin 
Thank you for your feedback! Good to hear that your photobooth worked reliably 👍 

@reuterbal I see your point. How much would you estimate the framerate of liveview over USB?

alopezbu, 2019-05-31:
Hi @mfausin I would like to know if you are controling the light with the RPi or it is always on.
Btw it looks awesome ;D

mfausin, 2019-05-31:
@alopezbu 
The light is a studio flash, it’s controlled by the camera. 
But it has a built in „modelling light“, so even in dark conditions it is possible for the camera to focus.