reuterbal/photobooth #98: Print to Bluetooth device / MAC Address

dfordclist, 2019-01-30:
**Describe the bug**

Not really a bug, the project is going really well so far and I love your photobooth. I've got an HP Sprocket Plus which will print anything pushed to it over Bluetooth (unfortunately not supported by CUPs/Gutenberg - as far as I can tell, if anyone has suggestions on this I'd prefer it to messing with the source code) - I've looked through the code but I'm very new to python and not 100% sure where to find the Printer handling settings.

What I'd like to do is change the default printer to the MAC address of the Bluetooth printer. I can manually push the photos on terminal using obexftp so hoping with a bit of tweaking I can get the JPG to print to the MAC address rather than a default printer.

Would this be possible and could you direct me to where in the program I could find the relevant code?

Thanks!

**To Reproduce**

Steps to reproduce the behavior:
1. Try to print image file to Bluetooth

**Hardware (please complete the following information)**

 - Device [Raspberry Pi 3B+,]
 - Camera [PiCamera v2]
 - GPIO: [No]

**Software (please complete the following information)**

 - OS [Raspbian Stretch]
 - Python version [3.5.3]
 
 **Installed packages**
 
 astroid==1.4.9
automationhat==0.1.0
blinker==1.3
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
cffi==1.11.5
chardet==2.3.0
click==6.6
colorama==0.3.7
cryptography==1.7.1
docutils==0.13.1
drumhat==0.1.0
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==0.12.1
fourletterphat==0.1.0
gphoto2==1.9.0
gphoto2-cffi==0.4.2
gpiozero==1.4.1
idna==2.2
isort==4.2.5
itsdangerous==0.24
jedi==0.10.2
Jinja2==2.8
keyring==10.1
keyrings.alt==1.3
lazy-object-proxy==1.2.2
MarkupSafe==0.23
mcpi==0.1.1
microdotphat==0.2.1
mote==0.0.4
motephat==0.0.2
mypy==0.470
numpy==1.12.1
oauthlib==2.0.1
opencv-python==3.4.4.19
pantilthat==0.0.7
pgzero==1.2
phatbeat==0.1.1
-e git+https://github.com/reuterbal/photobooth.git@1b152c6fd4bdadecbb898f2ad6f0fa155d2098cd#egg=photobooth
pianohat==0.1.0
picamera==1.13
picraft==1.0
piglow==1.2.4
pigpio==1.38
Pillow==4.0.0
pkg-resources==0.0.0
pyasn1==0.1.9
pycparser==2.19
pycrypto==2.6.1
pygame==1.9.3
Pygments==2.2.0
pygobject==3.22.0
pyinotify==0.9.6
PyJWT==1.4.2
pylint==1.6.5
pyOpenSSL==16.2.0
pyperclip==1.5.27
pyserial==3.2.1
python-apt==1.1.0b5
pyxdg==0.25
rainbowhat==0.1.0
requests==2.12.4
requests-oauthlib==0.7.0
roman==2.0.0
RPi.GPIO==0.6.5
RTIMULib==7.2.1
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
sense-emu==1.1
sense-hat==2.2.0
simplejson==3.10.0
six==1.10.0
skywriter==0.0.7
sn3218==1.2.7
spidev==3.3
ssh-import-id==5.6
thonny==3.0.8
thonny-pi==1.1
touchphat==0.0.1
twython==3.4.0
typed-ast==0.6.3
unicornhathd==0.0.4
urllib3==1.19.1
virtualenv==15.1.0
Werkzeug==0.11.15
wrapt==1.9.0

reuterbal, 2019-01-30:
Printing is implemented in [PrinterPyQt5.py](https://github.com/reuterbal/photobooth/blob/master/photobooth/printer/PrinterPyQt5.py), however, it simply passes the picture to the standard printer configured in the operating system. Thus, it is more a question of how to configure your bluetooth printer as the default printer in the operating system.

Usually, this is done in CUPS (see installation instructions of the photobooth), but might be different in your case. Basically, once you can print with the command `lp <filename>` it should also work with the photobooth.

dfordclist, 2019-01-31:
Thanks for the feedback, I'll get into a bit more investigation.

closed 2019-02-23 by dfordclist