reuterbal/photobooth #100: Can't install Photobooth

jul1792, 2019-02-03:
Hi everyone,

I already had the photobooth up and running, but I had some issues so I wanted to do a fresh install from scratch, but I ran into some problems. 

I installed the latest raspbian stretch and configured Gutenprint with my Canon Selphy CP800. Printing the test page worked fine. 
Then I went on to install the rest accordingly to the installation instruction. But something didn't work. 

First of all this command didn't work `sudo apt install qt5-default pyqt5-dev pyqt5-dev-tools # for PyQt5-GUI`

I got this messages in the terminal:
`sudo apt install qt5-default pyqt5-dev pyqt5-dev-tools # for PyQt5-GUI
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 qt5-default : Hängt ab von: qtbase5-dev soll aber nicht installiert werden
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.
`
then I installed pyqt5 with this `sudo apt-get install python3-pyqt5` which worked fine.
I don't need gphoto2 because I want to use the picamera. 

So I cloned the photobooth and did everything accordingly to the instructions. But something went wrong when I want to install the photobooth. This is the output:

`pi@raspberrypi:~ $ sudo git clone https://github.com/reuterbal/photobooth.git
Klone nach 'photobooth' ...
remote: Enumerating objects: 34, done.
remote: Counting objects: 100% (34/34), done.
remote: Compressing objects: 100% (27/27), done.
remote: Total 1757 (delta 17), reused 18 (delta 7), pack-reused 1723
Empfange Objekte: 100% (1757/1757), 6.10 MiB | 469.00 KiB/s, Fertig.
Löse Unterschiede auf: 100% (1121/1121), Fertig.
pi@raspberrypi:~ $ cd photobooth
pi@raspberrypi:~/photobooth $ virtualenv -p python3 --system-site-packages .venvAlready using interpreter /usr/bin/python3
Using base prefix '/usr'
Traceback (most recent call last):
  File "/usr/bin/virtualenv", line 11, in <module>
    load_entry_point('virtualenv==15.1.0', 'console_scripts', 'virtualenv')()
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 724, in main
    symlink=options.symlink)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 946, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 1157, in install_python
    mkdir(lib_dir)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 329, in mkdir
    os.makedirs(path)
  File "/usr/lib/python3.5/os.py", line 231, in makedirs
    makedirs(head, mode, exist_ok)
  File "/usr/lib/python3.5/os.py", line 231, in makedirs
    makedirs(head, mode, exist_ok)
  File "/usr/lib/python3.5/os.py", line 241, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: '/home/pi/photobooth/.venv'
pi@raspberrypi:~/photobooth $ sudo virtualenv -p python3 --system-site-packages .venv
Already using interpreter /usr/bin/python3
Using base prefix '/usr'
New python executable in /home/pi/photobooth/.venv/bin/python3
Also creating executable in /home/pi/photobooth/.venv/bin/python
Installing setuptools, pkg_resources, pip, wheel...done.
pi@raspberrypi:~/photobooth $ source .venv/bin/activate
(.venv) pi@raspberrypi:~/photobooth $ pip install -e .[picamera]
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/photobooth
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/home/pi/photobooth/setup.py", line 257, in <module>
        'build': CustomBuild,
      File "/home/pi/photobooth/.venv/lib/python3.5/site-packages/setuptools/__init__.py", line 144, in setup
        _install_setup_requires(attrs)
      File "/home/pi/photobooth/.venv/lib/python3.5/site-packages/setuptools/__init__.py", line 139, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/home/pi/photobooth/.venv/lib/python3.5/site-packages/setuptools/dist.py", line 724, in fetch_build_eggs
        replace_conflicting=True,
      File "/home/pi/photobooth/.venv/lib/python3.5/site-packages/pkg_resources/__init__.py", line 846, in resolve
        dist = best[req.key] = env.best_match(req, ws, installer)
      File "/home/pi/photobooth/.venv/lib/python3.5/site-packages/pkg_resources/__init__.py", line 1118, in best_match
        return self.obtain(req, installer)
      File "/home/pi/photobooth/.venv/lib/python3.5/site-packages/pkg_resources/__init__.py", line 1130, in obtain
        return installer(requirement)
      File "/home/pi/photobooth/.venv/lib/python3.5/site-packages/setuptools/dist.py", line 783, in fetch_build_egg
        install_dir = self.get_egg_cache_dir()
      File "/home/pi/photobooth/.venv/lib/python3.5/site-packages/setuptools/dist.py", line 752, in get_egg_cache_dir
        os.mkdir(egg_cache_dir)
    PermissionError: [Errno 13] Permission denied: './.eggs'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /home/pi/photobooth/
(.venv) pi@raspberrypi:~/photobooth $ sudo pip install -e .[picamera]
Obtaining file:///home/pi/photobooth
photobooth requires Python '>=3.5' but the running Python is 2.7.13
`
It is really frustrating, so can anyone help me here please? That would be great!

Thanks a lot in advance!

reuterbal, 2019-02-04:
Well, you really need to learn some basics about Linux. Read some of the vast number of tutorials available online. For example, pay attention to file access, user management, sudo, ...

1. Try running `sudo apt install qt5-default pyqt5-dev pyqt5-dev-tools qtbase5-dev` to install the missing packages.
2. DO NOT USE SUDO when installing the photobooth.  You cloned the photbooth as sudo, thus making root the owner of the photobooth folder. Therefore, creating the virtualenv as user `pi` did not work due to missing rights. Follow the instructions meticulously if you don't know what you're doing.

reuterbal, 2019-05-11:
Closing this due to inactivity

closed 2019-05-11 by reuterbal