reuterbal/photobooth #101: when the camera takes a picture, each picture is displayed separately, right?

MarioJanssen27, 2019-02-05:
Hello,

when the camera takes a picture, each picture is displayed separately, right?
and finally the collage.

Or do I get the wrong idea?

in deutsch
----------------------------------------------------------------------------------------------
Hallo,

wenn die Kamera ein Foto macht wird doch jedes Foto einzeln angezeigt oder?
und zum Schluss die Collage.

Oder versehe ich mich da.

reuterbal, 2019-02-06:
Hi,
the photobooth shows the preview (if enabled) during countdown, takes a picture (without displaying it) and assembles the collage at the end, which is then shown to the user.

reuterbal, 2019-02-06:
What does the log say?

MarioJanssen27, 2019-02-06:
I misspoke. 

The Photobooth starts super with the command: sudo ./autostart.sh

But at autostart. When starting the Raspberry pi I can't start it directly. The screen remains black..... . But why?

___________________________________________________________________
in deutsch

Ich habe mich falsch ausgedrückt. 

Die Photobooth startet super mit dem Befehl: sudo ./autostart.sh

Aber beim Autostart. Beim Start der Raspberry pi kann ich das nicht direkt starten. Der Bildschirm bleibt schwarz..... . Aber warum?

reuterbal, 2019-02-06:
Why do you run it as sudo? There is no need for that.
This is certainly the reason why it does not autostart, as it will probably have problems with access rights.

MarioJanssen27, 2019-02-06:
What other command can I use to start this?
With Raspberry I am a beginner.....

in deutsch
_____________________________________________

Mit welchen Befehl kann ich das denn sonst starten?
Mit Raspberry bin ich Anfänger.....

reuterbal, 2019-02-06:
Please read this: https://www.digitalocean.com/community/tutorials/an-introduction-to-linux-permissions

MarioJanssen27, 2019-02-06:
Okay, take a look at this. Thank you.

My goal: the Rasberrybox start sofrt the Photobooth.

Funny it worked otherwise also.... I don't know what I would have changed.

_____________________________________________________________________
in deutsch

Okay schaue mir das an. Danke.

Mein Ziel: die Rasberrybox start sofrt die Photobooth.

Komisch es hat sonst auch funktioniert.... ich weiß nicht was ich verändert hätte.

MarioJanssen27, 2019-02-06:
Okay, I get that, but what does that have to do with the automatic start?

--------------------------------------------------------------------------
in deutsch

Okay das habe ich verstanden aber was hat das mit dem automatischen Start zu tun?

reuterbal, 2019-02-06:
If you have 
- cloned the source code as root
- created the virtualenv as root
you might lack permissions to run or even read the relevant files as the usual desktop user.
If you have run the photobooth as root before, you might be unable to write to the log file, causing the application to crush.

Try executing `chown -R pi /home/pi/photobooth` or wherever you have put your files.

MarioJanssen27, 2019-02-06:
Which settings did you use for the collage?

---------------------------------------------------
in deutsch

Welche Einstellungen habt ihr für die Collage verwendet?

MarioJanssen27, 2019-02-06:
That worked. The Photobooth starts directly.

But after the first photo comes now the following error message:

Error: [Errno 13] Permission
Denied: '/media/pi/
4006-594E/Photobooth
2019-02-06_shot_00001.jpg'

The Photobooth is supposed to store the photo on a USB stick. This works without autostart.

in deutsch
____________________________________________________

Das hat funktioniert. Die Photobooth startet direkt.

Aber nach dem ersten Foto kommt jetzt folgende Fehlermeldung:

Error: [Errno 13] Permission
Denied: '/media/pi/
4006-594E/Photobooth
2019-02-06_shot_00001.jpg'

Die Photobooth soll die Foto einen USB-Stick speichern. Das klappt ohne Autostart problemlos.

reuterbal, 2019-02-06:
See #93 for that

MarioJanssen27, 2019-02-11:
I don't understand what you mean!

My USB stick has the format FAT32. Saving on the USB stick only doesn't work when starting Photobooth directly.

What do I have to do ?
#93 I don't understand. Because this is for NTFS, isn't it?

----------------------------------------------------------------------------
in deutsch
----------------------------------------------------------------------------
Ich verstehe nicht was du meinst?!

Mein USB-Stick hat das Format FAT32. Das Speichern auf dem USB-Stick funktioniert nur beim direkten starten der Photobooth nicht.

Was muss ich machen ?
#93 Verstehe ich nicht. Weil das doch für NTFS ist oder?

reuterbal, 2019-02-12:
The problem has nothing to do with the file system. Simply replace `ntfs` by `vfat`.

MarioJanssen27, 2019-02-14:
The manual didn't work.  The Photobooth does not start anymore.  This shows it:

MarioJanssen27, 2019-02-17:
Hello,

the USB stick is integrated but cannot be written to. I am in despair.....

This error message shows the Photobooth:
[image]

MarioJanssen27, 2019-02-19:
Closed. 
The write command has fallen.

closed 2019-03-11 by reuterbal