reuterbal/photobooth #104: Success story: Photobooth Birthday

qwertyFrank, 2019-02-16:
**Describe your photobooth**
Photobooth done for some birthday parties, both kids and parents. I would like to thanks [reuterbal] for his great effort and outstanding Photobooth project. I realy appreciate his effort to make such project happen.

**Hardware**
- Raspberry Pi 3B+
- Raspberry Camera V2
- Canon Selphy CP1300
 - GPIO: Yes
 
 **Additional equipment**
- Waveshare 10.1 inch HDMI Raspberry Pi Display 1024x600 Touchscreen LCD WS40023
- Paulmann FlexLED 3D 78964 LED 12V 150cm RGB
- Jumbo Arcade Button 98mm
- wood case
- components for GPIO
- some 3D printed parts

**Software (please complete the following information):**
- Raspbian

**Modifications:**
- as it is / but mirrored arrow for pressing the button ;-)

**Problems:**
- no issues

**Pictures:**
- 
[image]
[image]
[image]
[image]

oelegeirnaert, 2019-02-17:
Hi @qwertyFrank ,

I saw you used the RaspiCam V2 in your project, I also have one but i'm experiencing some problems... During countdown, the live preview isn't that smooth. Do you have the same problem? How do you have solved this?

Kind Regards,
Oele Geirnaert

qwertyFrank, 2019-02-18:
Hi Oele,
No. I do not have such issues. But I am also using a "Raspberry Pi 3B+" which is very performant.
Regards,
Frank

oelegeirnaert, 2019-02-18:
Hello @qwertyFrank ,
Do you have installed any additional drivers?
Did you just followed the instructions as Reuterbal wrote in his ReadMe?
Thanks for your reply btw!

Kr,
Oele

qwertyFrank, 2019-02-18:
Hi Oele,
I used 
[image]
without any additional driver and followed Reuterbal's ReadMe.
Regards,
Frank

oelegeirnaert, 2019-02-19:
Aaaaah, If I remember I've taken the Raspbian 'Stretch with desktop' only, so without the 'recommended software'.

I'll give it a try later the week and I'll keep the community posted because it's also related to an issue I've previously logged (https://github.com/reuterbal/photobooth/issues/91)!

Thanks for replying!

reuterbal, 2019-03-11:
Thanks for sharing your success story! The differences in the Raspbian editions seem to have changed recently - we should update the installation instructions accordingly. Thanks for pointing that out!