reuterbal/photobooth #106: Key/Mouse presses instead of GPIO

octahate, 2019-03-06:
Hello everyone!
First of all thank you for this piece of software, I haven't had such amount of success in such a short period of time! I am planning to use an old laptop instead of a pi for my photobooth, and checking every issue/question here I still couldn't find my answer to my question:
Is there any way to config the GPIO button presses to regual keyboard or much better mouse clicks? (left-rigt click?)
Thank you for the answer!

mfausin, 2019-03-06:
Hello octahate,

You don’t have to use GPIO, the UI has buttons you can click with your mouse or you can take a picture by pressing the space bar. 
There is no need to config something.

octahate, 2019-03-08:
I think I need to clarify a bit:
how can I use the mouse's buttons as input (so I am not using it as a click like a mosue but more like a button on the keyboard like if I right click (but not on the dedicated button) it prints and if I use right it cancels it
also I can use space to print picture but I cannot use the same imput to take a pic, is there a way to reconfigure it?

reuterbal, 2019-03-11:
You will have to change / add additional buttons / click-sensitive areas in [photobooth/gui/Qt5Gui/Frames.py](https://github.com/reuterbal/photobooth/blob/master/photobooth/gui/Qt5Gui/Frames.py) to make that happen.