reuterbal/photobooth #107: issue with GPIO 23 and 24 for control of Photobooth with active Printing option

gpsless, 2019-03-11:
Hi all

I have installed the latest Raspbian Full package (2018-11-13), including sudo apt-get update etc., the latest gPhoto2 package (v.2.5.22 IIRC) and the latest Photobooth package.
It runs on an RPi 3B in conjunction with a Canon EOS 550D. Printing via USB / Canon Selphy 1300 also works flawlessly.

So, first off, @reuterbal - great job! I love the capabilities of your software and do appreciate very much your efforts in building and maintaining it.

Before I start with the description of my issue, please consider that the last time I created code was in 1999, in C... so I candidly admit that I do have a solid understanding of Linux, but I have no clue of Python.
I want to build a photobooth before coming Saturday, as on that day my better half will celebrate her 30th birthday, and this photobooth is one of my gifts to her :)

Enough foreplay now.
I activated the printing option and GPIO control (default settings, 23 = trigger, 24 = exit).
Pleass note that I have tested with a little Python script my cabling to the GPIO i/f. I am positive that the RPi gets LOW states from pins 16 and 18, which, as per my understanding, are GPIO 23 and 24.
I boot up the RPi, it autostarts phobooth, and when I set GPIO23 to LOW, the countdown starts, the picture is shot and displayed as expected.

Issue 1 and expected behaviour:
I get on screen the options "Print" and "Start Over" for selection. If I do GPIO23=LOW, nothing happens. If I do GPIO24=LOW, I get back to the welcome menu "Start Photobooth", "Set Date/Time", "Settings", "Quit".
I would expect GPIO23=LOW to trigger the "Start Over" command (because, if GPIO23 is controlled by a pedal switch, the average semi-drunk user, aka"ASDU", would associate pedal action with picture shooting IMHO), and GPIO24=LOW would trigger the "Print" command (because the ASDU would first have to walk to the photobooth, inspect the picture, and only then decide to use a precious piece of photo paper)

Issue 2 and expected behaviour:
Furthermore, in the welcome menu, neither GPIO23=LOW nor GPIO24=LOW do anything.
In this context, I would expect 23 to trigger the selected option, and 24 to switch to the next option (just like hitting Space and Tab respectively, when controlling via keyboard).

I would appreciate very much if anyone could help me in rapidly solving at least Issue 1 before Saturday!

Many thanks in advance!

reuterbal, 2019-03-12:
As I have already mentioned in the email you sent me, you have a misunderstanding of the purpose of the GPIO buttons.

Pin 23 (Trigger) is to be used as a trigger and as such only works when the photobooth is idle (to avoid triggering a new set of photos while something else is happening). You might consider enabling the trigger in the PostprocessState to return to idle - see the [state chart](https://github.com/reuterbal/photobooth/blob/master/supplementals/state-chart/state-chart.pdf) for that.

Pin 24 (Exit) is to be kept hidden - it is intended to give you (as the operator of the photobooth) a means to return to the menu. E.g., to shut down the photobooth cleanly or change settings.

If you want to have additional GPIO buttons, please refer to the existing issues: #44, #58 and others. Add the functionality yourself, I won't find time to do this in the next few months.

reuterbal, 2019-05-11:
Closing this due to inactivity.

closed 2019-05-11 by reuterbal