reuterbal/photobooth #108: OpenCV bug

octahate, 2019-03-12:
**Describe the bug**

Using OpenCV and multitudes of webcams only gives a black picture when taking photos , even in the collage .jpg they are black

I spent the last afteroon picking the code, and found a solution to make the camera pictures visible:

I commented out 

    def capturePicture(self, state):

        ## self.setIdle()
        picture = self._cap.getPicture()
        if self._rotation is not None:

in __init__.py and then the camera pictures started to appear BUT this resulted in a new bug:
After the 3rd restart of the photo taking cycle the preview didn't appear, and stays blank on every second cycle (it appears on the first photo, then on the second photo then doesn't then does then does't etc.)
I tried positioning self.setIdle() to somehwere else in the code or substitute it with self.setActive() but it didn't help either.

 - Device: Intel Atom Laptop
 - Camera: Microsoft, and and the built-in webcam
 - GPIO: NO
**Software (please complete the following information)**

 - OS: Debian
 - Python version latest
 
 **Installed packages**
 
Everything needed but gpio2

reuterbal, 2019-03-12:
Please include the list of installed packages - including version numbers. 
For example, I heavily doubt that you have the "latest" Python version on a Debian OS.
This means, please paste the output of `python3 --version` and `pip list`.

Please also include your photobooth.log

octahate, 2019-05-30:
Sorry for the late reply, just had to switch out PC, but the problem stayed the same.

To me it seems when I initiate the capture in the program, it re-starts the camera and the camera doesn't have enough time to fix the exposure since it takes it like 2 seconds to do it so. When I removed the put it in idle an restart it thing in my og post it seemed to solve this specific problem, but on the new pc commenting that line out makes the program unusable

Python version: 3.5.3
Installed packages:
```
Brlapi              0.6.5              
chardet             2.3.0              
colorzero           1.1                
cryptography        1.7.1              
cupshelpers         1.0                
gphoto2             2.0.0              
gpiozero            1.5.0              
httplib2            0.9.2              
idna                2.2                
keyring             10.1               
keyrings.alt        1.3                
louis               3.0.0              
Mako                1.0.6              
MarkupSafe          0.23               
numpy               1.16.4             
opencv-python       4.1.0.25           
pexpect             4.2.1              
photobooth          0.4.dev81+g471fb07 /home/emustation/photobooth
Pillow              4.0.0              
pip                 19.1.1             
pkg-resources       0.0.0              
ptyprocess          0.5.1              
pyasn1              0.1.9              
pycrypto            2.6.1              
pycups              1.9.73             
pycurl              7.43.0             
pygobject           3.22.0             
pysmbc              1.0.15.6           
python-apt          1.4.0b3            
python-debian       0.1.30             
python-debianbts    2.6.1              
pyxdg               0.25               
reportbug           7.1.7              
reportlab           3.3.0              
requests            2.12.4             
SecretStorage       2.3.1              
setuptools          41.0.1             
six                 1.10.0             
unattended-upgrades 0.1                
urllib3             1.19.1             
virtualenv          15.1.0             
wheel               0.33.4
```
and the log: 
```
2019-05-30 20:43:23,206 - root - INFO - Initializing photobooth
2019-05-30 20:43:23,207 - root - INFO - Photobooth version: 0.4.dev81+g471fb07
2019-05-30 20:43:23,208 - root - INFO - Reading config file "/home/emustation/photobooth/photobooth/defaults.cfg"
2019-05-30 20:43:23,210 - root - INFO - Reading config file "photobooth.cfg"
2019-05-30 20:43:23,240 - root - INFO - Number of last existing file: 5
2019-05-30 20:43:23,242 - root - INFO - Saving pictures as "2019-05-30/photoboothXXXXX.jpg"
2019-05-30 20:43:23,243 - root - INFO - Number of last existing file: 0
2019-05-30 20:43:23,243 - root - INFO - Saving pictures as "2019-05-30/photobooth_shot_XXXXX.jpg"
2019-05-30 20:43:23,244 - root - INFO - GPIO disabled
2019-05-30 20:43:24,748 - root - INFO - Using printer ""
2019-05-30 20:43:26,257 - root - INFO - Using OpenCV
2019-05-30 20:43:26,257 - root - INFO - Using camera with preview functionality
2019-05-30 20:44:09,507 - root - INFO - Saving picture as 2019-05-30/photobooth00006.jpg
2019-05-30 20:44:19,099 - root - INFO - Exiting photobooth with status code 0
```

And some screenshots:
[image]
[image]
[image]
[image]

reuterbal, 2019-05-31:
Yes, initializing the camera might take a second or two. 
And it might have something to do with exposure settings or similar - I don't know if the camera or opencv selects wrong values for that.
Since the camera does in fact deliver some picture, it seems to work fine from the photobooth side of things. However, the picture is very dark - is this due to the room being dark or is this actually wrong?
In the former case, I would try first to make the room as bright as possible to see if the final pictures are simply underexposed.
Else, I would disable the preview and see if that changes anything - just to get a step further.

octahate, 2019-05-31:
yes it was a very dark room around midnight, I just wanted to show the results.
I turned off the preview and it resulted that not even the camera led is turning on and still gives pitch black pictures, like it's not initialized yet not because of exposure settings (I had very very similiar issues with another photobooth program, but it was bulit for windows. I tried to dig up the bug report from my emails but sadly I couldn't find them, but it got fixed somehow.)
Cheese seems to react normal with my camera, I am attaching some extra pictures:
[image]
[image]

octahate, 2019-06-01:
OK so I toyed around with the code and the settings a bit more today, and I sort of found the solution, but I am still not sure about the origin of the error.

First of all I edited the camera python code again as I described it in the original post AND set the countdown to 8 instead of 4 seconds. Now the program is seemingly working, I get picture output but the thing is, after every third shot or so, the camera needs a bit of a more loading time than usual, resulting that you can see yourself only on the last 2 seconds of the 8 seconds of preview
[image]
[image]

I digged around at home and found another webcam, wanted to try out how it works.
The version without set.idle worked like a charm! Like how it's supposed to do its job.
[image]
At that point I thought it's probably just the other camera that has the error, so I added that code back, and now the other camera started misbehaving too: this gave a yellow picture:
[image]

I don't know right now what should be done to the code. As a temporary solution I am fine with waiting for the camera to load a bit more than usual, but I guess there has to be something with how the photobooth initializes cameras?

reuterbal, 2019-06-13:
I guess this has something to do with the OpenCV module. The photobooth uses only high-level simple functions from that library, maybe try working with the Python OpenCV module interactively and see if that works as expected.