reuterbal/photobooth #109: Exposure compensation/ Flash compensation

MarioJanssen27, 2019-03-16:
How to set exposure compensation, flash compensation, and exposure time.  My Photobox takes a very long time to take pictures and they are overexposed? 

Canon 1000d

In deutsch

Wie kann ich Belichtungskorrektur, Blitzkorrektur und Belichtungszeit einstellen. Meine Photobox braucht sehr lange um Bilder machen und sie sind überbelichtet?!?[](url)

serpentes80, 2019-03-17:
as the software only triggers the camera you have to set the correct values in your camera and/or your flash

MarioJanssen27, 2019-03-17:
I'm not looking for the camera settings for the flash. Shutterpeed...

In deutsch

Ich suche die Einstellungen für die Kamera. Nicht für den Blitz

closed 2019-03-17 by MarioJanssen27

reuterbal, 2019-03-17:
If you are using a flash, I recommend using manual mode and thus setting a fixed exposure time and aperture value. If you don't do that, your camera might select a long exposure time to account for low light situations (thus taking long to take the picture), which then results in overexposed pictures when a flash is triggered. As @serpentes80 already mentioned, these settings have to be selected in your camera - if your pictures are well-exposed in the lighting situation when triggered manually, they should also be fine when used with the photobooth app.