reuterbal/photobooth #110: I can't use photobooth

loicabme, 2019-03-17:
Photobooth doesn't work on my PI. 

Sorry for my english, I'm french and english isn't easy for me. I will try to describe my issue.
I'm a beginner with the PI. I have some notion dating from my studies.
I installed photobooth as explained in the INSTALL.md
I start Photobooth, I change parameters and after Photobooth is very slow. I can't use the software after "hit the buton". Photobooth remain blocked on this page or on page "get ready start countdown"
I don't understand why doesn't work.
(I don't use GPIO, I use touch screen, I need to touch the text directly to start the countdown ? it's right ?)

When I start Photobooth I have this message in the terminal "libegl warning dri2 failed to authenticate". Is that the source of the problem ?

I make a photobooth for a birthday, it's a gift. I need to finish the project before 30 april.

My PI seems very slow, it's the micro sd ?

Thank you for your help.

**Hardware**

 - Device: Raspberry Pi 3B+
 - Camera Canon EOS 1200D
 - GPIO: No
 - Printer: Canon 900
 - Display: Waveshare 10.1" 1280*800 touch screen
 - micro sd: transcend 16Gb

**Software**

 - OS Raspbian Stretch last version
 - Python version last version

wikijm, 2019-03-17:
Salut @loicabme, 

Pourrais-tu filmer le lancement de Photobooth grâce au logiciel https://asciinema.org/?

call-me-matt, 2019-03-17:
Hello loicabme,
two things that you could check: make sure that you are using a 2.5A power supply. try the latest raspberian image to install everything.
(mon setup était aussi lente avant que j'ai changé l'alimentation pour un bloc de 2.5A. Par contre ton problème me semble d'être plustôt software. Comment est-ce que tu as installé raspberian? T'as pris la dernière version où réutilisé un ancien Pi?)

loicabme, 2019-03-17:
Je regarde pour l'enregistrement du démarrage.

L'alimentation est une ankru 3A. Le PI est neuf et j'ai pris le dernier raspbien sur le site officiel (+ mise à jour après premier démarrage)

loicabme, 2019-03-17:
> Salut @loicabme,
> 
> Pourrais-tu filmer le lancement de Photobooth grâce au logiciel https://asciinema.org/?

Comment je partage l'enregistrement ? 
J'ai des nouveaux problèmes avec gphoto2.

wikijm, 2019-03-17:
Au besoin, je peux me rendre dispo un peu ce matin sur Skype ou tout autre moyen de communication.

loicabme, 2019-03-17:
> Au besoin, je peux me rendre dispo un peu ce matin sur Skype ou tout autre moyen de communication.

Merci beaucoup de la proposition mais je ne vais pas pouvoir ce matin, journée chargée.

loicabme, 2019-03-17:
Voici l'nregistrement, je ne sais pas si c'est la bonne méthode.

[tmp5t26flj_-ascii.zip](https://github.com/reuterbal/photobooth/files/2974914/tmp5t26flj_-ascii.zip)

wikijm, 2019-03-17:
Pas de soucis. Pour le partage de l'enregistrement asciineam : https://asciinema.org/docs/getting-started

loicabme, 2019-03-17:
merci de l'info, je vais voir pour recommencer.

wikijm, 2019-03-17:
La commande suivante (à jouer en tant que robot) devrait solutionner le problème :
```shell
sudo ln -s /usr/lib/arm-linux-gnueabihf/libGLESv2.so.2 /usr/lib/arm-linux-gnueabihf/libGLESv2.so
```
Source : 
https://raspberrypi.stackexchange.com/questions/61078/qt-applications-dont-work-due-to-libegl

wikijm, 2019-03-17:
Dans ton enregistrement, je vois également cette erreur :
```shell
inotify_add_watch("/home/pi/.config/qt5ct") failed: "No such file or directory"
```

loicabme, 2019-03-18:
Bonjour,

Je viens de refaire une installation complète, j'ai les même messages (je n'avais pas vu également pour inotify_add_watch("/home/pi/.config/qt5ct") failed: "No such file or directory")

Alors après test je peux maintenant faire les photos. Cela semble être fonctionnel sur ce point là.

J'ai essayé "sudo ln -s /usr/lib/arm-linux-gnueabihf/libGLESv2.so.2 /usr/lib/arm-linux-gnueabihf/libGLESv2.so" il me retourne : "ln: impossible de créer le lien symbolique '/usr/lib/arm-linux-gnuebinf/libGLESv2.so' : Aucun fichier ou dossier de ce type".

Pour la deuxième erreur il faut que je crée le dossier ?

Si je résume : 
- 2 erreurs (les précédentes)
- je peux maintenant faire les photos, l'assemblage jusqu'à la demande d'impression
- Les photo sont bien stockées
- J'ai testé le style dark, il semble fonctionner

loicabme, 2019-03-22:
Hello,

New information
1/ Now I can take pictures, assemble them, and print. 
I'm happy.
I can modify the stylesheet but something when I modify a parameters photobooth doesn't work. I have some errors and photobooth doesn't start with the terminal. It opens full of "worker" window.
I need to re-install photobooth.
This problem is known ?

2/ When I start Photobooth I have this message in the terminal "libegl warning dri2 failed to authenticate". Is that the source of the problem ?

I tried "sudo ln -s /usr/lib/arm-linux-gnueabihf/libGLESv2.so.2 /usr/lib/arm-linux-gnueabihf/libGLESv2.so" and it return me: "ln: impossible de créer le lien symbolique '/usr/lib/arm-linux-gnuebinf/libGLESv2.so' : Aucun fichier ou dossier de ce type".

3/ Now that I can print I would like to center the photos when printing.
The margins are differents :
- Left : 4,3 mm
- Right: 9,5 mm
- Top: 3,3 mm
- Bottom: 7,5 mm

What parameters I need to modify and in what file ?
(I tested print parameters but it's not good, the printer cut photos)

Thank you for your help.

loicabme, 2019-03-27:
3/ I can print the photos with similar margin (right, left, top, bottom) if I change the size of paper in section printer. I need to use 154x104 mm paper size and the real paper size is 148x100 mm.
Can you explain me why I need to do this change ?

reuterbal, 2019-05-11:
1/ should be fixed when you update Raspbian and all PIP packages to the latest versions.

2/ That is no problem, see #128 

3/ This is an open issue, I am also looking for a good solution.

reuterbal, 2019-06-27:
Closing this due to inactivity.

closed 2019-06-27 by reuterbal