reuterbal/photobooth #112: Is it possible to print 2 copies per page ?

kun93, 2019-03-21:
Is there anyway to modify cups or add something in the config to print 2 copies per page (2x6 x 2) ? I would like to get a photobooth style print on a 4x6 photo paper. Similar to the image on https://github.com/reuterbal/photobooth/issues/86

oelegeirnaert, 2019-03-27:
Hi @kun93 ,

Check the file: https://github.com/reuterbal/photobooth/blob/79c7ef1affab78b947fe3940f2d70dd646ccb49c/photobooth/camera/__init__.py#L173

On line 173 I've added some code to do the same.
This will take the original picture, and copy it.

kun93, 2019-03-27:
@oelegeirnaert thanks so much for this ! I’ll try it out tonight .

reuterbal, 2019-05-11:
Thanks @oelegeirnaert for suggesting this. As there were no follow-ups, I assumed it worked and I'll close this.

closed 2019-05-11 by reuterbal

alopezbu, 2019-06-06:
@oelegeirnaert is it possible that the "second copy" could be in black and white and the fisrt one in colour?

oelegeirnaert, 2019-06-06:
Not tested but I think it should work by replace line 175 with:

`new_image.paste(picture.convert('LA'), (self._pic_dims.outputSize[0],0))`