reuterbal/photobooth #114: Succès but freeze

thyristor59, 2019-03-28:
**Describe your photobooth**

I use photobooth in various nights to test it and be reliable for the wedding of my best friend!

**Hardware (please complete the following information):**
 - Raspberry Pi 3
 - Nikon d3200/d5500
 - GPIO: No
 
 **Additional equipment**
  - Flash/Light 
https://www.amazon.fr/Yongnuo-YN-568EX-Master-Flash-Speedlite/dp/B00ENFF896/ref=asc_df_B00ENFF896/?tag=googshopfr-21&linkCode=df0&hvadid=52075807406&hvpos=1o2&hvnetw=g&hvrand=15055726846128071980&hvpone=&hvptwo=&hvqmt=&hvdev=m&hvdvcmdl=&hvlocint=&hvlocphy=9055711&hvtargid=pla-125337967515&psc=1
  - https://www.amazon.fr/Pixtic-Adaptateur-Batterie-appareil-numérique/dp/B00ITGPLT2/ref=mp_s_a_1_3?keywords=batterie+secteur+nikon&qid=1553787488&s=gateway&sr=8-3
  - https://www.amazon.fr/Multi-Touches-Infrarouge-Tactile-Overlay-Superposition/dp/B07D235GFF/ref=mp_s_a_1_4?keywords=cadre+tactile&qid=1553787552&s=gateway&sr=8-4

**Software (please complete the following information):**
 - latest version of the raspberry website with desktop

**Modifications:**
 - changing the stylesheet for a better layout

- raspap installation in order to remove printer problems via my smartphone

**Problems:**
 - However, even with the radiators on the raspberry, there are still some beugs / freeze completely, I have to turn off and turn on the raspberry every time, Do you have any idea of ​​an improvement?
I think it's not software but I would like to improve this point, being a novice I'm looking for a track!
In advance, thank you if anyone can inform me

**Pictures:**
[image]
[image]
[image]

thyristor59, 2019-04-03:
I had a solution, I bought a USB hub results I had only 2 frezze on a party!

Monstertuner, 2019-05-07:
Bonjour! Est il possible de me faire une copie de ta SD avec le logiciel ? (Envoi via MEGA ou une plateforme d'upload quelconque) 
Je rencontre beaucoup de bug que je n'arrive pas à gérer...
Merci!

Monstertuner, 2019-05-08:
> 
> 
> Bonjour! Est il possible de me faire une copie de ta SD avec le logiciel ? (Envoi via MEGA ou une plateforme d'upload quelconque)
> Je rencontre beaucoup de bug que je n'arrive pas à gérer...
> Merci!

C'est bon j'ai mis à jour, heureusement que j'ai lu ta page, mon Debian était obsolète.
Bon courage!

reuterbal, 2019-05-11:
Thank you for the freedback! We will include it in our list of success stories.

With all the radiators, and your working solution it is probably a power issue. Maybe try replacing the power source for your raspi?

thyristor59, 2019-05-12:
another evening yesterday, a lot of frost, I can not understand why ...
and now the camera is no longer detected at startup, I pull out the hair lol.
I have to continually restart the system.

Thé power of alimentation it's ok

reuterbal, 2019-05-13:
This sounds really strange, I'm sorry that you have these kinds of problems.
Regarding the camera detection issue: I sometimes have this as well after a crash. Typically, the following strategy helps:
1. Switch camera off and back on (helps mostly)
2. Switch camera off, unplug USB cable, switch back on, plug in USB cable (helps in 99% of the remaining cases)
3. Reboot with camera off and unplugged (fixed it every time)
However, I'm using a Canon and Nikon might have its own quirks.

thyristor59, 2019-05-13:
that's exactly what I need to do but sometimes 2-3 shutdown of the set should be done with resume normal operation!
do not apologize your work remains incredible
no less than 90 impressions were made.
I try a next cannon ready!

reuterbal, 2019-05-13:
One thing that you could look into for me, since you're running raspap:
When the photobooth freezes, does raspap still work? I.e. are you able to connect to Raspi via WiFi?
If it does and you are familiar with SSH, you could also try to log in remotely once a freeze happens and look at system usage with `top` and see if anything is logged in `/var/log/syslog`.
This would tell us whether only the photobooth freezes or the entire system is stuck. In the latter case, it is typically not a software issue but rather a hardware thing - on a Raspi this usually means one of the following:
- overheating
- too little / flaky power supply
- SD card

thyristor59, 2019-05-13:
I would do tests in the week, not stupid the idea, I had not thought.
I certainly think of the SD card that is not new, I flashed a lot of times

thyristor59, 2019-05-15:
I remembered that line that I did not delete at installation because it did not exist at the command " sudo rm /usr/share/gvfs/remote-volume-monitors/gphoto2.mounts"
 it causing the cable problem?

about freezing it:
I can recreate the beug wish!
when several click on start over. the page freezes.
try with raspap, I can connect to the raspberry successfully.
open a Terminal, a folder etc ...
I made the command "top", a page is displayed but I am not in the measure of the interpreter.
I do not understand the command / var / ... / ...
is it possible to correct that?

reuterbal, 2019-05-31:
I have experienced similar issues and tried to make it a bit more resilient against errors. There seems to be some issue when already on idle and receiving an idle event.
If you like, try out the development branch, i.e., run 
```
git checkout development
```
in your photooboth directory, or alternatively, install it in a new folder following the instructions but use
```
git clone -b development https://github.com/reuterbal/photobooth
```
instead of the regular clone command.

With that version I did not have any freeze in two events, only some error messages once in a while without actually crashing.

thyristor59, 2019-05-31:
great, I try this afternoon on my test version. I hope this simple order is effective !!! look forward to seeing the result! the wedding is the 22/06.

Lot of thanks reuterbal 👍

thyristor59, 2019-05-31:
unfortunately it did not work for my case, during the installation of development, I have a crash pycups and I have not managed to install it. I try to find the time to watch. I make small progress thanks to you, I understand at least what I do

thyristor59, 2019-06-02:
good news for me! the development version is working now. however, I do not understand the @mail mode at all. I downloaded a simple server but without result. the problem of multiple clicks on the tabs are set, no more frezze. however now the printer is extremely long, (still wired)

reuterbal, 2019-06-13:
You have to run
```
pip install -e .
```
again to make sure all requirements are met.

thyristor59, 2019-06-13:
Yes

thyristor59, 2019-07-02:
Hello !! The photobooth was a huge hit at my best friend's wedding!

 no beug !!!

 just recharge it in ink and paper. however, I did not have time to make some pictures of the material in the room for you to share ... 

1000x thanks

reuterbal, 2019-07-05:
That is great to hear!

PGotzmann, 2019-08-30:
First, I want to thank you for your absolutely great work. This is for sure one of the best free photo booth software on github.

I have decided to comment this issue rather than opening a new one as this appears to be related.
If you want me to open a new Issue instead please just let me know.

I am experiencing the same freezes when clicking the start over button more the once (which unfortunately happens pretty often due to impatient users).
As suggested above I tried to update to the latest version, however I am still getting a type error ( unknown event “idle” in State Greater)
Looks like a timing issue to me, as this signal clearly belongs to the previous state and is supposed to trigger the change in State to “State Greater” in the first place.

As a workaround I added an additional if clause to “handle event” to both class IdleState and GreeterState in StateMachine.py to simply ignore the repeated/delayed event “idle” respectively “trigger”.

This clearly is not the root cause but worked fine for me so far.

oelegeirnaert, 2019-09-03:
Hi @thyristor59 ,

As you have a mirror photobooth, i guess you're not previewing the camera-output during countdown to your screen? Because mine seems to become slow when using full-screen modus on a big screen...

But indeed, another option could be to create a mirror and only show the final assembled picture on the big screen.

Kind regards & good evening,
Oele Geirnaert

thyristor59, 2019-09-04:
Hello, @oelegeirnaert 
In fact at the beginning I had an old camera that did not take the liveview, that's why I had a mirror, to position (+/-) in the frame. it was not optimal .... the filter without complexion / mirror 
I had it by a friend, but it is sold in DIY store.

oelegeirnaert, 2019-09-04:
So now you're using the liveview option as well? Is it showing with delays?

thyristor59, 2019-09-04:
Yes now I use the liveview without worries, I always have a latency but this is not disturbing. 3 seconds is actually 6 seconds. (must be counted as a child 1111 222222 33333) 

The mirror without complexion makes its effect of style! 

for my question: https://github.com/reuterbal/photobooth/issues/158