reuterbal/photobooth #115: Can't seem to install Pi Camera

djgarethm, 2019-04-09:
**Describe the bug**

When trying to install the photobooth with the Pi Camera extra, the command pip install -e .picamera does not complete.

**To Reproduce**

Steps to reproduce the behavior:
1. Enter the virtual environment as per install instructions (source .venv/bin/activate) 
2. Type the command to use PIP to install photobooth and the extra module for Pi Camera (pip install -e .picamera)

**Expected behavior**

Installation completes successfully.

**Screenshots/Screencast**

[image]
Error:
".picamera should either be a path to a local project or a VCS url beginning with svn+, git+, hg+, or bzr+"

**Hardware (please complete the following information)**

 - Device: Raspberry Pi Zero W (*does* work with EOS 500D plugged in via USB)
 - Camera: Pi Camera
 - GPIO: Yes

**Software (please complete the following information)**

 - OS: Raspbian Stretch
 - Python version 3.5.3

Packages:
(.venv) root@raspberrypi:/var/photobooth# pip freeze

automationhat==0.1.0
blinker==1.3
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
chardet==2.3.0
click==6.6
colorama==0.3.7
colorzero==1.1
cryptography==1.7.1
drumhat==0.1.0
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==0.12.1
fourletterphat==0.1.0
gphoto2==1.9.0
gpiozero==1.5.0
idna==2.2
itsdangerous==0.24
Jinja2==2.8
keyring==10.1
keyrings.alt==1.3
MarkupSafe==0.23
microdotphat==0.2.1
mote==0.0.4
motephat==0.0.2
numpy==1.12.1
oauthlib==2.0.1
opencv-python==3.4.4.19
pantilthat==0.0.7
pgzero==1.2
phatbeat==0.1.1
-e git+https://github.com/reuterbal/photobooth.git@471fb07a9190883a6a7efadab8f95b02b95b8ef8#egg=photobooth
pianohat==0.1.0
picamera==1.13
picraft==1.0
piglow==1.2.4
pigpio==1.38
Pillow==4.0.0
pkg-resources==0.0.0
pyasn1==0.1.9
pycrypto==2.6.1
pygame==1.9.3
pygobject==3.22.0
pyinotify==0.9.6
PyJWT==1.4.2
pyOpenSSL==16.2.0
pyserial==3.2.1
python-apt==1.1.0b5
pyxdg==0.25
rainbowhat==0.1.0
requests==2.12.4
requests-oauthlib==0.7.0
RPi.GPIO==0.6.5
RTIMULib==7.2.1
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
sense-hat==2.2.0
simplejson==3.10.0
six==1.12.0
skywriter==0.0.7
sn3218==1.2.7
spidev==3.3
ssh-import-id==5.6
touchphat==0.0.1
twython==3.4.0
unicornhathd==0.0.4
urllib3==1.19.1
virtualenv==15.1.0
Werkzeug==0.11.15

Hope that I've just done something silly - but all ideas welcome!

djgarethm, 2019-04-10:
Would delete this - but I don't seem to be able to. 

Turns out that you don't need to install the picamera as a separate module - it seemed to just work for me! 

Problem I had with activation was erm... I had put the ribbon cable the wrong way round!! 

Cracking piece of software - thank you!

closed 2019-04-10 by djgarethm

reuterbal, 2019-05-11:
On Raspbian, the picamera module is by default already installed on the system.