reuterbal/photobooth #122: raspicam

apm1007, 2019-04-21:
im using raspicam as option as mentioned under settings. The picture it takes are just 800x480 pixel per picture. How to change? the current resolution is horrible

xiberger, 2019-04-21:
you can change the resolution in the PiCamera by setting `self._cap.resolution = (x, y)` 
in the `getPicture()` call - check out the Pi Camera manual for the max resolution of your version

xiberger, 2019-04-24:
@apm1007 did you manage to adjust the resolution? - if so, this issue can be closed

apm1007, 2019-04-26:
is it to be set in CameraPicamery.py? Didnt found the line :(

reuterbal, 2019-05-11:
It is even better to specify it in the constructor, i.e., replace l. 51 in https://github.com/reuterbal/photobooth/blob/master/photobooth/camera/CameraPicamera.py by
```
 self._cap = PiCamera(resolution=(x, y))
```
with `(x, y)` the tuple of your resolution.

reuterbal, 2019-06-27:
Closing this due to inactivity

closed 2019-06-27 by reuterbal