reuterbal/photobooth #123: Sluggishness During Live Preview

maddog9486, 2019-04-24:
I after a little difficulty, I finally got the software running with my Nikon D3300 DSLR, however, when I start the photobooth and push the button tontake the pictures, the Live Preview pops up with the countdown, but is it so slow. The image on the screen is sluggish and the countdown takes well over a second to change numbers.

Is this a limitation of the Raspberry Pi or is there some settings I can set to optimize? I would expect it to be real-time and fast so people can easily adjust before the image is taken.

NOTE: my output of the Raspberry Pi was to a 42” television, so not sure if this was the issue due to scaling. (I did up the GPU memory in the Raspbian Configuration, but it didn’t appear to make a difference)

Also, is it normal for the camera to fire when clicking start photobooth?

xiberger, 2019-04-24:
I have the same issue with a PiCameraModueV2, a Pi3 B+ and all variations of GPUMemory - couldn't determin where the problem is, might be an issue with the update rate and or the size of the images from the camera. Therefore drawing might take up a lot of computing.. but that is just a lucky guess w/ data to support the claim

maddog9486, 2019-05-01:
xiberger - assuming you have not found a way around this? I hate to turn the live preview off because everyone likes to see what they look like beforehand.

xiberger, 2019-05-02:
@maddog9486: I haven't had time to look into this. 
We used it with the sluggish CountDown at my brother in laws bachelor party and it was fine. People really liked it and didn't mind the slow response.

What I did is to turn down the time of the countdown to "4 seconds" which in reality are somewhere around 6 - but the PiCam needs 2secs to start up, so that is fine.

reuterbal, 2019-05-11:
It is really an issue of the Raspberry Pi - either the limited IO bandwidth or, more likely, the slow CPU are the cause for that. Feel free to add performance improvements that accelerate the rendering of the picture.

closed 2019-05-11 by reuterbal