reuterbal/photobooth #124: High latency after pressing print button and printer quality?

Tinker232, 2019-04-25:
Hey @reuterbal 
the photobooth works very well on my raspberry pi 3B+.
I had bought a canon selphy cp1300 printer and installed the drivers and stuff like cups etc. according to your instructions.
Then I tried the print function with the photobooth software. After pressing the button "print" the printer didn´t get the data immediately. There are about 10 seconds latency before the printer is doing anything. As for comparision, I printed a photo from the shell and the printer gets the data immediate without latency. 
Do you know why? 

The second issue is the quality of the photos, but i think it causes by the cups driver?! For the cp1300 printer I use the cp1200 driver (gutenprint 5.3.1). In comparision to a photo, which i had printed from my smartphone directly via wifi to the printer, the photos looks blurred, faded and stump. So, i tried to improve the quality with adjusting some settings on the cups GUI, but without success :/
Does somebody still have an idea?

Many Thanks,
Tinka

reuterbal, 2019-05-11:
The high latency is due to the way printing works: For portability reasons, I relied on the printing functionalities of Qt5, which creates a PDF file from the picture, which is then passed to the printer stack of the operating system, which typically converts then the PDF to PS or something the printer can understand. This simply takes a while.
You can try out the new PyCups printer module contributed by @oelegeirnaert - in my tests this was actually faster.

The printer quality is also something that bothers me, but it is simply due to the terrible driver of gutenprint. Colors are very off (typically to blue-ish) for me.
This is also true when printing from other programs, while the same pictures printed from a USB drive on the printer look a lot better.

reuterbal, 2019-05-13:
Something that could be worth investigating is whether one could talk to the printer in the same way as Apple devices do ("AirPrint"). I read somewhere that at least jpg-files are printed correctly that way. It requires to talk to the printer via WiFi (which might actually be a plus: one less cable). I'll try this if I find time and let you know if it worked.
(Some background on AirPrint: https://wiki.debian.org/AirPrint , https://www.finnie.org/2010/11/13/airprint-and-linux/)

reuterbal, 2019-05-13:
I managed to get printing via WiFi working (at least on Ubuntu 18.04, haven't tested it on Raspbian, yet), but I'm not sure how reliable it is.
This page was very helpful: https://wiki.debian.org/DriverlessPrinting
The basic ideas are the following:
- Connect the printer to the same WiFi as the computer/raspi
- Verify that the printer is detected by one of the following methods:
```
$ driverless
ipp://CP130012cc11.local:631/ipp/print
$ ippfind
ipp://CP130012cc11.local:631/ipp/print
$ lpinfo -v
network ipp://CP130012cc11.local:631/ipp/print
```
- `driverless` is supposed to create the PPD file itself, however, that did not work in my case (but it did for my regular Samsung Laser Printer):
```
$ driverless ipp://CP130012cc11.local:631/ipp/print
ERROR: Unable to create PPD file: No IPP attributes.
```
- Luckily, someone else had the same issue before and was smart enough to generate the PPD on macOS and provide it online: https://www.voss.earth/2018/08/31/kurztipp-canon-selphy-wlan-drucker-cp910-oder-cp1300-unter-linux-cups-verwenden/
In there, I had to modify the address for `APSupplies` and remove the lines starting with `cupsICCProfile...` as those are file locations on a macOS system (this should probably be replaced by the file links to the sRGB (`/usr/share/color/icc/colord/sRGB.icc`?) and Gray (...?) profiles.
- With that, I could add the printer in CUPS using the Add Printer dialogue and selecting the entry with `(driverless)` in its name. When asked for the model, I provided the PPD file instead and afterwards was able to print using that queue via WiFi.
- The printer lost its WiFi connection repeatedly. I don't know if this was due to our WiFi being unstable at that specific location, or if it is something the printer does (The printer reports itself as connected but doesn't have an IP from the DHCP anymore).

Comparing a printout via this setup to one using the USB cable and the Gutenprint driver, colors appear to be much nicer.

reuterbal, 2019-05-13:
A little follow up:
- Using the system sRGB profile did not change anything. So this line can probably remain deleted
- Giving the printer a fixed IP address and replacing the Bonjour/Autodiscovery addresses ("CP1300xxxxxx.local") by the fixed address seems to overcome the disconnect issues. This is probably a good idea in any case.
- I have not managed to get borderless printing working. I tried copying over some of the paper specifications from the Gutenprint driver but without any success.

thyristor59, 2019-05-30:
hello reuterball, sorry not to be in the subject but how should I do to test the version of oelegeirnaery?

wikijm, 2019-05-30:
Hi @thyristor59

You just have to follow install instruction but replacing https://github.com/reuterbal/photobooth by ttps://github.com/oelegeirnaert/photobooth 😉

Next time, I invite you to create a new and specific issue ☺️

reuterbal, 2019-05-31:
That is actually wrong. Simply switch to the development branch, there the new printer module is available.

For example, run
```
git clone -b development https://github.com/reuterbal/photobooth
```

wikijm, 2019-05-31:
Hi @reuterbal 

Hi @reuterbal 

Due to "sorry not to be in the subject" sentence, my thought was that @thyristor59 wants to give a try on https://github.com/reuterbal/photobooth/issues/75 features.

My bad for the misunderstanding 😞

thyristor59, 2019-08-16:
I restart the subject ... 
I fight with the printer for 2 days and no way to do it in WiFi. 

however you have removed the tips and tricks for USB installation. (change folder for cp1200 drivers) 

can you rewrite it please?

Tinker232, 2019-10-11:
Hi guys,
are there any new findings on this topic?
Nice weekend :)
Greetz Tina

Tinker232, 2019-10-13:
Sorry, I overlook the solution with the wifi connection. It works fine, but I struggle with the borderless printing. There aren´t possibilites to adjust some settings with your Canon .ppd file in the cups environment, so the printed photos have left 1mm and at the bottom 3mm white space. How can I adjust the printed size? Changes in the photobooth menu for the paper size doesn´t affect! 
Thanks Tina

Tinker232, 2020-02-12:
No idea????

Tinker232, 2021-09-28:
Hey @reuterbal and @all, 
I use the photbooth for weddings and birthday parties. I modified some things and it works very well, but the printer makes me crazy... 
The CP1300 ist connected via airprint over a fixed IP-Adress 192.168.180.100. At first, the photobooth and printer works as expected, but after about 6 hours uptime, the printer looses the connection to the pi. The only thing you can do is restart the printer manually.
I try to provoke the behavior with a long existing printjob at the queue (no paper in tray) and filling the queue with more and more jobs. And after a while the printer lost the connection, but this occurs only when the uptime is over ~6 hours...
I noticed the "lpinfo -v" and "ippfind" commands shows the printer with his CP130053a843.local address. @reuterbal Please, 

How can I replace the bonjour adress to my static IP-Adress? 
Does anyone have an idea to fix the problem?

Thanks Tina :-)