reuterbal/photobooth #125: Enable pull-up/pull-down resistors in code

Jaahn123, 2019-04-26:
**Describe the bug**

I build the circuit basically according to the scheme provided here. Instead of RGB LEDs I changed the code a bit to drive three white LED strips and change the brightness.
There is a common ground for the Rpi as well as for the LED power supply. 
I noticed some odd behaviour of the trigger if the LED power source was on. If the PSU was on, the camera was always triggered. After a while I figured that the issue is that there is no pull-up resistor for the trigger and the pin I choose has the internal one configured as pull-down on start-up. I therefore added a 10kOhm pull-up resistor and it worked.

The question is if one could activate the pull-up/down resistors directly in the code? 
That would render the three/four resistors superfluous.

**Hardware (please complete the following information)**

 - Device [Raspberry Pi 3B+, Canon 6D, Meanwell HLG240H]
 - Camera [e.g. Canon EOS 500D]
 - GPIO: [Yes]

**Software (please complete the following information)**

Raspbian Stretch

Jaahn123, 2019-05-05:
Here is an update:
Since the external pull-up resistor didnt solve my issue I modified the GPIO _init_.py.

`def setButton(self, bcm_pin, handler):
        
try:
            self._buttons.append(self.Button(bcm_pin, hold_time=0.1))
            self._buttons[-1].when_held = handler
        except self.GPIOPinInUse:
logging.error('Pin {} already in use!'.format(bcm_pin))`

This way any noise that I might have on the trigger pin will be canceled out as the trigger needs to be pushed at least 0.1 second before the function will be run.

reuterbal, 2019-05-11:
See here: https://gpiozero.readthedocs.io/en/stable/api_input.html#button
Pull up resistors are enabled by default.

closed 2019-06-27 by reuterbal