reuterbal/photobooth #128: libEGL Failed and

Monstertuner, 2019-05-05:
Hello everybody!

I was looking for a GUI to use, i've planned to make a surprise for my sister's wedding.
I found your GUI and installed it as you explained in your install file.
I've got some issues on my Raspberry pi using the photobooth.
First one, when i run the photobooth i've got the message

`libEGL warning: DRI2: failed to authenticate`

I tried to fix this by many ways, even a method that has already been posted on the gitub issues but it won't work, problem is not fixed.
Anyway, even with the bug the Gui launches fine, everything can be customized and my GPIO button works too.

Second Issue: When i trigger the button to take a picture, nothing happens, on the terminal i've got a surprise:

`Traceback (most recent call last):`
 ` File "/home/pi/photobooth/photobooth/gui/GuiSkeleton.py", line 91, in handleState
    self.showGreeter(state)`
`  File "/home/pi/photobooth/photobooth/gui/Qt5Gui/PyQt5Gui.py", line 187, in showGreeter
    lambda: self._comm.send(Workers.MASTER, GuiEvent('countdown'))))`
`TypeError: __init__() missing 1 required positional argument: 'countdown_action'`

I've got some issues before those which i fixed:

Running GUI
=> Error with the gphoto2 
Fixed with editing Defaults.Cfg
=> `module=picamera`

**Hardware**
- Raspberry Pi 3b
- Picamera V2
- GPIO: Yes
- Raspberry pi original power source

**Software**

 - Raspbian  8
 - Python version 3.4.2
 - Accessing RaspberryPi Via VNC Viewer on my computer (windows 10)
 
 **Installed packages**
 
 ```
 Flask==0.10.1
Jinja2==2.7.3
MarkupSafe==0.23
Pillow==2.6.1
RPi.GPIO==0.6.3
RTIMULib==7.2.1
Werkzeug==0.9.6
asciinema==0.9.8
cffi==0.8.6
chardet==2.3.0
codebug-i2c-tether==0.2.3
codebug-tether==0.7.0
colorama==0.3.2
cryptography==0.6.1
gpiozero==1.4.0
guizero==0.5.4
html5lib==0.999
itsdangerous==0.24
mcpi==0.1.1
numpy==1.8.2
pgzero==1.1
picamera==1.13
picraft==1.0
pifacecommon==4.2.1
pifacedigitalio==3.1.0
pigpio==1.35
ply==3.4
pyOpenSSL==0.14
pycparser==2.10
pygame==1.9.2a0
pygobject==3.14.0
pyinotify==0.9.4
pyserial==2.6
python-apt==0.9.3.12
python-debian==0.1.27
requests==2.4.3
sense-emu==1.0
sense-hat==2.2.0
six==1.8.0
smbus==1.1
spidev==3.0
twython==3.1.2
urllib3==1.9.1
virtualenv==1.11.6
wheel==0.24.0

 ```

**Photobooth log**

```
2019-05-05 12:53:56,695 - root - INFO - Initializing photobooth
2019-05-05 12:53:56,697 - root - INFO - Photobooth version: unknown
2019-05-05 12:53:56,699 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2019-05-05 12:53:56,704 - root - INFO - Reading config file "photobooth.cfg"
2019-05-05 12:53:56,738 - root - DEBUG - New state is "WelcomeState"
2019-05-05 12:53:56,755 - root - DEBUG - Start CameraProcess
2019-05-05 12:53:56,768 - root - DEBUG - Start WorkerProcess
2019-05-05 12:53:56,773 - root - DEBUG - Start GpioProcess
2019-05-05 12:53:56,777 - root - INFO - GPIO disabled
2019-05-05 12:53:56,781 - root - INFO - Number of last existing file: 0
2019-05-05 12:53:56,784 - root - INFO - Saving pictures as "2019-05-05/photoboothXXXXX.jpg"
2019-05-05 12:53:56,788 - root - INFO - Number of last existing file: 0
2019-05-05 12:53:56,789 - root - INFO - Saving pictures as "2019-05-05/photobooth_shot_XXXXX.jpg"
2019-05-05 12:53:56,789 - root - DEBUG - Start GuiProcess
2019-05-05 12:54:26,367 - root - DEBUG - Handling event "start"
2019-05-05 12:54:26,369 - root - DEBUG - New state is "StartupState"
2019-05-05 12:54:26,372 - root - INFO - Using PiCamera
2019-05-05 12:54:26,838 - root - INFO - Using camera with preview functionality
2019-05-05 12:54:27,825 - root - DEBUG - Handling event "ready"
2019-05-05 12:54:27,826 - root - DEBUG - New state is "IdleState"
2019-05-05 12:54:29,936 - root - DEBUG - Handling event "trigger"
2019-05-05 12:54:29,937 - root - DEBUG - New state is "GreeterState"
2019-05-05 12:54:31,973 - root - DEBUG - Handling event "Teardown(2)"
2019-05-05 12:54:31,974 - root - DEBUG - New state is "TeardownState"
2019-05-05 12:54:31,980 - root - DEBUG - Handling event "welcome"
2019-05-05 12:54:31,982 - root - DEBUG - New state is "WelcomeState"
2019-05-05 12:54:33,962 - root - DEBUG - Handling event "Teardown(0)"
2019-05-05 12:54:33,964 - root - DEBUG - New state is "TeardownState"
2019-05-05 12:54:33,964 - root - DEBUG - Exit GuiProcess
2019-05-05 12:54:33,967 - root - DEBUG - Exit GpioProcess
2019-05-05 12:54:33,968 - root - DEBUG - Exit WorkerProcess
2019-05-05 12:54:33,977 - root - DEBUG - Exit CameraProcess
2019-05-05 12:54:34,032 - root - DEBUG - All processes joined, returning code 0
2019-05-05 12:54:34,040 - root - INFO - Exiting photobooth with status code 0
```

Thank you for your work Reuterbal

Monstertuner, 2019-05-08:
I updated my Debian from 8 to 9
Took Raspbian Stretch with Desktop and installed:
-> Photobooth works fine!

libEGL Failed to authenticate always there but the GUI works.

reuterbal, 2019-05-11:
I do get these warnings as well but they don't cause any issues. Other projects see them, too. For example: https://pi3d.github.io/html/FAQ.html#glx-dri2-not-supported-or-failed-to-authenticate

For the moment I would simply ignore it. It is not pretty but as long as nothing is actually broken, I don't want to waste time on it. Closing this for now.

closed 2019-05-11 by reuterbal