reuterbal/photobooth #129: Error automatic printing, issue with paper format

Chefsessel, 2019-05-09:
Hey everyone,

after a pauseI got back to finish my project. Still stuck at printing this one on an actual photo paper. This might not be a photobooth issue but it could be so I hope you bear with me.

I'm using a PI3 + raspian + CUPS and a hp envy photo 6230 as a setup. I do want to have the photos printed (using the PyQT5 module) after they are taken.

The error I'm getting is that my hp photosmart 6230 tells me that detected paper format is not the format that was chosen for this print job.

I'm using 15cm x 10cm photo paper. In photobooth settings / printer I've set paper size(mm) to 150 x 100. In CUPS it's the same, set to photo paper 15x10cm. However, it will bring up the above error.

Thing is, I can print the photos from terminal command with no problem. So it's not a genuine printer issue, it's more like how the data gets to the printer. Is photobooth assuming a border?

Does anyone have a similiar experience or an idea on how to fix this?

Any help highly appreciated!

Sebastian

reuterbal, 2019-05-11:
Do the jobs show up in the CUPS queue? I.e., can you see the printed jobs in your browser when opening http://localhost:631/jobs/ ? If they do, does this contain any error information?

Photobooth doesn't assume anything: it initially sets the page size to the specified values, and then squeezes the picture in that size and hands it over to the printing stack.

reuterbal, 2019-06-27:
Closing this due to inactivity.

closed 2019-06-27 by reuterbal