reuterbal/photobooth #130: Proposition : add a paper & ink background countdown for Selphy CP1300

RonandoDeLaVega, 2019-05-13:
Hello,

Thanks for your great work !
I managed to realise a photobooth & I thought about an idea...

After browsing all the success stories, I found that the printer Canon Salphy CP1300 was the most used so I bought one as it seems to be a "standard" for the photobooth !

During the party, the people don't think about changing paper and/or ink.

What I thought about was some kind of background countdown which displays a warning message : no more paper/ink please change if you want to print (I've changed / I won't print).
Maybe even a video can be thrown & show how to do !

Thanks again !
Ronando

reuterbal, 2019-05-14:
Instead of a countdown (wrong too easily) I would rather extend the printer interface to report back on errors - with the new pycups printer interface this should be relatively easy to do. 
The problems with the counter that I see:
- There are 36 and 54 print cartridges (I buy the 108 print packages and mix those, depending which is cheaper at the moment)
- Paper comes in 18 print packages
- You might not start with a fresh cartridge
When I'm present at the event I usually avoid people letting change paper/cartridge themselves. Too often they swapped out the cartridge too early or struggled inserting it correctly. Usually a number of printouts piles up then and eventually somebody tells me about it.

oelegeirnaert, 2019-05-21:
Hi All,
I saw a new mailfeature is available in development.
Maybe PyCups can warn someone by mail?

reuterbal, 2019-05-31:
By a customers request I added a very simple mailer that can send the assembled pictures by email directly after postprocessing. Something that could be imagined is a worker task that regularly checks for certain things, e.g. supply levels on the printer, and then can notify by mail, telegram, etc. I'll look into that. But it will have low priority because on most of my events the photobooth is offline and thus I would have little benefit.

thyristor59, 2019-06-02:
how to manage the mail function and how to create the page to see the photos? in advance thank you for your patience

oelegeirnaert, 2019-06-03:
> how to manage the mail function and how to create the page to see the photos? in advance thank you for your patience

The mail function is currently only available in the development branch.
I don't know when the new release is planned.
https://github.com/reuterbal/photobooth/tree/development

reuterbal, 2019-06-13:
I'll merge the development branch into master in the next days.

Tinker232, 2019-06-21:
Hey, I ´m looking for the same issue and tried to fix the problem with other linux printer driver called Turboprint. There is a 30-days free trial version with full functions available and after expiry you can buy a license for less money, about 20€. With that, you can adjust a lot of printer functions, like paper format, color adaption and a printer status monitor. The printer status monitor pops up, if a printer error occurs. 
The installation on raspberry is very easy and it seems that the quality of the photos is increased. 
But, the response time of the printer is very slow, which means that the printer gets the data at least 14s after the photo preview is displayed. 
Well, I had the same problems with the response time by using the Cups-Driver, but i fixed it by disabling the "print" button, which is displayed at the photo preview. Thus, all pictures will print automatically and with a quick response time. The printer starts the job even the preview is displayed, thats quick enough :-) 
The other issue is, that the status monitor doesn´t pop up so far while a printer error occur. I think I have to delete the whole CUPS-Driver before and try it again?! Maybe it could fix the problem with the slow response time as well?! 
Anybody who has experience with turboprint or other solutions?

Tinker232, 2019-10-11:
Are there new findings regarding to the possibility to inform the user about paper and ink status?
I think the best way to inform people about status like recharge paper is a small popup textmessage.
How can I realize that?
Thanks Tina