reuterbal/photobooth #131: Can´t run photobooth (error)

Jaghova, 2019-05-18:
When I try to run it, I get this error

2019-05-18 22:24:17,486 - root - INFO - Initializing photobooth
2019-05-18 22:24:17,487 - root - INFO - Photobooth version: 0.4.dev81+g471fb07.d20190518
2019-05-18 22:24:17,490 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2019-05-18 22:24:17,496 - root - INFO - Reading config file "photobooth.cfg"
2019-05-18 22:24:17,564 - root - INFO - Number of last existing file: 0
2019-05-18 22:24:17,569 - root - INFO - Saving pictures as "2019-05-18/photoboothXXXXX.jpg"
2019-05-18 22:24:17,572 - root - INFO - GPIO disabled
2019-05-18 22:24:17,574 - root - INFO - Number of last existing file: 0
2019-05-18 22:24:17,575 - root - INFO - Saving pictures as "2019-05-18/photobooth_shot_XXXXX.jpg"
libEGL warning: DRI2: failed to authenticate
qt5ct: using qt5ct plugin
Process GuiProcess-3:
Traceback (most recent call last):
  File "/usr/lib/python3.5/multiprocessing/process.py", line 249, in _bootstrap
    self.run()
  File "/home/pi/photobooth/photobooth/main.py", line 88, in run
    retval = Gui(self._argv, self._cfg, self._comm).run()
  File "/home/pi/photobooth/photobooth/gui/Qt5Gui/PyQt5Gui.py", line 49, in __init__
    self._initUI(argv)
  File "/home/pi/photobooth/photobooth/gui/Qt5Gui/PyQt5Gui.py", line 75, in _initUI
    self._gui = PyQt5MainWindow(self._cfg, self._handleKeypressEvent)
  File "/home/pi/photobooth/photobooth/gui/Qt5Gui/PyQt5Gui.py", line 273, in __init__
    self._initUI()
  File "/home/pi/photobooth/photobooth/gui/Qt5Gui/PyQt5Gui.py", line 279, in _initUI
    if self._cfg.getBool('Gui', 'fullscreen'):
  File "/home/pi/photobooth/photobooth/Config.py", line 76, in getBool
    return self._cfg.getboolean(section, key)
  File "/usr/lib/python3.5/configparser.py", line 826, in getboolean
    raw=raw, vars=vars, fallback=fallback, **kwargs)
  File "/usr/lib/python3.5/configparser.py", line 806, in _get_conv
    **kwargs)
  File "/usr/lib/python3.5/configparser.py", line 800, in _get
    return conv(self.get(section, option, **kwargs))
  File "/usr/lib/python3.5/configparser.py", line 1152, in _convert_to_boolean
    raise ValueError('Not a boolean: %s' % value)
ValueError: Not a boolean: Truue

My hardware is:

Raspberry Pi 3B+
Still not installed camera or printer

Jaghova, 2019-05-18:
After reinstalling it again, it works

closed 2019-05-18 by Jaghova

reuterbal, 2019-05-19:
You had a typo in your `photobooth.cfg` file: There was `Truue` instead of `True` somewhere (see last line of your error message).