reuterbal/photobooth #133: Margins.

roboticsguy1988, 2019-05-31:
I would like to be able to add margins for the top, bottom, left, and right side of the page. The images should respect the margins but NOT the background. It seems the thumbnails are right up to the left and right edges of the paper and I would like to put a 40px margin on the left and right without adding 40px in-between each picture.

reuterbal, 2019-05-31:
The config values `min_dist_x` and `min_dist_y` are adding margins. Simply increase these values.

roboticsguy1988, 2019-05-31:
If I change the x does that affect both left and right? And y affect both top and bottom?

reuterbal, 2019-05-31:
Yes.

roboticsguy1988, 2019-06-03:
This is not a sufficient solution. This puts space in-between each picture. I should have a left, top, right, and bottom margin. I only want additional space on the left and right of the page i do not want additional space in-between the pictures.

reuterbal, 2019-06-13:
Now I understood - I will see if I can do something this evening.

roboticsguy1988, 2019-06-13:
Awesome! Thanks so much. If you don't mind telling me what file you modified to do it, I would like to see the differences.  Python is something I hope to learn.

reuterbal, 2019-06-14:
I added the ability to have different margins in the interior and exterior. Please try the development branch (see https://github.com/reuterbal/photobooth/wiki/Tips-and-tricks#switch-to-development-branch)
The changes can be seen here: https://github.com/reuterbal/photobooth/commit/8b14ec29141b4b519bf80350012d266754905e9f

commit 8b14ec29 (2019-06-14): Added additional parameters to have different border sizes between thumbnails and between thumb and border. Fixes #133
closed 2019-06-27 by reuterbal