reuterbal/photobooth #138: Imprimante DNP DS140, dev écrans ?

ThiibWatatax, 2019-06-08:
Bonjour,

J'ai pu utiliser le programme avec une configuration assez standard cependant j'envisage de nombreuses modifications mais je suis un peu bloqué.

Auriez-vous quelques pistes ?

1. Intégration d'une imprimante DNP DS40, celle-ci ne fonctionne pas aujourd'hui, nous utilisions jusqu'à présent une Canon CP1300

2. Intégration d'un live preview. J'ai réussi dans un programme en stand-alone... mais je n'ai pas réussi à comprendre le fonctionnement pour réussir à créer un écran ou un widget permettant d'afficher le live preview. Idéalement j'aimerai l'intégrer en superposition du décompte

3. Dans le même type j'aimerai pouvoir rajouter une étape dans le programme permettant de choisir de simplement enregistrer sans imprimer la photo.

4. Un écran de choix au début également pour choisir un mode de photo afin de sélectionner un template ou un autre par exemple

5. Enfin dans un état final j'aimerai pouvoir intégrer un envoie par mél 

Bref au final mon plus gros soucis est de réussir à comprendre l'imbrication du tout pour réussir à créer une étape / un écran... Ensuite le code pour les fonctionnalités j'ai réussi à m'en sortir... Et enfin galère pour réussir à utiliser une nouvelle imprimante...

Merci de ton aide et à bientôt

reuterbal, 2019-06-13:
Sorry, I don't understand french

wikijm, 2019-06-13:
Hi @reuterbal, let me help for the translation:

> Hello,
> 
> I was able to use the program with a fairly standard configuration however I am considering many changes but I am a little stuck.
> 
> Do you have any leads?
> 
>     1. Integration of a DS40 DNP printer, this one does not work with color.... Only with black and white.... I have the following error message in the logs "CreateProfile failed: org.freedesktop.ColorManager.AlreadyExists:profile id'DS40-RGB...' already exists"
>     Driver used: STP01258.PPD
>     Driver Version: 5.3.1
>     Make and Model: Dai Nippon Printing DS40 - CUPS+Gutenprint v5.3.1 (color)
>     Until now we used a Canon CP1300 which worked well with color....
> 
>     2. Integration of a Live preview: I succeeded in a stand-alone program... but I couldn't understand how it works to successfully create a screen or widget to display the live preview.  Ideally I would like to integrate it into the superposition of the countdown
> 
>     3. In the same type, I would like to be able to add a step in the program allowing to choose to simply save without printing the photo.
> 
>     4. A choice screen at the beginning also to choose a photo mode to select a template or another for example
> 
>     5. Finally in a final state I would like to be able to integrate a sending by e-mail
> 
> In short, in the end, my biggest concern is to be able to understand the interweaving of everything in order to create a step / screen... Then, the code for the features I managed to get out of it.... And finally, it's hard to succeed in using a new printer....
> 
> Thank you for your help and goodbye.
>

reuterbal, 2019-06-13:
For the overall picture, see [here](https://github.com/reuterbal/photobooth/blob/master/supplementals/state-chart/state-chart.pdf) for the state-chart.

Regarding your individual questions:

1. Can you print in color from any other application? Make sure that color printing is chosen as the default for the printer (see http://localhost:631).
2. Live Preview is already included. What camera module are you using?
3. The picture is always saved, printing must be explicitly triggered (unless configured otherwise).
4. That would require adding a new step to the state chart.
5. There is a simple mailing functionality available in the development branch. It sends every picture to a pre-configured mail address. I would love to have it such that users can send it themselves, however, this requires implementing a touchscreen keyboard.

wikijm, 2019-06-13:
🇫🇷
@ThiibWatatax :

Pour te donner une réponse globale, je t'invite à :

- Communiquer en anglais, peu de personnes vont être en mesure de comprendre et t'apporter de l'aide. Si tu n'es pas à l'aise avec l'anglais, tu peux utiliser le traducteur [Deepl](https://www.deepl.com/translator).
- Parcourir les tickets [déjà existants](https://github.com/reuterbal/photobooth/issues)
- Créer un ticket par incident / demande de fonctionnalité
- Parcourir le Wiki, qui regorge de pas mal d'astuces et qui est régulièrement mis à jour

----------------------------------------------------------------------------
 @ThiibWatatax 
To give you a global answer, I invite you to:

- Communicate in English, few people will be able to understand and help you. If you are not comfortable with English, you can use the translator[Deepl] (https://www.deepl.com/translator).
- Browse[already existing] tickets (https://github.com/reuterbal/photobooth/issues)
- Create one ticket per incident / feature request
- Browse the Wiki, which is full of many tips and is regularly updated