reuterbal/photobooth #139: [-2]  Bad Parameters

chino-lu, 2019-06-09:
**Describe the bug**

Starting photobooth in Gphoto2 mode (not command line in order to have liveview) the box tells me [-2] Bad Parameters

**To Reproduce**

Steps to reproduce the behavior:
1. Connect Sony Alpha 7 III
2. Start Photobooth set up to Gphoto2-Python
3. Click Start photobooth
4. See error

**Expected behavior**

The first screen should appear saying "press the button"

**Screenshots/Screencast**

```bash                                                                                                         
2019-06-10 04:58:54,276 - root - INFO - Using python-gphoto2 bindings
2019-06-10 04:58:54,838 - root - INFO - Camera summary: Manufacturer: Sony Corporation
Model: ILCE-7M3
  Version: 2.0
  Serial Number: 00000000000000003282943003847211
Vendor Extension ID: 0x11 (1.0)
Vendor Extension Description: Sony PTP Extensions

Capture Formats:
Display Formats: JPEG, Unknown(b301), ARW

Device Capabilities:
        File Download, No File Deletion, No File Upload
        No Image Capture, No Open Capture, Sony Capture

Storage Devices Summary:

Device Property Summary:
Compression Setting(0x5004):(readwrite) (type=0x2) Enumeration [2,3,4,16,19,20,18] value: 3
White Balance(0x5005):(readwrite) (type=0x4) Enumeration [2,4,32785,32784,6,32769,32770,32771,32772,7,32816,32786,32800,32801,32802] value: Automatic (2)
F-Number(0x5007):(readwrite) (type=0x4) Range [0 - 65535, step 1] value: f/4 (400)
Focus Mode(0x500a):(readwrite) (type=0x4) Enumeration [1,2,32772,32773,32774,32777] value: Automatic (2)
Exposure Metering Mode(0x500b):(readwrite) (type=0x4) Enumeration [32769,32770,32772,32773,32771,32774] value: 32769
Flash Mode(0x500c):(readwrite) (type=0x4) Enumeration [2,1,4,3,5,32769,32771,32817,32818,32833,32834] value: Fill flash (3)
Exposure Program Mode(0x500e):(readwrite) (type=0x4) Enumeration [32768,2,3,4,1,32848,32849,32850,32851,32900,32901,32902,32903,7,32785,32789,32788,32786,32787,32791] value: M (1)
Exposure Bias Compensation(0x5010):(readwrite) (type=0x3) Enumeration [0,5000,4700,4500,4300,4000,3700,3500,3300,3000,2700,2500,2300,2000,1700,1500,1300,1000,700,500,300,-300,-500,-700,-1000,-1300,-1500,-1700,-2000,-2300,-2500,-2700,-300$
Still Capture Mode(0x5013):(readwrite) (type=0x4) Enumeration [1,32784,2,32789,32786,32773,32771,32772,32776,32777,32780,32781,32782,32783,33591,34103,35127,33623,34135,35159,33655,34167,35191,33553,34065,35089,33569,34081,33585,34097,33$
DOC Compensation(0xd200):(readwrite) (type=0x3) Enumeration [0,3000,2700,2500,2300,2000,1700,1500,1300,1000,700,500,300,-300,-500,-700,-1000,-1300,-1500,-1700,-2000,-2300,-2500,-2700,-3000] value: 0
DRangeOptimize(0xd201):(readwrite) (type=0x2) Enumeration [1,31,17,18,19,20,21,32,33,34,35,36,37,38] value: 1
Image size(0xd203):(readwrite) (type=0x2) Enumeration [1,2,3] value: 3
Shutter speed(0xd20d):(readwrite) (type=0x6) Range [0 - 4294967295, step 1] value: 65540
Property 0xd20e:(readwrite) (type=0x2) Enumeration [1,2,3,11,8,9,10,4,12,5,13,6,14,7,15,16] value: 13
Color temperature(0xd20f):(readwrite) (type=0x4) Range [2500 - 9900, step 100] value: 0
Aspect Ratio(0xd211):(readwrite) (type=0x2) Enumeration [1,2] value: 1
Focus status(0xd213):(readwrite) (type=0x2) Enumeration [1,2,3,5,6,7] value: 1
ISO(0xd21e):(readwrite) (type=0x6) Enumeration [16777215,25,50,64,80,100,125,160,200,250,320,400,500,640,800,1000,1250,1600,2000,2500,3200,4000,5000,6400,8000,10000,12800,16000,20000,25600,32000,40000,51200,64000,80000,102400,128000,1600$
Picture Effect(0xd21b):(readwrite) (type=0x4) Enumeration [32768,32769,32770,32771,32772,32773,32784,32800,32801,32816,32832,32848,32849,32850,32851,32864,32912] value: 32768
Property 0xd21d:(readwrite) (type=0x2) Range [0 - 2, step 1] value: 0
Property 0xd21f:(readwrite) (type=0x2) Enumeration [2,1] value: 1
Property 0xd217:(readwrite) (type=0x2) Enumeration [2,1] value: 1
Battery Level(0xd218):(readwrite) (type=0x1) Range [-1 - 100, step 1] value: 41
Property 0xd219:(readwrite) (type=0x2) Enumeration [2,1] value: 1
Property 0xd212:(readwrite) (type=0x2) Range [0 - 15, step 1] value: 0
CC Filter(0xd210):(readwrite) (type=0x2) Range [164 - 220, step 1] value: 192
AB Filter(0xd21c):(readwrite) (type=0x2) Range [164 - 220, step 2] value: 192
Property 0xd222:(readwrite) (type=0x4) Enumeration [1,17] value: 17
Property 0xd22c:(readwrite) (type=0x4) Enumeration [1,2,3,257,258,259,260] value: 1
Property 0xd22d:(readwrite) (type=0x2) Enumeration [0,1,2] value: 0
Property 0xd22e:(readwrite) (type=0x4) Enumeration [] value: 59
Property 0xd22f:(readwrite) (type=0x4) Enumeration [] value: 0
Property 0xd230:(readwrite) (type=0x6) Enumeration [] value: 20971734
[... 64 lines cut ...]
```
Gphoto2 version:
```
gphoto2 2.5.20

Copyright (c) 2000-2018 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.20         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
```
**Hardware (please complete the following information)**

 - Raspberry Pi 3B+ and Asus TinkerBoard S
 - Sony Alpha 7 III

**Software (please complete the following information)**

 -  Raspbian Stretch, Armbian and Lubuntu 19.04
 - Python 3.7.3

chino-lu, 2019-06-09:
I get the error as well if connected with my Pentax K20D

reuterbal, 2019-06-13:
The error message should actually not be fatal. 
A number of questions:
1. Does it work if you use command line?
2. If yes, please remove lines 74-99 in [photobooth/camera/CameraGphoto2.py](https://github.com/reuterbal/photobooth/blob/master/photobooth/camera/CameraGphoto2.py). Does it work now?

reuterbal, 2019-06-13:
3. Is your camera supported by gphoto? I don't see Alpha 7 III or Pentax K20D in the list:
http://www.gphoto.org/doc/remote/

chino-lu, 2019-06-13:
it works with command line, but would like to have the liveview.
The Sony Alpha 7 III is listed here:
http://www.gphoto.org/proj/libgphoto2/support.php

and since I updated to gphoto2 2.5.23 it also shows Sony Alpha 7 III (PC Control)

Is this not enough?

reuterbal, 2019-06-13:
That looks absolutely fine. Have you tried 2. above?

chino-lu, 2019-06-13:
I tried to comment out some lines, but I think never that whole part. Will try again tonight

chino-lu, 2019-06-14:
the bad parameters error stays but there is one line difference

ERROR - CameraProcess: Exception "[-2] Bad parameters"

reuterbal, 2019-06-14:
Please try the development branch, it should give some more details on the cause of the issue. If the issue persists, please post the new error messages.
See here on how to change branches: https://github.com/reuterbal/photobooth/wiki/Tips-and-tricks#switch-to-development-branch

chino-lu, 2019-06-20:
I have the dev version installed

`2019-06-20 19:27:38,460 - root - INFO - Initializing photobooth
2019-06-20 19:27:38,461 - root - INFO - Photobooth version: 0.4.dev93+g12fc1a8
2019-06-20 19:27:38,464 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2019-06-20 19:27:38,470 - root - INFO - Reading config file "photobooth.cfg"
2019-06-20 19:27:38,532 - root - INFO - Number of last existing file: 0
2019-06-20 19:27:38,538 - root - INFO - Saving pictures as "2019-06-20/photoboothXXXXX.jpg"
2019-06-20 19:27:38,538 - root - INFO - GPIO disabled
2019-06-20 19:27:38,542 - root - INFO - Number of last existing file: 0
2019-06-20 19:27:38,543 - root - INFO - Saving pictures as "2019-06-20/photobooth_shot_XXXXX.jpg"
2019-06-20 19:27:40,365 - root - INFO - Using printer ""
2019-06-20 19:27:43,312 - root - INFO - Using python-gphoto2 bindings
2019-06-20 19:27:43,636 - root - INFO - Camera summary: Manufacturer: Sony Corporation
Model: ILCE-7M3
  Version: 3.0
  Serial Number: 00000000000000003282943003847211
Vendor Extension ID: 0x11 (1.0)
Vendor Extension Description: Sony PTP Extensions

Capture Formats: 
Display Formats: JPEG, Unknown(b301), ARW

Device Capabilities:
	File Download, No File Deletion, No File Upload
	No Image Capture, No Open Capture, Sony Capture

Storage Devices Summary:

Device Property Summary:
Compression Setting(0x5004):(readwrite) (type=0x2) Enumeration [2,3,4,16,19,20,18] value: RAW (4)
White Balance(0x5005):(readwrite) (type=0x4) Enumeration [2,4,32785,32784,6,32769,32770,32771,32772,7,32816,32786,32800,32801,32802] value: Automatic (2)
F-Number(0x5007):(readwrite) (type=0x4) Range [0 - 65535, step 1] value: f/10 (1000)
Focus Mode(0x500a):(readwrite) (type=0x4) Enumeration [1,2,32772,32773,32774,32777] value: 32773
Exposure Metering Mode(0x500b):(readwrite) (type=0x4) Enumeration [32769,32770,32772,32773,32771,32774] value: 32769
Flash Mode(0x500c):(readwrite) (type=0x4) Enumeration [2,1,4,3,5,32769,32771,32817,32818,32833,32834] value: Fill flash (3)
Exposure Program Mode(0x500e):(readwrite) (type=0x4) Enumeration [32768,2,3,4,1,32848,32849,32850,32851,32900,32901,32902,32903,7,32785,32789,32788,32786,32787,32791] value: A (3)
Exposure Bias Compensation(0x5010):(readwrite) (type=0x3) Enumeration [0,5000,4700,4500,4300,4000,3700,3500,3300,3000,2700,2500,2300,2000,1700,1500,1300,1000,700,500,300,-300,-500,-700,-1000,-1300,-1500,-1700,-2000,-2300,-2500,-2700,-3000,-3300,-3500,-3700,-4000,-4300,-4500,-4700,-5000] value: 0.0 stops (0)
Still Capture Mode(0x5013):(readwrite) (type=0x4) Enumeration [1,32784,2,32789,32786,32773,32771,32772,32776,32777,32780,32781,32782,32783,33591,34103,35127,33623,34135,35159,33655,34167,35191,33553,34065,35089,33569,34081,33585,34097,33590,34102,35126,33622,34134,35158,33654,34166,35190,33552,34064,35088,33568,34080,33584,34096,32792,32808,32793,32809] value: Single Shot (1)
DOC Compensation(0xd200):(readwrite) (type=0x3) Enumeration [0,3000,2700,2500,2300,2000,1700,1500,1300,1000,700,500,300,-300,-500,-700,-1000,-1300,-1500,-1700,-2000,-2300,-2500,-2700,-3000] value: 0
DRangeOptimize(0xd201):(readwrite) (type=0x2) Enumeration [1,31,17,18,19,20,21,32,33,34,35,36,37,38] value: 1
Image size(0xd203):(readwrite) (type=0x2) Enumeration [1,2,3] value: 3
Shutter speed(0xd20d):(readwrite) (type=0x6) Range [0 - 4294967295, step 1] value: 65566
Property 0xd20e:(readwrite) (type=0x2) Enumeration [1,2,3,11,8,9,10,4,12,5,13,6,14,7,15,16] value: 15
Color temperature(0xd20f):(readwrite) (type=0x4) Range [2500 - 9900, step 100] value: 0
Aspect Ratio(0xd211):(readwrite) (type=0x2) Enumeration [1,2] value: 1
Focus status(0xd213):(readwrite) (type=0x2) Enumeration [1,2,3,5,6,7] value: 1
ISO(0xd21e):(readwrite) (type=0x6) Enumeration [16777215,25,50,64,80,100,125,160,200,250,320,400,500,640,800,1000,1250,1600,2000,2500,3200,4000,5000,6400,8000,10000,12800,16000,20000,25600,32000,40000,51200,64000,80000,102400,128000,160000,204800,256000,320000,409600,33554431,16777241,16777266,16777280,16777296,16777316,16777341,16777376,16777416,16777466,16777536,16777616,16777716,16777856,16778016,16778216,16778466,16778816,16779216,16779716,16780416,16781216,16782216,16783616,16785216,16787216,16790016,16793216,16802816,16828416,16879616,16982016,17186816,50331647,33554532,33554632,33554832,33555232,33556032,33557632,33560832,33567232,33580032,33605632,33656832,33759232,33964032] value: 16777215
Picture Effect(0xd21b):(readwrite) (type=0x4) Enumeration [32768,32769,32770,32771,32772,32773,32784,32800,32801,32816,32832,32848,32849,32850,32851,32864,32912] value: 32768
Property 0xd21d:(readwrite) (type=0x2) Range [0 - 2, step 1] value: 0
Property 0xd21f:(readwrite) (type=0x2) Enumeration [2,1] value: 1
Property 0xd217:(readwrite) (type=0x2) Enumeration [2,1] value: 1
Battery Level(0xd218):(readwrite) (type=0x1) Range [-1 - 100, step 1] value: 99
Sensor Crop(0xd219):(readwrite) (type=0x2) Enumeration [2,1] value: 1
Property 0xd212:(readwrite) (type=0x2) Range [0 - 15, step 1] value: 0
CC Filter(0xd210):(readwrite) (type=0x2) Range [164 - 220, step 1] value: 192
AB Filter(0xd21c):(readwrite) (type=0x2) Range [164 - 220, step 2] value: 192
Property 0xd222:(readwrite) (type=0x4) Enumeration [1,17] value: 17
Property 0xd22c:(readwrite) (type=0x4) Enumeration [1,2,3,257,258,259,260] value: 1
Property 0xd22d:(readwrite) (type=0x2) Enumeration [0,1,2] value: 0
Property 0xd22e:(readwrite) (type=0x4) Enumeration [] value: 59
[... cut ...]
Property 0xd2cf:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2d0:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2d2:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2d3:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2d4:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2d1:(readwrite) (type=0x3) Range [-7 - 7, step 1] value: 0

2019-06-20 19:27:43,644 - root - ERROR - CameraProcess: Exception "[-2] Bad parameters"
2019-06-20 19:27:43,652 - root - ERROR - Camera: [-2] Bad parameters
`

reuterbal, 2019-06-21:
Are you sure that is the development branch? I would have expected a trace back from the exception. Could you please post the output of `git status`?

chino-lu, 2019-06-21:
pi@raspberrypi:~/photobooth $ git status
Auf Branch development
Ihr Branch ist auf dem selben Stand wie 'origin/development'.
Änderungen, die nicht zum Commit vorgemerkt sind:
  (benutzen Sie "git add <Datei>...", um die Änderungen zum Commit vorzumerken)
  (benutzen Sie "git checkout -- <Datei>...", um die Änderungen im Arbeitsverzeichnis zu verwerfen)

	geändert:       photobooth/camera/CameraGphoto2.py

Unversionierte Dateien:
  (benutzen Sie "git add <Datei>...", um die Änderungen zum Commit vorzumerken)

	Downloads

keine Änderungen zum Commit vorgemerkt (benutzen Sie "git add" und/oder "git commit -a")

reuterbal, 2019-06-21:
This looks fine. Could you please update your copy to the latest version of the development branch? I have introduced an interface that allows to define model specific configuration options and doesn't apply anything if those don't exist. Thus, in your case, this will hopefully overcome this issue.

chino-lu, 2019-06-23:
where can I specify the model? I could only find an additional option "Upload" (which is by the way super nice!!!!)

Here the log without specifying the camera some where
`2019-06-23 16:29:41,451 - root - INFO - Initializing photobooth
2019-06-23 16:29:41,452 - root - INFO - Photobooth version: 0.4.dev93+g12fc1a8
2019-06-23 16:29:41,465 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2019-06-23 16:29:41,473 - root - INFO - Reading config file "photobooth.cfg"
2019-06-23 16:29:41,567 - root - INFO - Number of last existing file: 0
2019-06-23 16:29:41,577 - root - INFO - Saving pictures as "2019-06-23/photoboothXXXXX.jpg"
2019-06-23 16:29:41,581 - root - INFO - Number of last existing file: 0
2019-06-23 16:29:41,583 - root - INFO - Saving pictures as "2019-06-23/photobooth_shot_XXXXX.jpg"
2019-06-23 16:29:41,583 - root - INFO - GPIO disabled
2019-06-23 16:29:43,460 - root - INFO - Using printer ""
2019-06-23 16:30:30,551 - root - INFO - Writing config file "photobooth.cfg"
2019-06-23 16:30:30,647 - root - INFO - Restarting photobooth and reloading config
2019-06-23 16:30:30,648 - root - INFO - Photobooth version: 0.4.dev93+g12fc1a8
2019-06-23 16:30:30,650 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2019-06-23 16:30:30,671 - root - INFO - Reading config file "photobooth.cfg"
2019-06-23 16:30:30,754 - root - INFO - Number of last existing file: 0
2019-06-23 16:30:30,766 - root - INFO - Saving pictures as "2019-06-23/photoboothXXXXX.jpg"
2019-06-23 16:30:30,774 - root - INFO - Number of last existing file: 0
2019-06-23 16:30:30,761 - root - INFO - GPIO disabled
2019-06-23 16:30:30,776 - root - INFO - Saving pictures as "2019-06-23/photobooth_shot_XXXXX.jpg"
2019-06-23 16:30:37,408 - root - INFO - Using python-gphoto2 bindings
2019-06-23 16:30:38,937 - root - INFO - Camera summary: Manufacturer: Sony Corporation
Model: ILCE-7M3
  Version: 3.0
  Serial Number: 00000000000000003282943003847211
Vendor Extension ID: 0x11 (1.0)
Vendor Extension Description: Sony PTP Extensions

Capture Formats: 
Display Formats: JPEG, Unknown(b301), ARW

Device Capabilities:
	File Download, No File Deletion, No File Upload
	No Image Capture, No Open Capture, Sony Capture

Storage Devices Summary:

Device Property Summary:
Compression Setting(0x5004):(readwrite) (type=0x2) Enumeration [2,3,4,16,19,20,18] value: 16
White Balance(0x5005):(readwrite) (type=0x4) Enumeration [2,4,32785,32784,6,32769,32770,32771,32772,7,32816,32786,32800,32801,32802] value: 0
F-Number(0x5007):(readwrite) (type=0x4) Range [0 - 65535, step 1] value: f/0 (0)
Focus Mode(0x500a):(readwrite) (type=0x4) Enumeration [1,2,32772,32773,32774,32777] value: 0
Exposure Metering Mode(0x500b):(readwrite) (type=0x4) Enumeration [32769,32770,32772,32773,32771,32774] value: 0
Flash Mode(0x500c):(readwrite) (type=0x4) Enumeration [2,1,4,3,5,32769,32771,32817,32818,32833,32834] value: Fill flash (3)
Exposure Program Mode(0x500e):(readwrite) (type=0x4) Enumeration [32768,2,3,4,1,32848,32849,32850,32851,32900,32901,32902,32903,7,32785,32789,32788,32786,32787,32791] value: A (3)
Exposure Bias Compensation(0x5010):(readwrite) (type=0x3) Enumeration [0,5000,4700,4500,4300,4000,3700,3500,3300,3000,2700,2500,2300,2000,1700,1500,1300,1000,700,500,300,-300,-500,-700,-1000,-1300,-1500,-1700,-2000,-2300,-2500,-2700,-3000,-3300,-3500,-3700,-4000,-4300,-4500,-4700,-5000] value: 0.0 stops (0)
Still Capture Mode(0x5013):(readwrite) (type=0x4) Enumeration [1,32784,2,32789,32786,32773,32771,32772,32776,32777,32780,32781,32782,32783,33591,34103,35127,33623,34135,35159,33655,34167,35191,33553,34065,35089,33569,34081,33585,34097,33590,34102,35126,33622,34134,35158,33654,34166,35190,33552,34064,35088,33568,34080,33584,34096,32792,32808,32793,32809] value: 0
DOC Compensation(0xd200):(readwrite) (type=0x3) Enumeration [0,3000,2700,2500,2300,2000,1700,1500,1300,1000,700,500,300,-300,-500,-700,-1000,-1300,-1500,-1700,-2000,-2300,-2500,-2700,-3000] value: 0
DRangeOptimize(0xd201):(readwrite) (type=0x2) Enumeration [1,31,17,18,19,20,21] value: 0
Image size(0xd203):(readwrite) (type=0x2) Enumeration [1,2,3] value: 0
Shutter speed(0xd20d):(readwrite) (type=0x6) Range [0 - 4294967295, step 1] value: 4294967295
Property 0xd20e:(readwrite) (type=0x2) Enumeration [1,2,3,11,8,9,10,4,12,5,13,6,14,7,15,16] value: 15
Color temperature(0xd20f):(readwrite) (type=0x4) Range [2500 - 9900, step 100] value: 0
Aspect Ratio(0xd211):(readwrite) (type=0x2) Enumeration [1,2] value: 0
Focus status(0xd213):(readwrite) (type=0x2) Enumeration [1,2,3,5,6,7] value: 1
ISO(0xd21e):(readwrite) (type=0x6) Enumeration [16777215,25,50,64,80,100,125,160,200,250,320,400,500,640,800,1000,1250,1600,2000,2500,3200,4000,5000,6400,8000,10000,12800,16000,20000,25600,32000,40000,51200,64000,80000,102400,128000,160000,204800,256000,320000,409600,33554431,16777241,16777266,16777280,16777296,16777316,16777341,16777376,16777416,16777466,16777536,16777616,16777716,16777856,16778016,16778216,16778466,16778816,16779216,16779716,16780416,16781216,16782216,16783616,16785216,16787216,16790016,16793216,16802816,16828416,16879616,16982016,17186816,50331647,33554532,33554632,33554832,33555232,33556032,33557632,33560832,33567232,33580032,33605632,33656832,33759232,33964032] value: 0
Picture Effect(0xd21b):(readwrite) (type=0x4) Enumeration [32768,32769,32770,32771,32772,32773,32784,32800,32801,32816,32832,32848,32849,32850,32851,32864,32912] value: 32768
[... cut ...]
  File "/home/pi/photobooth/photobooth/camera/CameraGphoto2.py", line 71, in _setupCamera
    self._printConfig(self._cap.get_config())
  File "/home/pi/photobooth/photobooth/camera/CameraGphoto2.py", line 105, in _printConfig
    config_txt += CameraGphoto2._configTreeToText(config)
  File "/home/pi/photobooth/photobooth/camera/CameraGphoto2.py", line 86, in _configTreeToText
    config_txt += str(chld.get_value())
gphoto2.GPhoto2Error: [-2] Bad parameters
2019-06-23 16:30:39,021 - root - ERROR - Camera: [-2] Bad parameters`

reuterbal, 2019-06-25:
You don't have to specify the model - it is taken automatically from the data provided by the camera.

But now we finally have a useful backtrace! And, most interesting, the issue is due to the output of the camera configuration.
Please remove (or comment out) l. 71:
```
 self._printConfig(self._cap.get_config())
```
and see if it works.

chino-lu, 2019-06-25:
one step forward. The bad parameters message is gone...
What happens now:
1) I see the countdown and the live preview
2) 1st Picture Taken after 8 seconds
3) I see again the countdown but livepreview freezes with a message: Error: Unknown Event type "Capture"
4) I click retry
5) Error Message: Error: [-1] Unspecified Error
6) I click retry
7 ) Error Message: Error: [-1] Unspecified Error but coundown running
8) back at 3)

Log here:
```
2019-06-25 21:22:39,768 - root - INFO - Initializing photobooth
2019-06-25 21:22:39,769 - root - INFO - Photobooth version: 0.4.dev93+g12fc1a8
2019-06-25 21:22:39,774 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2019-06-25 21:22:39,782 - root - INFO - Reading config file "photobooth.cfg"
2019-06-25 21:22:39,863 - root - INFO - Number of last existing file: 0
2019-06-25 21:22:39,870 - root - INFO - Saving pictures as "2019-06-25/photoboothXXXXX.jpg"
2019-06-25 21:22:39,874 - root - INFO - Number of last existing file: 0
2019-06-25 21:22:39,876 - root - INFO - Saving pictures as "2019-06-25/photobooth_shot_XXXXX.jpg"
2019-06-25 21:22:39,877 - root - INFO - GPIO disabled
2019-06-25 21:22:45,840 - root - INFO - Using python-gphoto2 bindings
2019-06-25 21:22:47,304 - root - INFO - Camera summary: Manufacturer: Sony Corporation
Model: ILCE-7M3
  Version: 3.0
  Serial Number: 00000000000000003282943003847211
Vendor Extension ID: 0x11 (1.0)
Vendor Extension Description: Sony PTP Extensions

Capture Formats: 
Display Formats: JPEG, Unknown(b301), ARW

Device Capabilities:
	File Download, No File Deletion, No File Upload
	No Image Capture, No Open Capture, Sony Capture

Storage Devices Summary:

Device Property Summary:
Compression Setting(0x5004):(readwrite) (type=0x2) Enumeration [2,3,4,16,19,20,18] value: 16
White Balance(0x5005):(readwrite) (type=0x4) Enumeration [2,4,32785,32784,6,32769,32770,32771,32772,7,32816,32786,32800,32801,32802] value: 0
F-Number(0x5007):(readwrite) (type=0x4) Range [0 - 65535, step 1] value: f/0 (0)
Focus Mode(0x500a):(readwrite) (type=0x4) Enumeration [1,2,32772,32773,32774,32777] value: 0
Exposure Metering Mode(0x500b):(readwrite) (type=0x4) Enumeration [32769,32770,32772,32773,32771,32774] value: 0
Flash Mode(0x500c):(readwrite) (type=0x4) Enumeration [2,1,4,3,5,32769,32771,32817,32818,32833,32834] value: Fill flash (3)
Exposure Program Mode(0x500e):(readwrite) (type=0x4) Enumeration [32768,2,3,4,1,32848,32849,32850,32851,32900,32901,32902,32903,7,32785,32789,32788,32786,32787,32791] value: A (3)
Exposure Bias Compensation(0x5010):(readwrite) (type=0x3) Enumeration [0,5000,4700,4500,4300,4000,3700,3500,3300,3000,2700,2500,2300,2000,1700,1500,1300,1000,700,500,300,-300,-500,-700,-1000,-1300,-1500,-1700,-2000,-2300,-2500,-2700,-3000,-3300,-3500,-3700,-4000,-4300,-4500,-4700,-5000] value: 0.0 stops (0)
Still Capture Mode(0x5013):(readwrite) (type=0x4) Enumeration [1,32784,2,32789,32786,32773,32771,32772,32776,32777,32780,32781,32782,32783] value: 0
DOC Compensation(0xd200):(readwrite) (type=0x3) Enumeration [0,3000,2700,2500,2300,2000,1700,1500,1300,1000,700,500,300,-300,-500,-700,-1000,-1300,-1500,-1700,-2000,-2300,-2500,-2700,-3000] value: 0
DRangeOptimize(0xd201):(readwrite) (type=0x2) Enumeration [1,31,17,18,19,20,21,32,33,34,35,36,37,38] value: 0
Image size(0xd203):(readwrite) (type=0x2) Enumeration [1,2,3] value: 0
Shutter speed(0xd20d):(readwrite) (type=0x6) Range [0 - 4294967295, step 1] value: 4294967295
Property 0xd20e:(readwrite) (type=0x2) Enumeration [1,2,3,11,8,9,10,4,12,5,13,6,14,7,15,16] value: 14
Color temperature(0xd20f):(readwrite) (type=0x4) Range [2500 - 9900, step 100] value: 0
Aspect Ratio(0xd211):(readwrite) (type=0x2) Enumeration [1,2] value: 0
Focus status(0xd213):(readwrite) (type=0x2) Enumeration [1,2,3,5,6,7] value: 1
ISO(0xd21e):(readwrite) (type=0x6) Enumeration [16777215,25,50,64,80,100,125,160,200,250,320,400,500,640,800,1000,1250,1600,2000,2500,3200,4000,5000,6400,8000,10000,12800,16000,20000,25600,32000,40000,51200,64000,80000,102400,128000,160000,204800,256000,320000,409600,33554431,16777241,16777266,16777280,16777296,16777316,16777341,16777376,16777416,16777466,16777536,16777616,16777716,16777856,16778016,16778216,16778466,16778816,16779216,16779716,16780416,16781216,16782216,16783616,16785216,16787216,16790016,16793216,16802816,16828416,16879616,16982016,17186816,50331647,33554532,33554632,33554832,33555232,33556032,33557632,33560832,33567232,33580032,33605632,33656832,33759232,33964032] value: 0
Picture Effect(0xd21b):(readwrite) (type=0x4) Enumeration [32768,32769,32770,32771,32772,32773,32784,32800,32801,32816,32832,32848,32849,32850,32851,32864,32912] value: 32768
Property 0xd21d:(readwrite) (type=0x2) Range [0 - 2, step 1] value: 0
Property 0xd21f:(readwrite) (type=0x2) Enumeration [2,1] value: 0
Property 0xd217:(readwrite) (type=0x2) Enumeration [2,1] value: 1
Battery Level(0xd218):(readwrite) (type=0x1) Range [-1 - 100, step 1] value: 78
Sensor Crop(0xd219):(readwrite) (type=0x2) Enumeration [2,1] value: 1
Property 0xd212:(readwrite) (type=0x2) Range [0 - 15, step 1] value: 0
CC Filter(0xd210):(readwrite) (type=0x2) Range [164 - 220, step 1] value: 192
AB Filter(0xd21c):(readwrite) (type=0x2) Range [164 - 220, step 2] value: 192
Property 0xd222:(readwrite) (type=0x4) Enumeration [1,17] value: 17
Property 0xd22c:(readwrite) (type=0x4) Enumeration [1,2,3,257,258,259,260] value: 0
Property 0xd22d:(readwrite) (type=0x2) Enumeration [0,1,2] value: 0
[... cut ...]
Traceback (most recent call last):
  File "/home/pi/photobooth/photobooth/main.py", line 164, in mainloop
    exit_code = context.handleEvent(event)
  File "/home/pi/photobooth/photobooth/StateMachine.py", line 84, in handleEvent
    self.state.handleEvent(event, self)
  File "/home/pi/photobooth/photobooth/StateMachine.py", line 289, in handleEvent
    raise TypeError('Unknown Event type "{}"'.format(event))
TypeError: Unknown Event type "capture"
2019-06-25 21:30:53,858 - root - ERROR - Gpio: Unknown Event type "capture"
```

reuterbal, 2019-06-25:
That sounds a lot like there are plenty of upstream issues in gphoto2 with your camera model. In particular, since the error appears during capturing preview (which is a single, parameter-less call to the library) and fails with "[-1] Unspecified error".

I suggest to do the following: Disable preview in the settings and see, if everything works as expected (except for the missing preview, of course). If it doesn't, post the issues.

If everything works, you will probably have to try to figure out the issues upstream with the gphoto2-developers. It looks very much like the camera model is not properly support by the library, yet.

reuterbal, 2019-06-27:
Another thought: the [support list](http://gphoto.org/proj/libgphoto2/support.php) mentions the required capabilities only for "PC Control" mode, not for "MTP" mode. Your log output says
```
Vendor Extension Description: Sony PTP Extensions
```
suggesting that your camera is in MTP-mode. Maybe try to find out (e.g., in the camera settings) if you can switch to the "PC Control"-mode.

nuscreativelabs, 2019-07-26:
Even I'm having the parameters error originally mentioned. I have an A7R III. I tried the option with PC Mode manually selected in camera. 

Currently trying your dev branch.