reuterbal/photobooth #140: Slide Show when idle

Blokhuizen, 2019-06-10:
Over 4 weeks my son has his last day at his Elementary School.
We as parents are organising a party and i'm creating a photobooth using your fantastic solution.

But, since we are not using a printer, we would like to show the taken pictures of that day, on a separate screen. I'm looking for a solution so the photo's are stored on a network drive or uploaded to a secure website.

But, I can't seem to find the solution for this.
So, a alternetive is asking you if it is possible to create a slideshow when the photobooth is idle.

I'm really new at programming for raspberry .....

Thank you for your reply (even it is bad news hehe)

reuterbal, 2019-06-13:
Hi,
I won't have time to implement any slideshow feature, but you can use a second computer and copy the files via SSH in a loop. I did this with the first version of the photobooth - have a look at the old code version here: https://github.com/reuterbal/photobooth/tree/v0.1
Simply use the slideshow.py and change the configuration in the beginning of the file to match your requirements.

Blokhuizen, 2019-06-30:
Hi ,

Thank you for your reply and i understand completley.
I've found another solution for presenting the pictures. I've made a share on the raspberry and a other computer will take the pictures from there to show it in a slideshow.

BUT.

Now I've got another problem.

I've got one big red button on a pole.
I want to use the buttan as a trigger for making the picture (as yet implemented on the GPIO pin)
But I also want to use the same button to cancel the taken picture.
If i fill in the same GPIO-pin it does not work...

Can you help me with that?
I tried to find it in the code, but i'm just a rookie and cant find how to solve this....

hope you can help.

reuterbal, 2019-07-05:
What do you mean by cancel? All GPIO-related code is in [photobooth/gpio/__init__.py](https://github.com/reuterbal/photobooth/blob/master/photobooth/gpio/__init__.py)

Blokhuizen, 2019-07-05:
When the picture is taken, you can cancel it and take a new one.

What i want is use one button for taking the picture and canceling.

reuterbal, 2019-07-05:
That is not easily possible. However, you can simply insert
```python
self.enableTrigger()
```
in l. 181 of [`photobooth/gpio/__init__.py`](https://github.com/reuterbal/photobooth/blob/master/photobooth/gpio/__init__.py). This way, the trigger button is already active again when the final image is shown. This way, pressing the button does not bring you back to the idle screen but instead triggers a new set of photos.