reuterbal/photobooth #141: No module named 'gphoto2'

FizzyMUC, 2019-06-16:
Dear all,

I recently installed the photobooth, following the installation guideline provided here. However I'm experiencing an issue when starting the box via "python -m photobooth" out of the virtual environment. 
The application itself loads up, but when I hit start the spinning wheel just stays there forever. The terminal shows (besides some other lines) the following issue:

> File "/home/pi/photobooth/photobooth/camera/CameraGphoto2.py", line 25, in <module>
> import gphoto2 as gp
> ImportError: No module named 'gphoto2'

I tried installing it again, however the system then shows that I already have the latest version installed.

What am I missing here? :)

FizzyMUC, 2019-06-17:
I should add: It works with when I select gphoto-commandline! Can anyone help?

I should mention that I'm running on Raspbian Stretch Desktop, together with a Canon 6D.

oelegeirnaert, 2019-06-17:
Hi @FizzyMUC ,
First you need to activate your virtual environment.
`source .venv/bin/activate`
Afterwards you need to install the gphoto library via pip install.
`sudo apt install gphoto2 libgphoto2-dev # to use gphoto2`

Hope this solves your issue!

FizzyMUC, 2019-06-17:
Thanks, will try later! Then I suggest to update the manual, because there is not written to install the library within the virtual environment if I'm not mistaken.

oelegeirnaert, 2019-06-17:
If you install it globally it should work as well... But I guess you didn't installed it at all.

FizzyMUC, 2019-06-17:
Yes I did. In this case (I had gphoto2 already installed) it mentioned that one additional file will be downloaded and installed only (the libgphoto2-dev) and I remember wondering if the -dev was another package. So it IS installed, and also I assume globally. But let me check when I get home tonight...

FizzyMUC, 2019-06-17:
Just checked...
"libgphoto2-dev is already the latest version (2.5.12-1)"

Strange...

reuterbal, 2019-06-18:
apt always installs things globally - it is the package manager for the OS. However, PIP is a package manager for python and inside a virtual environment it installs packages locally.

I thing @oelegeirnaert was already on the right track but posted the wrong command to install the PIP package. In fact, gphoto must be installed correctly as the command line module works.

@FizzyMUC did you run ```pip install -e .``` after activating the virtual environment? See also https://github.com/reuterbal/photobooth/blob/master/INSTALL.md#install-photobooth-with-dependencies for the relevant step.

FizzyMUC, 2019-06-19:
Hi @reuterbal. I did run it but honestly nothing happened. After a couple minutes I killed the process. When I add the invidual package behind `pip install -e .` there is an error message...

reuterbal, 2019-06-19:
That very much sounds like there is something broken with your Python setup, virtual environment, etc. Please provide the output of the following commands after activating your virtual environment:
```python --version```
```pip freeze```

FizzyMUC, 2019-06-20:
`python --version`
Python 3.5.3

`pip freeze`
automationhat==0.1.0
blinker==1.3
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
chardet==2.3.0
click==6.6
colorama==0.3.7
colorzero==1.1
cryptography==1.7.1
drumhat==0.1.0
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==0.12.1
fourletterphat==0.1.0
gpiozero==1.5.0
idna==2.2
itsdangerous==0.24
Jinja2==2.8
keyring==10.1
keyrings.alt==1.3
MarkupSafe==0.23
microdotphat==0.2.1
mote==0.0.4
motephat==0.0.2
numpy==1.12.1
oauthlib==2.0.1
pantilthat==0.0.7
pgzero==1.2
phatbeat==0.1.1
pianohat==0.1.0
picamera==1.13
picraft==1.0
piglow==1.2.4
pigpio==1.38
Pillow==4.0.0
pkg-resources==0.0.0
pyasn1==0.1.9
pycrypto==2.6.1
pygame==1.9.3
pygobject==3.22.0
pyinotify==0.9.6
PyJWT==1.4.2
pyOpenSSL==16.2.0
pyserial==3.2.1
python-apt==1.1.0b5
pyxdg==0.25
rainbowhat==0.1.0
requests==2.12.4
requests-oauthlib==0.7.0
RPi.GPIO==0.6.5
RTIMULib==7.2.1
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
sense-hat==2.2.0
simplejson==3.10.0
six==1.12.0
skywriter==0.0.7
sn3218==1.2.7
spidev==3.3
ssh-import-id==5.6
touchphat==0.0.1
twython==3.4.0
unicornhathd==0.0.4
urllib3==1.19.1
virtualenv==15.1.0
Werkzeug==0.11.15

reuterbal, 2019-06-21:
There is indeed `gphoto2` missing, which should have been installed when you run `pip install -e .`. You can try to install it manually using `pip install gphoto2`.