reuterbal/photobooth #142: RPI 4B

oelegeirnaert, 2019-06-25:
Hello,

Is their anybody who tried the photobooth on the new RPI4B?
https://www.raspberrypi.org/blog/raspberry-pi-4-on-sale-now-from-35/

I just bought the Odroid N2, but as it is 64bit I do have a lot of issues with modules that are only 64 bit.

Kind Regards,
Oele Geirnaert

reuterbal, 2019-06-27:
Great question - I don't have a RPI4B, yet. Also, I did not have time to test it on Raspbian buster - I'd be interested to learn whether that causes any issues.

oelegeirnaert, 2019-06-28:
Yeah, the shop I used before (https://www.sossolutions.nl) will have the RPI4B 4GB on the 2th of july. Once available I'll buy it and test it... I'll keep you posted!

SkiTheSlicer, 2019-06-28:
I don't have a RPi 4B, but I am running Rasbian Buster (the OS that RPi 4B's require) because I'm trying to work out some issues with qt5 on Rasbian Stretch.

As of right now, the folks over at [piwheels.org](https://blog.piwheels.org/raspberry-pi-4-and-raspbian-buster/) who make the pip packages for ARM haven't build all the latest packages for it. This step will fail: `pip install -e .` with `ERROR: No matching distribution found for opencv-python (from photobooth==0.5.dev3+g40f1cda)`. So, unless you want to build from source, you might need to wait a bit before the RPi 4B can run photobooth.

reuterbal, 2019-07-05:
That is a good point, thank you for pointing that out!

oelegeirnaert, 2019-09-03:
Hello,
I recently received my RPI4's.
Everything seems to work fine except 'opencv-python' needs to be removed from the setup.py.
Next you should be able to run the installation with the following command: pip install -e .

https://github.com/reuterbal/photobooth/blob/master/setup.py#L187