reuterbal/photobooth #144: Raspbian + photobooth autoinstall

wikijm, 2019-06-27:
Hi @reuterbal 

I would like to propose a solution that would allow us to:

- Download the latest Raspbian image
- Create, in a graphic way, a "recipe" to play at the 1st launch of the Raspberry (installation and configuration of the OS + photo booth)
- Prepare the 'recipe' to be played at each start (automatic launch of Photobooth)
- Write the prepared image in SD/MicroSD card, ready-to-use

This method of setting up the Photobooth would be based on the PiBakery software (https://www.pibakery.org/index.html)

Here is the result of a 1st version, mostly based on your install instructions:
[image]

What do you think?
If you're interested, I don't know what the most appropriate form would be. To integrate in the [INSTALL.md](https://github.com/reuterbal/photobooth/blob/master/INSTALL.md) file? In the [Wiki](https://github.com/reuterbal/photobooth/wiki)? Both?

reuterbal, 2019-06-27:
That is an excellent idea. I have a few suggestions:

- Is it possible to force the user to specify a password for the user `pi`? (to make sure people don't run the photobooth with default password enabled)
- Include [raspap](https://raspap.com/) into the installation
- The "on every boot" section can be reduced to a single "run command":
```/home/pi/photobooth/.venv/bin/python -m photobooth --run```
No need to change the directory or activate the venv before that.

I'd prefer to have the instructions in a wiki page - moving all the installation instructions to the wiki is on my todo list (you could do that if you want and find the time?!). In the long run I want to have only a bare minimum in the INSTALL.md file (like requirements and only the platform-independent part about the software).

reuterbal, 2019-06-27:
I prepared an [Installation](https://github.com/reuterbal/photobooth/wiki/Installation) page with the rough structure into subpages that I would like to have. Feel free to include the Pibakery option as a possible installation path in the Raspberry Pi specific documentation.

wikijm, 2019-06-27:
>     * Is it possible to force the user to specify a password for the user `pi`? (to make sure people don't run the photobooth with default password enabled)
Yes, like this:
[image]

>     * Include [raspap](https://raspap.com/) into the installation
Done
>     * The "on every boot" section can be reduced to a single "run command":
>       `/home/pi/photobooth/.venv/bin/python -m photobooth --run`
>       No need to change the directory or activate the venv before that.
Done too
> I'd prefer to have the instructions in a wiki page - moving all the installation instructions to the wiki is on my todo list (you could do that if you want and find the time?!). In the long run I [...]

Were are in the same page for this 😉 :

wikijm, 2019-06-28:
Hi @reuterbal 

I've just added some text on the Wiki:

- [X] https://github.com/reuterbal/photobooth/wiki/Installation
- [X] https://github.com/reuterbal/photobooth/wiki/Raspbian-and-photobooth-image-creation-with-PiBakery

If it's OK, i will close this issue.

reuterbal, 2019-06-28:
That looks very good, thank you! 
My suggestion would be to have the link to your PiBakery-Page at the top of https://github.com/reuterbal/photobooth/wiki/Installation-on-Raspberry-Pi (since this is specific for Raspi) but we can leave it like that for now. Feel free to close the issue.

wikijm, 2019-06-28:
I just applied what you recommended.
I took the opportunity to re-link to the INSTALL.md history file for the more adventure-seeker.

closed 2019-06-28 by wikijm