reuterbal/photobooth #145: [Success story] Portable and affordable webcam photobooth

wikijm, 2019-06-28:
Hi @reuterbal,

Main goal: Having a quick-and-dirty and portable photobooth solution for a birthday party.
This version is not made to be long-lasting, but has the merit of offering an affordable solution (recycled box, low-cost sports camera) and easily transportable.

**Hardware:**
 - Device:     Raspberry Pi 3 B
 - Camera:   [TOPELEK Caméra Sport Étanche HD 1080P 16MP Étanche 30M 170°](https://www.cdiscount.com/photo-numerique/camescope-numerique/topelek-camera-sport-etanche-hd-1080p-16mp-batteri/f-1120223-top7933463628791.html)
 - GPIO:     Yes

**Additional Equipment:**

| Picture | Name | Price
| :--------------: | :--------------: | :--------------:
| [image] | SKIP capsule plastic box | N/A
| [image] | [TOPELEK Caméra Sport Étanche HD 1080P 16MP Étanche 30M 170°](https://www.cdiscount.com/photo-numerique/camescope-numerique/topelek-camera-sport-etanche-hd-1080p-16mp-batteri/f-1120223-top7933463628791.html) | 35,99 €
| [image] | [Chargeur Batterie Externe Portable 25000mAh GRDE Ultra-haute Capacité Power Bank 2 Porte USB avec Ultra Lumineux LED Torche et 4 Lumiere Indicateur Puissance pour iphone ipad Xiaomi Huawei Sony etc](https://www.amazon.fr/Chargeur-GRDE-Ultra-haute-Indicateur-Puissance/dp/B07D78Y4DN/ref=sr_1_2_sspa?__mk_fr_FR=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=3065UIXKB0ARC&keywords=batterie+20000mah&qid=1561713006&s=gateway&sprefix=batterie+200%2Caps%2C296&sr=8-2-spons&psc=1)  | 25,99 €
| [image] | [Aukru Raspberry Pi 3 Power supply 5V 3000mA with switch](https://www.amazon.fr/Aukru-Chargeur-Adaptateur-Alimentation-Raspberry/dp/B01M58O9M9/ref=sr_1_19?ie=UTF8&qid=1529013208&sr=8-19&keywords=raspberry%2Bpi%2B2&th=1) | 9,99 €
| [image] | [Waveshare 7inch HDMI LCD (C) Capacitive Touch Screen Display Supports Various Systems for All Ver. Raspberry pi 3 Model B/ 2B/B+/B/A BeagleBone Black Banana Pi/Pro Video Photo Kit with Bicolor Case](https://www.amazon.fr/Waveshare-LCD-HDMI-r%C3%A9solution-Raspberry/dp/B01HPV7M4I/ref=sr_1_58?ie=UTF8&qid=1529013265&sr=8-58&keywords=raspberry%2Bpi%2B2&th=1)  | 67,99 €
| [image] | [EG Starts zéro Delay USB Encoder Pour Ordinateur Jeux Manette Rouge + 10 x LED lumineux 5 V boutons pour arcade joystick Kit DIY Parties Mame Raspberry Pi 2 3 3B ](https://www.amazon.fr/gp/product/B075DFNK24/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) | 25,99 €

**Schema:**
More to come

**Software:**
 - OS: Raspbian GNU/Linux 9 (stretch) (partial result of `cat /etc/*-release` command)
 - Python: 3

**Modifications:**
N/A

**Problems:**
   - Arcade buttons and joystick are not in use for now, more to come

**Pictures:**
[image]
[image]

If you have any questions or need feedback, please do not hesitate to contact me 😉

reuterbal, 2019-06-28:
That's a cool idea! Is there a reason why you chose a sports cam instead of the picam, which would have been small enough to hide it in the box.

wikijm, 2019-06-28:
PiCcam would have been convenient indeed.

The reasons are simple:

- I have a sports camera but no PiCam.
- I started by recycling Logitech USB webcams. It worked very well, but I prefer to use an ultra wide angle lens